iNavFlight/inav #7941: Inav4.1 motor issue

Grumpyrse, 2022-03-31:
updated to 4.1 from 3.02 on my geprc croc 7 and now motors glitch, skip and run hot .
Reverted back to 3.02 and they don't skip, run cool with no issues whatever. Anyone have any ideas?

StuweFPV, 2022-04-04:
did you check your ESC protocol in the Outputs tab? I don't remember in witch version it was but at one point the default protocol changed and it gave me a similar behavior as you describe. setting the correct protocol again fixed the issue.

Grumpyrse, 2022-04-04:
Thankyou for suggestion, I think I did check but will have another look...

StuweFPV, 2022-04-04:
I hope it works out for you! whenever i had issues like the one you described it had something to do with how the FC talks to the ESC.. maybe also worth checking (if you're using it) your bl_heli settings (and update) just to be safe. Personally I tend not to go over dshot300 - often even only 150. the benefits vs the issues just did not work out for me.. good luck!

Grumpyrse, 2022-04-04:
Thank you for your thoughts, yes I am on dshot600. I tried dshot300 but it still had the same affect, skipping ,clicky, glitching motors which run hot. I'm only a novice at this and not sure if the blackbox data could point me in the right direction.Not sure how to interpret it either !! Contemplating trying BF4.3 to see if the issues are there also. Anyways, thanks again.

DzikuVx, 2022-05-07:
@Grumpyrse please watch this; https://youtu.be/zukTkEER-24

closed 2022-05-07 by DzikuVx