iNavFlight/inav #7953: T-Motor F7 Pro V2 target missing internal 16MB flash blackbox

FPVMartin, 2022-04-06:
## Current Behavior

## could it be possible to add the flash blackbox to this target?

Let me know if you need a "dump" or "diff" file.

Thank you.

- FC Board name and vendor: T-Motor F7 Pro V2 from T-Motor
- INAV version string: INAV 4.1 on FC (see attached screenshot)

[image]

[image]

FPVMartin, 2022-04-06:
Here is some info on the FC

Specifications

Mounting Pattern: 30.5x30.5mm/M3
Dimensions: 37x37x7.5mm
MCU: STM32F722 RET6
Target: TMOTOR F7 V2
Blackbox: 16MB Flash
Built-in OSD: AT7456
ESC Signal: M1-M8
BEC: 10V/2A, 5V/3A
Barometer: BMP280
Input Voltage: 3-8S
Motor Outputs: 8
IMU: MPU6000
USB: Type-C
Weight: 8.1g
UARTS: 6
WiFi: Yes

OptimusTi, 2022-04-06:
Are you sure you are using the right target?

FPVMartin, 2022-04-06:
I've been using the TMOTOR F7 V2 target. That's the e only TMotor target I saw in INAV.

FPVMartin, 2022-04-06:
I did re-flash it just in case there was an issue with the first flash but still no blackbox login.

OptimusTi, 2022-04-06:
> I've been using the TMOTOR F7 V2 target. That's the e only TMotor target I saw in INAV.

That doesn't mean it's the correct target. There was another target that was removed on 4.1.

Does it work with Betaflight?  Check if it works in EmuFlight.

FPVMartin, 2022-04-06:
I check on BF and it worked. Do I need to download INAV 4.0?

The manufacturer mentions that the target is TMOTOTF7V2.

I'm new to INAV. I'm not familiar with finding proper targets for FC except for what the manufacturer says.

FPVMartin, 2022-04-06:
It was the same target for BF

erstec, 2022-04-06:
@FPVMartin I see TMOTOR F7 V2 differs from TMOTOR F7, mentioned by @OptimusTi, by `SPI3` pins used
But in this case you should lose UART3 TX/RX and ADC2 (Current meter)... as mentioned pins are mapped to those peripherals.

You can build custom FW for "removed" TMOTOR F7 target to test, if you want, by enabling it in CMakeList.txt, in same target forder...

NOTE: In BF code SPI3 related part are totally same to INAV, for both V1 and V2 accordingly

FPVMartin, 2022-04-06:
Ok, this jargon is a bit out of my understanding.

I can try the TMOTORF7 instead of V2

I have GPS on UART 3. I'll have to move it to another UART

erstec, 2022-04-06:
> Ok, this jargon is a bit out of my understanding.
> 
> I can try the TMOTORF7 instead of V2
> 
> I have GPS on UART 3. I'll have to move it to another UART

Ok :) In simple words, if you have working UART3, in your case it is GPS with TMOTOR F7 V2 target flahed, it seems you have correct target selected.
Can you show photo of your FC?

FPVMartin, 2022-04-06:
Ok, I'll send you a pic as soon as I'm home later today.

Here is a generic idea of what it looks like

FPVMartin, 2022-04-06:
[image]

[image]

FPVMartin, 2022-04-06:
[image]

FPVMartin, 2022-04-06:
Here is a pic of my set up I took when I completed the build last weekend. I'll send you more when I'm home.

erstec, 2022-04-06:
> Here is a pic of my set up I took when I completed the build last weekend. I'll send you more when I'm home.

Thank you. It is more than enough :) I will take a look to code tomorrow

Just to clarify, you are using TMOTORF7V2 on BF and BlackBox works ok?
And few more points: during flashing INAV Full chip erase are turned on? and BlackBox not working without any configuration, just after flashing INAV too?

FPVMartin, 2022-04-06:
Yes, I put on BF and blackbox worked then I flashed INAV with full chip erase and blackbox does not work.

FPVMartin, 2022-04-06:
With TMOTORF7V2 as target

FPVMartin, 2022-04-06:
Thank You for looking into it

FPVMartin, 2022-04-07:
I see. I guess I'll add a micro SD card reader to record the blackbox.

Thank you for looking into it.

I was hoping to copy/paste the blackbox pin out from BF to INAV but that won't do anything.

erstec, 2022-04-07:
@eksperts do it work with EmuFlight? Or @FPVMartin can check that too?

FPVMartin, 2022-04-07:
I didn't try emuflight. I don't have the configurator. I have BF and INAV. I'm looking to use poshold and RTH so I can got through the drone declaration thing here in Canada amd be able to fly in controlled airspace (where our local racetrack is located)

I just assume BF and emuflight are similar that way.

I can always try it if you need more data on this.

Let me know.

FPVMartin, 2022-04-07:
Is the issue with the code in INAV or it's a thing strictly with that board?

erstec, 2022-04-07:
@FPVMartin BetaFlight don't use targets, but use defaults, applied after flashing unified targets FW. EmuFlight uses targets, same as INAV, that why I asking you to try EF.

There is no answer to your second question rn, whithout having exact same FC it is hard to find what wrong and took time ;) I will get same FC in 7-9 workdays, but till that you can assist. You know, getting board doesn't mean having freetime to debug, dig code and etc.

FPVMartin, 2022-04-07:
Of course. I'll install emuflight and see it blackbox works.

eksperts, 2022-04-07:
I checked. Installed Emu
[image]
and BB is there
[image]

cc: @erstec

FPVMartin, 2022-04-07:
Oh wow. So it's strictly an INAV thing

erstec, 2022-04-07:
I just found FC with same behaviour, IFlight F4 SuceX-D AIO, with 8MB onboard flash, but it don't show it in Configurator. I'll can go deep now

erstec, 2022-04-08:
@eksperts @FPVMartin I need some assist, can you check FW attached. flash it, and show me CLI 'status' output, please.

[inav_4.1.0_TMOTORF7V2.hex.zip](https://github.com/iNavFlight/inav/files/8450812/inav_4.1.0_TMOTORF7V2.hex.zip)

eksperts, 2022-04-08:
roger @erstec:
```
# status
System Uptime: 13 seconds
Current Time: 2041-06-28T01:04:00.000+00:00
Voltage: 0.00V (1S battery - NOT PRESENT)
CPU Clock=216MHz, GYRO=MPU6000, ACC=MPU6000, BARO=BMP280
STM32 system clocks:
  SYSCLK = 216 MHz
  HCLK   = 216 MHz
  PCLK1  = 54 MHz
  PCLK2  = 108 MHz
Sensor status: GYRO=OK, ACC=OK, MAG=UNAVAILABLE, BARO=OK, RANGEFINDER=NONE, OPFLOW=NONE, GPS=NONE, IMU2=NONE
Stack size: 6144, Stack address: 0x20010000, Heap available: 2028
I2C Errors: 285, config size: 8830, max available config: 16384
ADC channel usage:
   BATTERY : configured = ADC 1, used = ADC 1
      RSSI : configured = ADC 3, used = none
   CURRENT : configured = ADC 2, used = ADC 2
  AIRSPEED : configured = none, used = none
System load: 6, cycle time: 1014, PID rate: 986, RX rate: 49, System rate: 10
Arming disabled flags: ACC HWFAIL RX CLI
VTX: not detected
SPI FLASH Status: 3 0 16777215 - 256 0 0 0 0
```

erstec, 2022-04-08:
> roger @erstec:
>
> ```
> # status
> System Uptime: 13 seconds
> Current Time: 2041-06-28T01:04:00.000+00:00
> Voltage: 0.00V (1S battery - NOT PRESENT)
> CPU Clock=216MHz, GYRO=MPU6000, ACC=MPU6000, BARO=BMP280 [...]

Thank you!

Here is next one. Do the same if you can.

[inav_4.1.0_TMOTORF7V2.hex.zip](https://github.com/iNavFlight/inav/files/8451920/inav_4.1.0_TMOTORF7V2.hex.zip)

FPVMartin, 2022-04-08:
I don't know how to open this file. 😬

erstec, 2022-04-08:
> I don't know how to open this file. 😬

You don't need to open it, just run INAV configurator, go to Firmware Flasher tab (don't press Connect after start) and then press Load Firmware [Local] instead of you doing [Online] normally, select provided HEX (unzip it of course before) and then Flash Firmware. Simple :)

FPVMartin, 2022-04-08:
here you go

status
System Uptime: 14 seconds
Current Time: 2041-06-28T02:04:00.000+01:00
Voltage: 0.00V (1S battery - NOT PRESENT)
CPU Clock=216MHz, GYRO=MPU6000, ACC=MPU6000, RANGEFINDER=MSP, OPFLOW=MSP
STM32 system clocks:
  SYSCLK = 216 MHz
  HCLK   = 216 MHz
  PCLK1  = 54 MHz
  PCLK2  = 108 MHz
Sensor status: GYRO=OK, ACC=OK, MAG=UNAVAILABLE, BARO=UNAVAILABLE, RANGEFINDER=FAILING, OPFLOW=FAILING, GPS=NONE, IMU2=NONE
Stack size: 6144, Stack address: 0x20010000, Heap available: 1924
I2C Errors: 65, config size: 8830, max available config: 16384
ADC channel usage:
   BATTERY : configured = ADC 1, used = ADC 1
      RSSI : configured = ADC 3, used = none
   CURRENT : configured = ADC 2, used = ADC 2
  AIRSPEED : configured = none, used = none
System load: 10, cycle time: 505, PID rate: 1980, RX rate: 49, System rate: 9
Arming disabled flags: NAV HWFAIL RX CLI
VTX: not detected
SPI FLASH Status: 3 0 0 16777215 - 1 1 2 - 256 0 0 0 0

#

erstec, 2022-04-10:
Okay, it is fixed and tested.

For those who need 4.1.0 version with fix -  HEX attached.
[inav_4.1.0_TMOTORF7V2.hex.zip](https://github.com/iNavFlight/inav/files/8459626/inav_4.1.0_TMOTORF7V2.hex.zip)

PR will follow

closed 2022-05-07 by DzikuVx