iNavFlight/inav #7967: Frsky sbus failsafe/inav/Matek problems

m-fuka, 2022-04-15:
### Discussed in https://github.com/iNavFlight/inav/discussions/7963

<div type='discussions-op-text'>

<sup>Originally posted by **m-fuka** April 13, 2022</sup>
I have 5 sets of aircraft in the configuration of Matek f405wing and f405wse with R9 receivers. 

After updating all Frsky R9 receivers, (custom) failsafe stopped working via sbus. (custom failsafe rx pwm output works fine)

  Even if I have set my own failsafe, fc matek evaluates failsafe HOLD !!!

When the transmitter is switched off, the fc Matek will be switched to rth due to the loss of the connection, but the own channels that I have preset will not be preset.

Matek channel 5 = modes
Transmitter off / failsafe on channel 5 = 100%
Matek mode nav rth on = 100%

I solved it with the representation of Frsky, allegedly it is not a mistake of Frsky but a mistake of misinterpretation of the sbus inav or matek.

The problem is mainly with through channels where I need to have my own values in case of connection failure.

I only have this problem with Frsky/Inav/Matek , for example with frsky/qgroundcontrol/pixhawk4 it works properly.

Has anyone solved the same problem?
Thanks.</div>

MrD-RC, 2022-04-15:
Please keep this in the discussion, don’t keep creating issues.

closed 2022-04-15 by MrD-RC

m-fuka, 2022-04-15:
I'm sorry, I placed it wrong, language barrier