iNavFlight/inav #7971: iNav not arming but it doesn't give the cause

francis916, 2022-04-17:
## Current Behavior

iNav 4.1 on Matek F411-WSE won't arm without giving the reason.

# status
System Uptime: 337 seconds
Current Time: 2022-04-16T13:19:53.796+00:00
Voltage: 10.92V (3S battery - OK)
CPU Clock=96MHz, GYRO=MPU6000, ACC=MPU6000, BARO=SPL06
STM32 system clocks:
  SYSCLK = 96 MHz
  HCLK   = 96 MHz
  PCLK1  = 48 MHz
  PCLK2  = 96 MHz
Sensor status: GYRO=OK, ACC=OK, MAG=NONE, BARO=OK, RANGEFINDER=NONE, OPFLOW=NONE, GPS=OK, IMU2=NONE
Stack size: 6144, Stack address: 0x20020000, Heap available: 1728
I2C Errors: 0, config size: 8808, max available config: 16384
ADC channel usage:
   BATTERY : configured = ADC 1, used = ADC 1
      RSSI : configured = ADC 3, used = none
   CURRENT : configured = ADC 2, used = ADC 2
  AIRSPEED : configured = ADC 4, used = none
System load: 15, cycle time: 1027, PID rate: 973, RX rate: 90, System rate: 9
Arming disabled flags: CLI
VTX: band: R, chan: 5, power: 1 (25 mW), freq: 5806 MHz

## Steps to Reproduce

1. Install iNave 4.1 on Matek F411-WSE
2. Connect GPS, Spektrum SPM4650
3. Turn ARM switch on after GPS fixing

## Expected behavior
 FC armed.

## Suggested solution(s)
 Not known. 

## Additional context

Hi. I setup iNav 4.1 on Matek F411-WSE but it won't arm without giving the reason. 

- I tried it after getting GPS fixing and also tried nav_extra_arming_safety OFF. 

- I noticed that throttle value is 6% when ARM switch is on and lowered it to 0 by trimming at NX8. Still didn't arm. 

- I didn't install the camera because it's meant to be a LOS plane for me but I installed the VTX to see error messages when I couldn't arm it.

- Motor works fine in 'Motors' tab in iNav. ESC calibration was done at the tab - increasing it to 100% before connecting the battery and decreasing it to 0% in a few seconds. 

I think it's a bug because it is not arming giving a reason. I hope it's not a stupid mistake of mine.

Thank you.

--------Components
FC Matek F411-WSE
ESC AtomRC 30A 2~4S BLS ESC
Model AtmoRC Mobula 650mm Delta Wing
RX Spektrum SPM4650
TX Spektrum NX8
GPS Beitian BN-220
VTX Goku VTX625
Cam None

https://user-images.githubusercontent.com/18643755/163711204-5998bf95-3f24-4efa-8f9c-ab20639bbee4.mov

[image]
[image]
[image]
[image]
[image]
[image]
[image]

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

https://pastebin.com/aXq4334S
---

- FC Board name and vendor: Matek F411-WSE
- INAV version string:# INAV/MATEKF411SE 4.1.0 Feb 11 2022 / 10:58:30 (611d26f4)
# GCC-10.2.1 20201103 (release)

francis916, 2022-04-17:
I added a capacitor to the FC, soldered at the 2nd motor tab. Could it be the cause by any chance? Thanks.

Jetrell, 2022-04-21:
As the OSD shows, its not seeing low throttle.
You don't want to touch the radio trims. Try setting the idle throttle to 0%... Idle is for multi-rotor use.
Also try adjusting the `min_check`, possibly lower.

francis916, 2022-04-21:
> As the OSD messages show, its not seeing low throttle.
> You don't want to touch the radio trims. Try setting the _Motor idle power_ to 0%... Idle is for multi-rotor use.
> Also try adjusting the `min_check`, possibly lower.

Thanks! Will try.

francis916, 2022-04-21:
Solved! I tried rxrange, Motor idle power to 0%. Thank you!

Jetrell, 2022-04-21:
@francis916 That's good to hear. Please close this help request now.  Thanks

closed 2022-04-22 by francis916