iNavFlight/inav #7982: Diatone Mamba h743 ADC issue

YellowMan87, 2022-04-23:
ADC for Vbat give correct values after "save and reboot" and after you plug in a battery, but is static and do not change until the next "save and reboot" or powercycle.
All other features seems to work fine.
I think its the same issue as with the Matek H743 (Matek H743 ADC issue #7813).
I havent tested the ADC for current readings, because the values are extremly off. There seems to be another problem.

To reproduce, just flash inav 4.1 on a Diatone Mamba H743 and aktivate ADC Vbat reading and display it on the OSD.

DzikuVx, 2022-04-26:
[inav_4.1.0_MAMBAH743.zip](https://github.com/iNavFlight/inav/files/8560947/inav_4.1.0_MAMBAH743.zip)
@YellowMan87 if you could quickly test the hex from here, it would be amazing

uralm1, 2022-04-26:
ADC current meter is broken too.
I already tried asking about current ADC https://github.com/iNavFlight/inav/issues/7885#issuecomment-1068764052

> Small offtopic question on issue for MK4 fc, I tried to look into...
> Do you have ADC external current meter working on this fc? It works in BF dev 4.3, but shows huge amps in inav 4.1. Not only huge amps, the values are inverted - more current -> gets lower values. Millivolts on current pad are increasing properly, so ESC is good.
> My hw is: Mamba MK4 H743 + HobbyWing Xrotor 60A.
> Or may be linked with https://github.com/iNavFlight/inav/commit/c73cdc9e5759e6ca208e71b507edcc0e9aa4467e or https://github.com/iNavFlight/inav/commit/bc4c8ffb79055d039305b72b47c90d4f7713c781

YellowMan87, 2022-04-26:
I tested the hex file and the voltage reading seems to work fine! Thank you very much!

i would like to ask one more question:
as i understand the target of the Mamba h743 is defined so that you cant use servos in combination with multicopters because the target looks something like this:

const timerHardware_t timerHardware[] = {

    DEF_TIM(TIM5, CH1, PA0, TIM_USE_MC_MOTOR | TIM_USE_FW_SERVO, 0, 0),   // S1
    DEF_TIM(TIM5, CH2, PA1, TIM_USE_MC_MOTOR | TIM_USE_FW_SERVO, 0, 1),   // S2
    DEF_TIM(TIM5, CH3, PA2, TIM_USE_MC_MOTOR | TIM_USE_FW_SERVO, 0, 2),   // S3
    DEF_TIM(TIM5, CH4, PA3, TIM_USE_MC_MOTOR | TIM_USE_FW_SERVO, 0, 3),   // S4

    DEF_TIM(TIM3, CH3, PB0, TIM_USE_MC_MOTOR | TIM_USE_FW_SERVO, 0, 4),   // S5
    DEF_TIM(TIM3, CH4, PB1, TIM_USE_MC_MOTOR | TIM_USE_FW_SERVO, 0, 5),   // S6
    DEF_TIM(TIM8, CH3, PC8, TIM_USE_MC_MOTOR | TIM_USE_FW_MOTOR, 0, 6),   // S7
    DEF_TIM(TIM8, CH4, PC9, TIM_USE_MC_MOTOR | TIM_USE_FW_MOTOR, 0, 7),   // S8

    DEF_TIM(TIM1,  CH1, PA8,  TIM_USE_LED, 0, 8),    // LED_2812
};

would it be possible to change it to this in the inav code:

const timerHardware_t timerHardware[] = {

    DEF_TIM(TIM5, CH1, PA0, TIM_USE_MC_MOTOR | TIM_USE_FW_SERVO, 0, 0),   // S1
    DEF_TIM(TIM5, CH2, PA1, TIM_USE_MC_MOTOR | TIM_USE_FW_SERVO, 0, 1),   // S2
    DEF_TIM(TIM5, CH3, PA2, TIM_USE_MC_MOTOR | TIM_USE_FW_SERVO, 0, 2),   // S3
    DEF_TIM(TIM5, CH4, PA3, TIM_USE_MC_MOTOR | TIM_USE_FW_SERVO, 0, 3),   // S4

    DEF_TIM(TIM3, CH3, PB0, TIM_USE_MC_SERVO | TIM_USE_FW_SERVO, 0, 4),   // S5
    DEF_TIM(TIM3, CH4, PB1, TIM_USE_MC_SERVO | TIM_USE_FW_SERVO, 0, 5),   // S6
    DEF_TIM(TIM8, CH3, PC8, TIM_USE_MC_MOTOR | TIM_USE_FW_MOTOR, 0, 6),   // S7
    DEF_TIM(TIM8, CH4, PC9, TIM_USE_MC_MOTOR | TIM_USE_FW_MOTOR, 0, 7),   // S8

    DEF_TIM(TIM1,  CH1, PA8,  TIM_USE_LED, 0, 8),    // LED_2812
};

i did this by my self and everything seems to work perfectly fine. But i have to compile my own .hex file with every new release...
i think more people are using a servo on a MC than peaople using 8 motors.

or maybe to something like that so people can choose:

const timerHardware_t timerHardware[] = {

    DEF_TIM(TIM5, CH1, PA0, TIM_USE_MC_MOTOR | TIM_USE_FW_SERVO, 0, 0),   // S1
    DEF_TIM(TIM5, CH2, PA1, TIM_USE_MC_MOTOR | TIM_USE_FW_SERVO, 0, 1),   // S2
    DEF_TIM(TIM5, CH3, PA2, TIM_USE_MC_MOTOR | TIM_USE_FW_SERVO, 0, 2),   // S3
    DEF_TIM(TIM5, CH4, PA3, TIM_USE_MC_MOTOR | TIM_USE_FW_SERVO, 0, 3),   // S4

    DEF_TIM(TIM3, CH3, PB0, TIM_USE_MC_MOTOR | TIM_USE_MC_SERVO, 0, 4),   // S5
    DEF_TIM(TIM3, CH4, PB1, TIM_USE_MC_MOTOR | TIM_USE_MC_SERVO, 0, 5),   // S6
    DEF_TIM(TIM8, CH3, PC8, TIM_USE_MC_MOTOR | TIM_USE_FW_MOTOR, 0, 6),   // S7
    DEF_TIM(TIM8, CH4, PC9, TIM_USE_MC_MOTOR | TIM_USE_FW_MOTOR, 0, 7),   // S8

    DEF_TIM(TIM1,  CH1, PA8,  TIM_USE_LED, 0, 8),    // LED_2812
};

but i havent tested this one.

closed 2022-04-27 by DzikuVx

DzikuVx, 2022-04-27:
@YellowMan87 please see #7993