iNavFlight/inav #8000: Strange failsafe RTH behavior when RC lost

flyingzzy, 2022-05-02:
## Current Behavior
Mini Ar wing, all conditions are good.
when about 5km far and 540m high, RC link got lost and the the Failsafe rth was trigered,but it doesnt just turn around and fly back ,instead, it loitered a few cycles and finally decided to emergency landing.
in the video we can see when rth trigered,zhe plane began to turn back ,but the yaw angle seemed to be "frozen",that caused the FC to make more cycles, but the angle was still very strange , so the FC decided to land.
the sensor was always good,even in earlly phase of this flight and in the following flights,it seems that somewhere hides a bug?
note that when i switched to cruise mode and fly towards home direction,fs rth happened again,but this time it doesnt do any strange behaviors. so why?

## Steps to Reproduce
1.FLY
2.far enough to lost rc link,live home behind
3.see what happens.

## Expected behavior
when rc lost,the plane should go straight back home

## Suggested solution(s)
Not sure.

## Additional context
# version
# INAV/OMNIBUSF4V3_S6_SS 4.1.0 Apr  3 2022 / 21:33:23 (6e1dacba)
# GCC-10.2.1 20201103 (release)

# start the command batch
batch start

# resources

# mixer

mmix reset

mmix 0  1.000  0.000  0.000  0.000

# servo mix
smix reset

smix 0 3 0 50 0 -1
smix 1 3 1 50 0 -1
smix 2 4 0 -50 0 -1
smix 3 4 1 50 0 -1

# servo
servo 0 1000 2000 1500 100
servo 1 1000 2000 1500 -100
servo 2 1000 2000 1500 100
servo 3 1000 2000 1500 -100
servo 4 1000 2000 1500 100
servo 5 1000 2000 1500 100
servo 6 1000 2000 1500 100
servo 7 1000 2000 1500 100
servo 8 1000 2000 1500 100
servo 9 1000 2000 1500 100
servo 10 1000 2000 1500 100
servo 11 1000 2000 1500 100
servo 12 1000 2000 1500 100
servo 13 1000 2000 1500 100
servo 14 1000 2000 1500 100
servo 15 1000 2000 1500 100

# safehome
safehome 0 0 0 0
safehome 1 0 0 0
safehome 2 0 0 0
safehome 3 0 0 0
safehome 4 0 0 0
safehome 5 0 0 0
safehome 6 0 0 0
safehome 7 0 0 0

[... 1511 lines cut ...]
set nav_mc_hover_thr = 1500
set nav_fw_cruise_thr = 1400
set nav_fw_min_thr = 1200
set nav_fw_max_thr = 1600
set nav_fw_pitch2thr = 10
set nav_fw_launch_thr = 1600
set nav_fw_launch_idle_thr = 1300
set limit_cont_current = 0
set limit_burst_current = 0
set limit_burst_current_time = 0
set limit_burst_current_falldown_time = 0
set limit_cont_power = 0
set limit_burst_power = 0
set limit_burst_power_time = 0
set limit_burst_power_falldown_time = 0

# end the command batch
batch end

# 
---

- FC Board name and vendor:  OMNIBUS  F4V3
- INAV version string:OMNIBUSF4V3_S6_SS 4.1.0
- Plane: Mini AR wing

breadoven, 2022-05-03:
The emergency landing was caused by Sanity checking, i.e. it flew 500m further away from home after the RTH was triggered. See https://github.com/iNavFlight/inav/blob/master/docs/Settings.md#nav_rth_abort_threshold.

Not sure why the RTH failed to go home though, difficult to say without a log. Were you manually correcting the CRUZ direction when the failsafe occurred ?

flyingzzy, 2022-05-04:
i have found the reason.
if rth is trigered while the plane is flying against home, fc try to turn around and fly abck,but the gyroscope seems working wrong,the yaw  angle changes very slow, maybe the sensor was affected by the wires of the VTX.i have changed the wire routing,and will do a test later

flyingzzy, 2022-05-04:
[image]

breadoven, 2022-05-04:
I doubt this is the reason. Far more likely related to this bug https://github.com/iNavFlight/inav/pull/7829.

If you were moving the pitch/roll stick when the failsafe occurred RTH Nav control would have become stuck in manual RC adjustment mode preventing RTH from navigating back home.

flyingzzy, 2022-05-05:
I have done a test with nothing  changed. 
set vtx at full power, pick up the plane and turn a few circles, left and right. the gyroscope work very well.
The reason seems to be a code  bug, as you have found.
I have met twice rth error recent days.

flyingzzy, 2022-05-05:
https://user-images.githubusercontent.com/50829593/166956489-4faa5256-13e4-4505-8ce7-8c80652d55a9.mp4

Here is another video, pay attention to the home arrow and the yaw direction change, it seems to be frozen at some moment

breadoven, 2022-05-05:
The bug mentioned above kicks in only if you're moving the controls at the moment the failsafe happens.

Looking at the recent video I'd say there is a problem with horizon drift. When the horizon in the camera view is level the roll angle is around 10 degrees when it should be zero obviously. That drunken rolling movement is what you see when horizon drift is affecting navigation. There is also maybe some drift on the pitch also, but not so obvious. A known problem with INAV, possibly worse on flying wings if they suffer from pitch oscillations.

flyingzzy, 2022-05-06:
> The bug mentioned above kicks in only if you're moving the controls at the moment the failsafe happens.
>
> Looking at the recent video I'd say there is a problem with horizon drift. When the horizon [...]

I understand the reason of the first video,its a code bug which has been found and will be solved in the following release.

The horizon drift is another puzzle, it happens in every flight.If u continuous move the roll stick toward one direction,the  horizon line will go wrong. To correct it,u can move the stick toward the other direction for a while, or u can switch to cruise mode to let the fc correct it. Will this problem be solved in the following release?

breadoven, 2022-05-06:
Horizon drift is a long standing issue with INAV and won't get fixed until someone works out how to fix it. All you can do to minimise it is make sure the plane is mechanically well trimmed and PIDs well tuned.

closed 2022-05-09 by flyingzzy