iNavFlight/inav #8004: Magnetometer auto calibration

sdellava, 2022-05-04:
## Current Behavior

The correct calibration of the magnetometer is essential to allow INAV to correctly activate the navigation modes. However, it happens very frequently that the calibration is not correct. Based on the many experiments performed, I can say that the main cause is linked to the presence of metal masses in the area where the calibration is performed. This means that as soon as the drone lifts up in flight and moves away from the metal masses, the earth's magnetic field is no longer deviated, resulting in an incorrect assessment of the drone's orientation.
This error can lead to the loss of the drone in case of activation of the RTH procedure, since INAV uses the data produced by the magnetometer to calculate the direction towards the destination point. If the magnetometer provides an indication that is not consistent with the real position of the drone, the point is not reached and after a while the route is recalculated, but it will be wrong again leading the drone to turn uncoordinated.
The incorrect calibration of the magnetometer also affects the stabilisation of the horizon.

## Desired Behavior
I suggest to add a magnetometer in flight auto-calibration procedure to the INAV system. 

## Suggested Solution
The operation works as follows:
- by default INAV does not consider valid the information provided by the magnetometer, so it does not use it.
- After take-off, once a minimum horizontal speed threshold is exceeded (e.g. 10 cm/s), a reliable heading value can be obtained from the GPS or calculated.
- At this point INAV knows the orientation of the drone in space (heading, roll, pitch) and can compare this data with those provided by the magnetometer. The difference between these data is the correction parameter that is be used to calibrate the magnetometer.

As long as the horizontal speed is higher than the minimum threshold, it is possible to repeat this operation every N seconds to eliminate measurement errors derived from the environment.  

In navigation:
As long as the horizontal speed is above the minimum threshold, Inav uses the calculated heading provided by the GPS.

When the speed drops below the minimum threshold, the GPS produces an unreliable data, therefore INAV uses the data provided by the magnetometer as it is currently done, but having calibrated it in flight just before the data provided will certainly be correct.

## Who does this impact? Who is this for?
Helicopters and multi-rotor users

breadoven, 2022-05-20:
Surely the fundamental problem with this is wind. Even if you control the motors so it should be flying forward there will be a mismatch between the compass heading and GPS course over ground if there is any wind affecting the quad so how will it know which is correct when it has no idea of the wind direction and strength ?

sdellava, 2022-05-21:
Correct, if the drone is not flying along the roll axis due to the wind or the manoeuvre required by the pilot, the difference between the course on the ground calculated by the GPS and the heading measured by the magnetometers may be high but not representative of the deviation of the magnetic field. 

We have at least two options to solve this problem
a) the pilot knows when the aircraft is not flying in line with the rool axis. A simple implementation of the self-calibration of the mangnetometer could then consist of making it controllable with a flight mode and then associating it with a channel of the transmitter. In practice, the pilot after take-off puts the drone in straight flight and activates the calibration for a few seconds. To check that it has worked, he can activate Position Hold and verify that the drone stands still. If desired, the self-calibration can be integrated as the initial phase of Position Hold, thus eliminating the need to create a dedicated calibration mode.

b) From the gyroscope and accelerometer data, zeroed at the time of the arm, the heading can be calculated. Of course, it is assumed that the drone does not take off sideways. 
This data can be updated and should be fairly reliable. It then allows us to detect whether the drone is flying with a heading equal to the course on the ground provided by the GPS. If yes we can calculate the magetometer correction parameter, if not we wait for the next loop of the magnetometer calibration process. If calibration fails because heading and course do not coincide after N loops, the pilot is warned with a system message.

sdellava, 2023-01-22:
Hi team! is this feature going to be available in the next 6.0 release?

DzikuVx, 2024-03-22:
As since INAv 7.1 mag is not required for RTH, we can close this ticked as not relevant anymore. Thanks

closed 2024-03-22 by DzikuVx