iNavFlight/inav #8005: KAKUTE H7 STRANGE RTH BEHAVIOR

abimaputra78, 2022-05-05:
## Current Behavior
~~~~~~~~~~~~~
5x attempted to perform RTH (Return To Home) had randomized emergency landing 3x could not auto disarm even land has been detected, before landing had a suddenly throttle increased  to reach out unreasonable altitude 

## Steps to Reproduce
<!-- Steps to reproduce the behavior: -->
1. Do.. Setup RTH after minimum distance 1000cm (customize cli) exceeded
2. Do.. assigned  3 pos switch NONE - WP- RTH switch 
3. Do.. activated RTH After exceed min rth distance
4. Do.. installed gps bn220T and QMC5883 generic mag
5. Data.... sat count = 14, HDOP 1.5

## Expected behavior
During several trials of Rth tests as found randomized / unresonable emergency landing with the same route, the same distance, the same sat count, the same everyrhing.  auto disarmed sometimes ok sometimes not ok, during landing throttle up first with unreasonable altitude and landed. 

## Suggested solution(s)
Stability of RTH behavior, landing behavior, auto disarm behavior 

## Additional context
<!-- Add any other context about the problem here.  -->
<!-- Go to CLI, execute `dump` command copy its output to [PasteBin](https://pastebin.com) and provide a link to a paste here -->

---
<!-- PLEASE FILL THIS OUT -->
<!-- Use CLI 'version' command to get version string -->

- FC Board name and vendor: KAKUTEH7 BT / HOLYBRO size 30x30
- INAV version string:4.1 RELEASE

DzikuVx, 2022-05-07:
Unable to say without a blackbox log. Working hypothesis: GPS/Sensor problems. Unable to say without BB log of the event

abimaputra78, 2022-05-10:
Dear Sir, 
After several trials, and correct what I miss in cli. 
I learned and tried using 
NAV_POS_TIMEOUT = 0 instead of 5 and it works. Default was too fast initiated by Inav to decide Emergency Landing Procedure. 

So far, the RTH works. 
Meanwhile we can postpone this issue until further rth issue can be found again.

abimaputra78, 2022-05-10:
Dear Sir,

abimaputra78, 2022-05-10:
Dear Sirs,
I thank you for your kind attention and help. 
Presently I had tried of using nav_pos_timeout=0 instead of default 5. Everything seemed good. 👍
I will do some test flight again and again and let u know while I had this rth issue again.

closed 2022-05-10 by abimaputra78