iNavFlight/inav #8006: HDZero Artificial Horizon very sluggish

MartinHugh, 2022-05-05:
## Current Behavior
Running iNav 4.1.0 with HDZero OSD enabled.

Movement of the artificial horizon lags actual IMU movement by a couple of seconds.

## Steps to Reproduce
1.  Configure OSD with artificial horizon
2.  Suddenly rotate or pitch the aircraft (or FC)
3.  Observe the OSD, and note that the artificial horizon only responds something like 2+ seconds later.

## Expected behavior
Movement expected to be more like analog systems

## Suggested solution(s)
Is the bottleneck actually iNav.  Or is the slow update on the HDZero side.  Or both ?

## Additional context

---

# INAV/MATEKF405SE 4.1.0 Feb 11 2022 / 10:52:32 (611d26f4)
# GCC-10.2.1 20201103 (release)

Jetrell, 2022-05-06:
The VTX/VRX OSD refresh rate is only 1hz at present.  The issue is known about, but it requires some work on the HDzero side, before the developer can implement a fix on the iNav side.

More information on this can be found on the HDzero Discord Community/inav support.

MartinHugh, 2022-05-06:
> The VTX/VRX OSD refresh rate is only 1hz at present. The issue is known about, but it requires some work on the HDzero side, before the developer can implement a fix on the iNav side.
> 
> More information on this can be found on the HDzero Discord Community/inav support.

Thanks for the info @Jetrell.  Is it known whether Divimath are intending to add this work to their next firmware ?

Jetrell, 2022-05-06:
> 
> Is it known whether Divimath are intending to add this work to their next firmware ?

Not that I'm aware. Even the guy who did the iNav side of the code, didn't have any news a week ago.

MartinHugh, 2022-05-07:
I can confirm HDZero are looking at this, and in fact in their development branch of the VRX firmware the lag for artificial horizon with iNav 4.1.0 drops from around 3s to around 1s.  Which suggests that what bottleneck does exist will be back on the iNav side.  They are currently working on another bug in that VRX firmware so no promise of a release date has been made.

MartinHugh, 2022-06-12:
The HDZero firmware update has been released.
https://www.facebook.com/groups/hdzero/posts/410033324357968/

Is there any update optimisation to be done on the iNav side ?

jasc76, 2022-08-15:
Hi

BF OSD is now really good with the 4Hz update rate. Any info when this will be adopted to inav?

geoffsim, 2023-01-03:
This issue can probably be closed now as the issue raised has been addressed.

MrD-RC, 2023-01-03:
The bottleneck is not on the INAV side. It is on the HDZero side. INAV updates the OSD to the VTx at approximately 60Hz. Much faster than 4Hz.

MrD-RC, 2023-01-03:
I agree @geoffsim. The rest of the work needs to be done on the HDZero side. It's not like the OSD can be made any smaller and still be useful.

closed 2023-01-03 by MrD-RC