iNavFlight/inav #8009: Virtual magnetomeeter

sdellava, 2022-05-05:
## Current Behavior
At present, INAV is unable to use flight modes in automatic navigation if a magnetometer is missing.
This also affects the RTH function in case of radio link failure.

I have recently issued a feature request to improve the calibration of the magnetometer, but there is also the possibility to avoid using the magnetometer at all, as I explain below.

## Desired Behavior
Use the GPS calculated heading in place of the magnetometer.

## Suggested Solution
The GPS (with 3D fix) provides a reliable direction of travel. It calculates the direction in which the vehicle is flying using the current coordinates and those of a point taken a few moments earlier.

If the speed of the vehicle is high enough, the calculation of the direction relative to north is very precise.

We can then use this information to navigate along a path of waypoints.

A trouble only occurs when the aircraft slows down below a minimum speed and stops at a point.

The solution I propose is to consider that the drone is still oriented as defined by the last GPS indication before the speed drops below the threshold.

This way if the aircraft has to stop and remain stationary, for example, it will not rotate on itself.

When a waypoint is reached, the last heading provided by the GPS before the horizontal speed drops below the threshold is used to calculate the new route to the next one. 

This mode can be set by selecting the Virtaul Magnetometer from the list of possible magnetometer devices.

## Who does this impact? Who is this for?
Every magnetomeeter users

dingrgh, 2022-05-06:
iNav already does this. The magnetometer is optional. iNav derives the track from GPS and assumes heading is the same as the track. I have several fixed-wing craft without a magnetometer. RTH works just fine.

sdellava, 2022-05-06:
> I have several fixed-wing craft without a magnetometer. RTH works just fine.

Thanks for your feedback. I only use INAV on copters so maybe the behaviour is different. 
Specifically the HOLD POSITION nav mode is not available without magnetometer and it's dones't work well if the magnetometer is not perfectly calibrated.

Anyway, which is the magnetometer hw you select in the configurator?

StuweFPV, 2022-05-06:
I agree with dingrgh - I don't have any compass/magnetometer on my planes and everything works just fine. My GPS modules don't even have one (I use mainly BN-220) and the hardware is set to "NONE" in the configurator.

rts18, 2022-05-06:
@sdellava The GPS course over ground heading is used to assist MC compass bearing.
Flying straight in a waypoint session, when the compass bearing is off by up to 20degrees from the GPS course. Reveals the multi-copter will face its bearing 20degrees from the target, while its forward motion will crab along sideways, as it advances towards the target.

Your idea is good. While I'm not sure it would be possible to calibrate the magnetometer perfectly, just from a Northern bearing on one axis. The compass requires the earths magnetic north to intersect all three coils sets. That's why the unit has to be rotated in the calibration phase.

sdellava, 2022-05-07:
@rts18 you are probably referring to my other feature request (the automatic calibration of the magnetometer). I am not sure if it is really necessary to have all magnetic axes perfectly calibrated. It might be better, but in reality, when the helicopter is moving, the course calculated by the GPS is enough to navigate and the gyroscopes/accelerometers provide the necessary information to keep the horizon.

The magnetometer is really necessary when the helicopter has to remain stationary on a point or otherwise fly very slowly. In this case the GPS may provide an incorrect course but the helicopter is almost flat, so in this condition we only need the course provided by the magnetometer on the x-axis or simply keep the last good GPS course value and calculate the actual course considering the rotation on the YAW provided by the gyro/accelerometer.

rts18, 2022-05-07:
There is an open pull request for such a feature. #6206. It is still a draft.
Such a feature requires a considerable amount of rework to the navigation code. This may be why its never been finished.

Things to be aware of are. The magnetic bearing on the Y axis needs to be correct when the copter is slowing down, while still having forward tilt..  The navigation code would need better data filtering for conflicts.
INAV's navigation code still has problems with accelerometer drift.  This can mess up the way the INS calculates course heading, if IMU, GPS and Compass are not in total agreement.

Bottom line. The idea and concept is good. But it would require a lot of work and testing to accomplish.

sdellava, 2022-05-07:
> Bottom line. The idea and concept is good. But it would require a lot of work and testing to accomplish.

@rts18 I understand your point, but on the other hand if the magnetometer had been a reliable aerial navigation tool, there would have been no need to invent GPS.

b14ckyy, 2024-03-20:
we have that now https://github.com/iNavFlight/inav/pull/9387

closed 2024-03-20 by b14ckyy