iNavFlight/inav #8010: Blackbox data selection

sdellava, 2022-05-07:
## Current Behavior
Currently blackbox collect all data provided by the FC. This lead to quicky fill the blackbox memory, even selecting  low logging rate.
Some fligth data, like i.e. the gryo/accelerometer, are required to elaborate video recorded with an oboard camera missing of gyro/acc.

Due to the huge ammount of data collecte by default, the collection last no more than few minutes, making it impossible to elaborate a long video footage.

Selecting only some data allow to extend the usable video length.

## Desired Behavior
Filter which data to collect into the BC memory, according with the configuration.

## Suggested Solution
Add a multichose menu in the configurator to select whick kind of data heve to be collected.
Filter the data collection according the configuration.

## Who does this impact? Who is this for?
Everyone need tha gyro/acc. data only.

stronnag, 2022-05-07:
https://github.com/iNavFlight/inav/pull/7378

DzikuVx, 2022-05-07:
Also see Blackbox improvements section of INAV 4.0 release notes

closed 2022-05-07 by DzikuVx

sdellava, 2022-05-07:
Thanks @DzikuVx. I totally missed it.