iNavFlight/inav #8017: INAV 4.1.0 RTH shoots up and falls down

akurekin, 2022-05-08:
Hi, I experience a very wrong behaviour of INAV in RTH mode. 
When RTH mode switches ON the copter shoots up at full power to the RTH height f 30 meters, then stops and starts returning home. At home position it starts the descent down at high speed to the 10 meter height (the hover height), goes past this height for a couple of meters and then climbs up at full power again and then descends down. I nearly crashed my drone at these manoeuvrers. What is going on?  Is INAV 4.1.0 really so bad?

In POS.HOLD the drone is stable and holds the height. It climbs up at proper speed, no complains, but it goes down too fast.

My hardware: MATEK 405SE control board.

Please help!

**PLEASE MAKE SURE YOU READ AND UNDERSTAND THE SOCIAL MEDIA SUPPORT CHANNELS. QUESTIONS ABOUT FLASHING, CONFIGURING, PILOTING MAY BE CLOSED WITHOUT FURTHER INTERACTION.**

* [Telegram channel](https://t.me/INAVFlight)
* [Facebook group](https://www.facebook.com/groups/INAVOfficial)
* [RC Groups thread](https://www.rcgroups.com/forums/showthread.php?2495732-Cleanflight-iNav-%28navigation-rewrite%29-project)

**Please double-check that nobody reported the issue before by using search in this bug tracker.**

**PLEASE DELETE THE TEXT ABOVE AFTER READING AND UNDERSTANDING IT**

****

## Current Behavior

## Steps to Reproduce

1.
2.
3.
4.

## Expected behavior

## Suggested solution(s)

## Additional context

---

- FC Board name and vendor:
- INAV version string:

DzikuVx, 2022-05-08:
Blackbox log required, in the mean time, watch this: https://www.youtube.com/watch?v=aE5oVbBtlig

akurekin, 2022-05-10:
Thanks for the video with explanation. It looks like I could, at least partly, solve the problem by:
- downgrading INAV to v3.0.1
- increasing the I2C clock rate and also the board clock rate to max values.
- shielding the barometer with foam

akurekin, 2022-05-10:
I'll bring the logs from my next flight

DzikuVx, 2022-05-11:
* 4.1 and 3.0.1 have the same RTH procedure, so I'm not sure what downgrade would change
* I2C clock would change nothing
* foam on baro might help

closed 2022-05-11 by DzikuVx

akurekin, 2022-05-15:
HI, the log file enclosed.  I moved to INAV v3.0.1 but it did not help to solve the problem
As the issue has not been close, may I ask to reopen the issue?
Many thanks.

[]([LOG00053.TXT](https://github.com/iNavFlight/inav/files/8695535/LOG00053.TXT))

akurekin, 2022-05-15:
[LOG00053.TXT](https://github.com/iNavFlight/inav/files/8695548/LOG00053.TXT)

akurekin, 2022-05-15:
I tried to add foam to the barometer but it did not help. 
A couple of times I succeeded to start RTH but was not successful next time.
The success in RTH was random.

akurekin, 2022-05-15:
I also tried to put a controller on foam pads to dampen the vibrations from motors but it did not help. I am using the motors from Mavic with folding blades and when the motors start they may vibrate until the blades will unfold completely.