iNavFlight/inav #8018: INAV Break mode Fly Away

akurekin, 2022-05-08:
## Current Behavior
Fly away issue in break mode. 
It seems to work at slow speed and no wind condition. 
At light wind, when I accelerate left and release Roll stick, it flies away at max roll angle. 
At smaller speed the copter breaks with rubber band effect. It passes through the break point while returning back, then changes its direction to the opposite, retrns back and misses the break point again. What is going on?
 
## Steps to Reproduce
1. Switch to position hold mode
2. Roll stick left
3. Release Roll stick
4. Watch the copter flies away at max speed.

## Expected behavior

The copter stops at break point the the sticks is returned to the central position.

---

# version
# INAV/MATEKF405SE 4.1.0 Feb 11 2022 / 10:52:32 (611d26f4)
# GCC-10.2.1 20201103 (release)

- FC Board name and vendor: MATEK 405SE
- INAV version string: 4.1.0

rts18, 2022-05-09:
There have been many reports of this bug. Its known about. But there seems to be problems resolving it.
The copter won't completely runaway. The effect is seen worse if you release the stick quickly.
My advice is not to use it !

akurekin, 2022-05-10:
Hopefully this may help someone. I solve the problem by increasing the I2C clock rate and also the board clock rate to max values.

akurekin, 2022-05-10:
Also I downgraded the INAV from v4.1.0 to v3.0.1 and the break mode now works very well.

closed 2022-05-23 by akurekin