iNavFlight/inav #8034: OSD is not centered in the camera image

MartinHugh, 2022-05-12:
## Current Behavior
Following a maiden of a wing with iNav 4.01, and carrying out auto level and autotune, the aircraft flew well in all modes configured (Horizon/Acro/Manual).  Checking for flying level using LOS the aircraft looked fine, but through FPV, the AHI was well below the horizon.  I initally suspected that this could be down to camera tilt, but I only have a tiny amount set normally. This could be a contributing factor to the exaggerated downward angle, but the OSD is not vertically centred anyway. 

When reviewing frames from the FPV recording, and expecting the vertical centre of the OSD display to be the vertical centre of the image, I found it was about 55-60% of the way down.  

## Steps to Reproduce
1.  Fly and observe or record FPV and judge where the vertical centre of the OSD is in relation to the camera image

## Expected behavior
The OSD should be vertically and horizontally centred.  If the aircraft is flying level, the camera is straight and the OSD is centred, the levelled AHI and aircraft symbol should be close to the real horizon rather than pointing down

## Suggested solution(s)
Is it possible to vertically centre the OSD?
Are there any CLI commands I can enter to adjust its offsets ?

## Additional context
Using HDZero Whoop VTX with latest firmware and HDZero Nano Micro V1 camera
---

[image]

INAV/MATEKF405SE 4.1.0 Feb 11 2022 / 10:52:32 (https://github.com/iNavFlight/inav/commit/611d26f4696fe1792516ef585a1c5603404ebeab)
GCC-10.2.1 20201103 (release)

Jetrell, 2022-05-13:
The analog character OSD has a similar offset, which makes sense, because of it even number of character positions. Maybe the HDzero rendering is based on it.
There are some AHI offset commands, that work on the analog Character and Pixel OSD.. I haven't tried them on HDzero. But they may work.
`osd_ahi_vertical_offset` 
`osd_horizon_offset`
`osd_ahi_height`
`osd_sidebar_horizontal_offset`

MartinHugh, 2022-05-13:
@Jetrell thanks for your post.
I have tried changing the following

osd_horizon_offset
to various values between -2 and +2
No change

osd_row_shiftdown
to 0 and 1
No change

osd_ahi_vertical_offset
to +40,-40,-18
No change

obviously in each case I have confirmed the value set properly by calling "get" on the parameter and then running "save" to save and reboot the FC.

Nothing is moving in the display.

Am I missing something?
Could it be that the HDZero generation of OSD is not subject to these parameters?

geoffsim, 2022-05-13:
I have not done anything special with the HDZero canvas mode. It should be the same as analogue, just with a larger canvas.

MartinHugh, 2022-05-13:
@geoffsim can you move the OSD within the image using the commands I have tried.  It is possible there is something I am missing here. Thanks

geoffsim, 2022-05-14:
"osd_ahi_vertical_offset"
"osd_ahi_height"

These are pixel OSD only.

geoffsim, 2022-05-14:
========
osd_horizon_offset 
"To vertically adjust the whole OSD and AHI and scrolling bars"

I have tried this and it does work. +/- 2 lines from the center.
Note that the configurator does not show this offset when set.

geoffsim, 2022-05-14:
"osd_sidebar_horizontal_offset"
"Sidebar horizontal offset from default position. Positive values move the sidebars closer to the edges."

I have tested this and the sidebars move as expected.

geoffsim, 2022-05-14:
Current iNav fixes the width and height of the AHI to 9 rows by 11 chars.
The location is also fixed to the center of the display size, adjustable only by the "osd_horizon_offset" parameter. Unfortunately, this is locked to +/-2.

When I implemented "HD" mode, I didn't want to change any of the AHI code as it would have just complicated things. I thought that this area could be re-visited at a later date.

MartinHugh, 2022-05-14:
> 

> "osd_horizon_offset" "To vertically adjust the whole OSD and AHI and scrolling bars"
> 
> I have tried this and it does work. +/- 2 lines from the center. Note that the configurator does not show this offset when set.

@geoffsim thank you for trying these.  I tried this one in particular and got no change on the actual OSD.
So there is some hope then that I am doing something wrong.  

I did not repower the FC, I assumed the reboot after saving the CLI setting would have been enough, but no change to the display.

I will try again.  If you can see anything obvious I may have done wrong I would appreciate a steer.

MartinHugh, 2022-05-14:
> Current iNav locks the width and height of the AHI to 9 rows by 11 chars. The location is also fixed to the center of the display size, adjustable only by the "osd_horizon_offset" parameter. Unfortunately, this is locked to +/-2.
> 
> When I implemented "HD" mode, I didn't want to change any of the AHI code as it would have just complicated things. I thought that this area could be re-visited at a later date.

@geoffsim , I realise you are stating that the dimensions of the AHI above, but in relation to this I note in info from HDZero, they reserve space for the VTX OSD at the top (not sure if they do the same at the bottom) and this might explain why there is a gap at the top before the first iNav OSD row.

The area that HDZero render is 50x18 chars (1200px * 648px)

The glaring problem there, which was alluded to by @Jetrell in an earlier post is that if there are an even number of rows, how can you have a centre one.

geoffsim, 2022-05-14:
The VRX retains one line at the top and one at the bottom. The 18 lines for the OSD should be central. Try it by placing widgets at the extreme edges of the OSD. Note there is also a margin left and right.

MartinHugh, 2022-05-14:
> The VRX retains one line at the top and one at the bottom. The 18 lines for the OSD should be central. Try it by placing widgets at the extreme edges of the OSD. Note there is also a margin left and right.

@geoffsim that sounds right.  The 18 rows should be central, the point I was making was that with an even number of rows there is no way any particular row (like the centre of the AHI) can ever be vertically central anyway.  

I hope to retry the osd_horizon_offset very soon and will report back
Thank you.

MartinHugh, 2022-05-14:
set osd_horizon_offset=-2
save

Does indeed move the AHI up 2 chars.
Whether we need scope for more movement is another topic.

Sorry for the finger trouble on this.  I think perhaps I had misunderstood and was expecting the whole osd to move and the bottom row stayed put.  But in fact this does what I need.

Thank you for your help with this @geoffsim and thank you for the work on the HDZero OSD feature itself.

closed 2022-05-14 by MartinHugh