iNavFlight/inav #8044: Support JHEMCU f4 NOXE V3

MoesMo, 2022-05-21:
Hello, maybe I missed something. I would like to see support for this flight controller. At the moment I have a suitable firmware from Betaflight but it is for a quadcopter. And I'm building a plane .FLYWOOF411 is not suitable, FC does not see the gyroscope and accelerometer

MoesMo, 2022-05-21:
https://disk.yandex.ru/d/xRLHnBgvrBbzSg

MoesMo, 2022-05-24:
I tried to sort out the ports myself, it turned out that they match. That is, everything works but there is no GYRO and ACC. When you turn it on in betafly, the gyroscope does not work until the settings that betafly offers are applied. I attached Dumps on both sides. also a screen of the frequency of the Gyroscope gyro 3.2 Khz
https://disk.yandex.ru/d/Wm_fkzJYYNyzUQ

MoesMo, 2022-05-30:
I'm talking about my attempts on this flight controller. Reassigning board contacts did not help, in flywhoof411 (inav there are no drivers for this acc \ gyro, I tried to transfer the drivers from betaflight - they did not appear even in the acc selection. I did not have enough experience in this matter
gyro presumably LSM6

trogalko, 2022-07-08:
Just bought this one, Noxe F4 V3, its using different imu than V2, V3 uses BMI270, I also hope this board supported by Inav.
Thank You

trogalko, 2022-07-10:
I copied some code from target.h from MatekF411TE to flywoof411 and it now detect gyro, acc, and baro (not flying yet, just flashed the Noxe V3 board). Here is the modified FlywooF411V2 target.h (I use V2 as its design closed to V3) :
`/*
 * This file is part of Cleanflight.
 *
 * Cleanflight is free software: you can redistribute it and/or modify
 * it under the terms of the GNU General Public License as published by
 * the Free Software Foundation, either version 3 of the License, or
 * (at your option) any later version.
 *
 * Cleanflight is distributed in the hope that it will be useful,
 * but WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 * GNU General Public License for more details.
 *
 * You should have received a copy of the GNU General Public License
 * along with Cleanflight.  If not, see <http://www.gnu.org/licenses/>.
 */
//wakata
#pragma once

#ifdef FLYWOOF411_V2
#define TARGET_BOARD_IDENTIFIER "FW42"
#define USBD_PRODUCT_STRING     "FLYWOOF411V2"
#else
#define TARGET_BOARD_IDENTIFIER "FW41"
#define USBD_PRODUCT_STRING     "FLYWOOF411"
#endif

#define LED0                    PC13

#define BEEPER                  PC14
#define BEEPER_INVERTED

// *************** SPI **********************
#define USE_SPI
#define USE_SPI_DEVICE_1
#define SPI1_SCK_PIN           PA5
#define SPI1_MISO_PIN   	   PA6
#define SPI1_MOSI_PIN   	   PA7

#define USE_SPI_DEVICE_2
#define SPI2_SCK_PIN           PB13
#define SPI2_MISO_PIN  		   PB14
#define SPI2_MOSI_PIN  		   PB15

// *************** SPI Gyro & ACC **********************

#define USE_EXTI
#ifdef FLYWOOF411_V2
#define USE_IMU_BMI270
#define BMI270_SPI_BUS          BUS_SPI1
#define BMI270_CS_PIN           PA4
#define IMU_BMI270_ALIGN        CW270_DEG_FLIP
#define USE_MPU_DATA_READY_SIGNAL
#else
#define USE_IMU_MPU6000
#define MPU6000_CS_PIN          PA4
#define MPU6000_SPI_BUS         BUS_SPI1
#define IMU_MPU6000_ALIGN       CW180_DEG

#define USE_IMU_ICM20689
#define ICM20689_CS_PIN         PA4
#define ICM20689_SPI_BUS        BUS_SPI1
#define IMU_ICM20689_ALIGN      CW180_DEG
#endif

#ifdef FLYWOOF411_V2
#define GYRO_INT_EXTI           PB5
#define BMI270_EXTI_PIN         GYRO_INT_EXTI
#else
#define GYRO_INT_EXTI           PB3
#endif
#define USE_MPU_DATA_READY_SIGNAL

// *************** Baro *****************************

#define USE_I2C
#define USE_I2C_DEVICE_1
#define I2C1_SCL                PB8
#define I2C1_SDA                PB9

#define USE_BARO
#define BARO_I2C_BUS		    BUS_I2C1
#define USE_BARO_BMP280
#define USE_BARO_MS5611
#define USE_BARO_SPL06

#define USE_MAG
#define MAG_I2C_BUS             BUS_I2C1
#define USE_MAG_HMC5883
#define USE_MAG_QMC5883
#define USE_MAG_IST8310
#define USE_MAG_IST8308
#define USE_MAG_MAG3110
#define USE_MAG_LIS3MDL
#define USE_MAG_AK8975

// *************** SPI OSD *****************************
#define USE_MAX7456
#define MAX7456_SPI_BUS         BUS_SPI2
#define MAX7456_CS_PIN          PB12

// *************** SPI FLASH **************************
#define USE_FLASHFS
#define USE_FLASH_M25P16
#define M25P16_CS_PIN           PB2
#define M25P16_SPI_BUS          BUS_SPI2
#define ENABLE_BLACKBOX_LOGGING_ON_SPIFLASH_BY_DEFAULT

// *************** UART *****************************
#define USE_VCP

#define USE_UART1
#ifdef FLYWOOF411_V2
#define UART1_TX_PIN            PA9
#define UART1_RX_PIN            PA10
#else
#define UART1_TX_PIN            PB6
#define UART1_RX_PIN            PB7
#endif

#define USE_UART2
#ifdef FLYWOOF411_V2
#define UART2_TX_PIN            PA2
#else
#define UART2_TX_PIN            NONE    //PA2
#endif
#define UART2_RX_PIN            PA3

#define USE_SOFTSERIAL1
#ifdef FLYWOOF411_V2
#define SOFTSERIAL_1_TX_PIN     PB6     // Clash with TX2, possible to use as S.Port or VTX control
#define SOFTSERIAL_1_RX_PIN     PB7
#else
#define SOFTSERIAL_1_TX_PIN     PA2     // Clash with TX2, possible to use as S.Port or VTX control
#define SOFTSERIAL_1_RX_PIN     PA2
#endif

#define SERIAL_PORT_COUNT       4       // VCP, USART1, USART2, SS1

#define DEFAULT_RX_TYPE         RX_TYPE_SERIAL
#define SERIALRX_PROVIDER       SERIALRX_SBUS
#ifdef FLYWOOF411_V2
#define SERIALRX_UART           SERIAL_PORT_USART1
#else
#define SERIALRX_UART           SERIAL_PORT_USART2
#endif
// *************** ADC *****************************
#define USE_ADC
#define ADC_INSTANCE                    ADC1
#define ADC_CHANNEL_1_PIN               PA1
#ifdef FLYWOOF411_V2
#define ADC_CHANNEL_2_PIN               PB1
#define ADC_CHANNEL_3_PIN               PB0
#else
#define ADC_CHANNEL_2_PIN               PA0
#define ADC_CHANNEL_3_PIN               PB1
#endif

#define CURRENT_METER_ADC_CHANNEL       ADC_CHN_1
#define VBAT_ADC_CHANNEL                ADC_CHN_2
#define RSSI_ADC_CHANNEL                ADC_CHN_3

// *************** LED2812 ************************
#define USE_LED_STRIP
#ifdef FLYWOOF411_V2
#define WS2811_PIN                      PA0
#else
#define WS2811_PIN                      PA15
#endif
// ***************  OTHERS *************************
#define DEFAULT_FEATURES                (FEATURE_TX_PROF_SEL | FEATURE_OSD | FEATURE_VBAT | FEATURE_TELEMETRY | FEATURE_SOFTSERIAL)

#define USE_DSHOT
#define USE_ESC_SENSOR
#define USE_SERIAL_4WAY_BLHELI_INTERFACE

#define TARGET_IO_PORTA         0xffff
#define TARGET_IO_PORTB         0xffff
#define TARGET_IO_PORTC         0xffff
#define TARGET_IO_PORTD        (BIT(2))

#ifdef FLYWOOF411_V2
#define MAX_PWM_OUTPUT_PORTS       6
#else
#define MAX_PWM_OUTPUT_PORTS       4
#endif`

trogalko, 2022-07-10:
Use Inav configurator developer mode to config the board with custom build firmware : [http://seyrsnys.myzen.co.uk/inav-configurator-next/](url)

MoesMo, 2022-07-13:
Hello, thanks for your help! Weird, I've added this gyro before and it didn't work for me. I'll try again next week. If you have a compiled firmware file for testing, then I would not refuse it for tests. If you don't mind of course

MoesMo, 2022-07-13:
Hello, thanks for your help! Weird, I've added this gyro before and it didn't work for me. I'll try again next week. If you have a compiled firmware file for testing, then I would not refuse it for tests. If you don't mind of course
@trogalko

trogalko, 2022-07-13:
Here is the firmware for Noxe V3 : https://drive.google.com/drive/folders/1ed0cnApQLq041-fcoeYrtr34jYF2IpzJ?usp=sharing    

And this is my crude modification, I'm build the firmwares using Github Action
https://github.com/trogalko/inav
Maybe You could request for pulling and merge if You had tested it working, while I must wait until Saturday or Sunday.

Geka-electron, 2022-07-17:
> Here is the firmware for Noxe V3 : https://drive.google.com/drive/folders/1ed0cnApQLq041-fcoeYrtr34jYF2IpzJ?usp=sharing
>
> And this is my crude modification, I'm build the firmwares using Github [...]

Hello. Thanks for you work! I was trying to test a code. But, unfortunately, don`t work SBUS on UART1 (and other). The gyro is ok!

Geka-electron, 2022-07-17:
> Here is the firmware for Noxe V3 : https://drive.google.com/drive/folders/1ed0cnApQLq041-fcoeYrtr34jYF2IpzJ?usp=sharing
>
> And this is my crude modification, I'm build the firmwares using Github [...]

Tested. Doesn`t work SBUS. No signals. I am use inav_6.0.0_FLYWOOF411_V3_ci-20220711-a1fe6f3.hex file. May be INAV 6.0 have some troubles?

trogalko, 2022-07-22:
Looks like JHEMCU has started provided the (Inav) firmware for this board, just the binary, no source code (or modiification) :
[https://github.com/jhemcu/FC-ESC-Firmware](https://github.com/jhemcu/FC-ESC-Firmware)

Geka-electron, 2022-07-22:
> Looks like JHEMCU has started provided the (Inav) firmware for this board, just the binary, no source code (or modiification) : https://github.com/jhemcu/FC-ESC-Firmware

Thanks! I will try today

Geka-electron, 2022-07-22:
> Looks like JHEMCU has started provided the (Inav) firmware for this board, just the binary, no source code (or modiification) : https://github.com/jhemcu/FC-ESC-Firmware

Thanks. I tasted today this binary: it working for drons, but not for airplane.

MoesMo, 2022-07-22:
> Похоже, что JHEMCU начал предоставлять прошивку (Inav) для этой платы, только двоичный код, без исходного кода (или модификации): [https://github.com/jhemcu/FC-ESC-Firmware .](https://github.com/jhemcu/FC-ESC-Firmware)

# set board
align_board_roll = -1800
Allowed range: -1800 - 3600

align_board_pitch = 0
Allowed range: -1800 - 3600

align_board_yaw = 0
Allowed range: -1800 - 3600

There are problems with gyro at the i2c bus frequency of 400-800 - it starts to freeze, everything is fine at 200, apparently for a reason other values ​​were on the beta. Also flip the board. When editing via cli, accelerometer calibration does not work correctly.the rest of the outputs have not yet been tested

Geka-electron, 2022-07-23:
Yes. The accel does`t calibrate. I tried to configure to flyingwing. The servs are working, but motor don`t have a signal

wawanlost, 2022-08-22:
Apakah tersedia di inav berikutnya.... bagaimana dengan softserial,

pavel0001, 2022-09-17:
Good day. Has anyone been able to set up this flight controller for the wing? The firmware from the jhemcu repo ( https://github.com/jhemcu/FC-ESC-Firmware/tree/main/F4%20NOXE%20V3#:~:text=inav_6.0.0_JHEMCUF411.rar )  is almost ok, but when I install 2 servos with 1 esc, esc doesn't work. And when I remove all servos and only esc remains, it works.

MoesMo, 2022-11-13:
Good night, I finally got to this controller. The meaning is the same on betaflight, when switching to the glider / airplane mode, only the motor works - the servo does not work. in drone mode, when checking all the motors at the outputs to which the servos are connected, everything works both the motor and the servos, but only until you select the glider or plane again.
I repeat on betaflight exactly the same problem
I built my firmware on which you do not need to flip anything, in fact, porting it from target.h target.c from beta. But still the same problem with the servo and motor

> Похоже, что JHEMCU начал предоставлять прошивку (Inav) для этой платы, только двоичный код, без исходного кода (или модификации): [https://github.com/jhemcu/FC-ESC-Firmware .](https://github.com/jhemcu/FC-ESC-Firmware)

p-fpv, 2022-11-13:
I helped with the firmware, but I don't have the controller itself to check properly.

[inav5.1.JHEF411.zip](https://github.com/iNavFlight/inav/files/9997827/inav5.1.JHEF411.zip)

s1 - servo
s2 - servo
s3 - servo
s4 - motor

softserial_tx - LED pin
softserial_rx - cam ctrl pin

Only the firmware in the archive from the 'firmware old' folder was checked. But it is without softserial.

MoesMo, 2022-11-13:
https://youtu.be/8VkhtbEG7fg    @p-fpv 
Pavel it works, though again not everything, but it's already cool!

Geka-electron, 2022-11-18:
It s working! One motor and two servs.

MartinGiersberg, 2022-11-20:
Two Servos + 1 ESC not working (Delta Wing on INAV) **[SOLVED] + Instructions**

Hellow Everyone. I ran into the Issue of the firmware not beeing able to run two Servos and one ESC at the same time. I've solved the Issue and here is how:

Issue:
Using the F4 NOXE V3 FC on a delta wing setup (Tiny Wing 450X) and the stock "Flying Wing" mixer setup on INAV the FC did not output any signal to the S1 channel (the designated motor/ESC channel). However when connecting the ESC to one of the servo channels the ESC worked as expected. This was not an issue with the ESC. The problem could be replicated using a wide rage of other ESCs. The FC indeed did not send out any signal on the designated ESC channel.

Solution:
Setup the INAV Mixer in a way that the ESC Channel is no longer a "Motor" channel but a "Servo" channel instead.

HowTo:
After watching [this video from INAV developer Paweł Spychalski](https://www.youtube.com/watch?v=zMc4iLz269Q) I typed the command `set output_mode=SERVOS` in CLI To override the default INAV presets and force all outputs to be servo outputs. Then I added a third Servo in the Mixer section of INAV and set it up to "RC Throttle". 
[image]

MoesMo, 2022-11-23:
@Geka-electron @trogalko @p-fpv @pavel0001  @MartinGiersberg 
Hello guys, I tested uart 1 uart 2 ibus, gps works, can any of you check on the firmware from @p-fpv Sbus, Crossfire. And I think @p-fpv can make a pull request. It looks like you managed to make a people's flight controller, given its cost, this is a gift. Many thanks to all. I got the plane

p-fpv, 2022-11-23:
@MoesMo I won't be able to make an extraction request due [NEW_HARDWARE_POLICY](https://github.com/iNavFlight/inav/blob/master/docs/policies/NEW_HARDWARE_POLICY.md)

[inav_6.0.0_JHEMCUF411NOXEv3.zip](https://github.com/iNavFlight/inav/files/10079456/inav_6.0.0_JHEMCUF411NOXEv3.zip) Not tested  
s1-3 - servo, s4 - motor  
softserial_tx - LED pin  
softserial_rx - cam ctrl pin

Geka-electron, 2022-11-24:
Pavel. Thanks a lot!

p-fpv, 2022-12-06:
@Geka-electron @MoesMo @pavel0001  @trogalko Fixed a bug in target.
[JHEMCUF411.zip](https://github.com/iNavFlight/inav/files/10162443/JHEMCUF411.zip) Not tested
@MartinGiersberg output_mode=SERVOS should work fine, but esc won't work with dshot.

Write if something doesn't work in the firmware, I can try to fix the error

mm3x, 2023-02-14:
Will the "output_mode=SERVOS" solution have issues with RTH, automatic landing, autolaunch or any other modes which require inav to manage motor speed?

Geka-electron, 2023-02-15:
I'm tested only autolaunch. It's worked

Viga7474, 2023-02-19:
> @MoesMoЯ не могу отправить запрос на участие из-за [NEW_HARDWARE_POLICY](https://github.com/iNavFlight/inav/blob/master/docs/policies/NEW_HARDWARE_POLICY.md)
> 
> [inav_6.0.0_JHEMCUF411NOXEv3.zip](https://github.com/iNavFlight/inav/files/10079456/inav_6.0.0_JHEMCUF411NOXEv3.zip) Не тестировался Upd [JHEMCUF411.zip](https://github.com/iNavFlight/inav/files/10162443/JHEMCUF411.zip) Исправлена ​​цель. Не испытано
> 
> s1-3 — сервопривод, s4 — мотор softserial_tx — контакт светодиода softserial_rx — контакт управления кулачком

Is it possible to use 4 motors for a quadrocopter?
Thanks.

odddollar, 2023-02-23:
I know an INAV 6.0 target has been provided by @p-fpv, but I'm assuming this is a beta release of incomplete firmware. Will there be an official target or a community compiled version when INAV 6.0 is fully released?

IvanBiv, 2023-03-16:
> ed

Can you made firmware for jhemcu F405-xsd?
And I want to know how can I self made this firmware.

odddollar, 2023-03-16:
> > ed
> 
> Can you made firmware for jhemcu F405-xsd?
> And I want to know how can I self made this firmware.

Open another issue on GitHub and ask for support there. I'm not familiar with how to custom compile firmware

p-fpv, 2023-03-16:
@Viga7474 yes, it should work on a quadcopter.
@odddollar for the official version, these conditions must be met [NEW_HARDWARE_POLICY](https://github.com/iNavFlight/inav/blob/master/docs/policies/NEW_HARDWARE_POLICY.md), maybe someone will make a pull request for this controller, adding it as an unofficial target.

[JHEMCUF411NOXEv3_INAV6RC3.zip](https://github.com/iNavFlight/inav/files/10995136/JHEMCUF411NOXEv3_INAV6RC3.zip) updated to inav 6 rc3

p-fpv, 2023-03-16:
@IvanBiv try this firmware [Test JHEMCUF405xsd_INAV6RC3.zip](https://github.com/iNavFlight/inav/files/10995296/JHEMCUF405xsd_INAV6RC3.zip)

There is no use from files from the archive, the normal byteflight firmware for jhemcu F405-xsd should be this [F405-XSD](https://github.com/jhemcu/FC-ESC-Firmware/tree/main/F405-XSD)
It may be useful to output the status command to determine the accelerometer if it is running in betaflight

odddollar, 2023-03-16:
> @Viga7474 yes, it should work on a quadcopter.
> @odddollar for the official version, these conditions must be met [...]

You absolute legend! Thanks a ton.

Is there any documentation on how to compile Inav for custom targets, like you've done here? I'd love to be able to compile a target for this flight controller when Inav 6 exits release candidate phase.

IvanBiv, 2023-03-17:
> @IvanBiv try this firmware [Test JHEMCUF405xsd_INAV6RC3.zip](https://github.com/iNavFlight/inav/files/10995296/JHEMCUF405xsd_INAV6RC3.zip)
> 
> There is no use from files from the archive, the normal betaflight firmware for jhemcu F405-xsd should be this [F405-XSD](https://github.com/jhemcu/FC-ESC-Firmware/tree/main/F405-XSD) It may be useful to output the status command to determine the accelerometer if it is running in betaflight

Thx! I will try soon.

Viga7474, 2023-03-17:
> @Viga7474да, это должно работать на квадрокоптер. 

Could you tell me how to reassign the servo to the motors. In your version, unlike the "official" one, there is a softserial, but unfortunately I don't know how to adapt it to the copter.

MoesMo, 2023-03-17:
@Viga7474 
Где-то выше трогалко предоставлял свою модификацию под дрон https://github.com/iNavFlight/inav/issues/8044#issuecomment-1182642601
только это inav 6.0
 Я пробовал она работает ,не могу сказать за softserial

p-fpv, 2023-03-17:
> Is there any documentation on how to compile Inav for custom targets, like you've done here?  

@odddollar Documentation is here [development](https://github.com/iNavFlight/inav/tree/master/docs/development). I am building in debian stable according to this instruction [Building in Linux](https://github.com/iNavFlight/inav/blob/master/docs/development/Building%20in%20Linux.md).
The directory with the files for the assembly must be left in [target](https://github.com/iNavFlight/inav/tree/master/src/main/target).
Bulding the firmware ```make JHEMCUF411NOXEv3```

>  I don't know how to adapt it to the copter.

@Viga7474 настройка как у любого другого контроллера inav, если выбрать коптер при первом включении, то моторы уже будут на s1-4. Это так же можно изменить на вкладке Mixer

MoesMo, 2023-03-17:
@p-fpv что-то поменял в UPD просто я на старой
 ( понял подвёз softserial на контакты и теперь это inav 6)

Так на 405 плату вроде есть прошивка под Jhemcu

> > @IvanBivпопробуй эту прошивку [Test JHEMCUF405xsd_INAV6RC3.zip](https://github.com/iNavFlight/inav/files/10995296/JHEMCUF405xsd_INAV6RC3.zip)
> > От файлов из архива толку нет, нормальная прошивка betaflight для jhemcu F405-xsd должна быть этой [F405-XSD](https://github.com/jhemcu/FC-ESC-Firmware/tree/main/F405-XSD) Может быть полезно вывести команду status для определения акселерометра, если он запущен в betaflight
> 
> Спасибо! Я попробую в ближайшее время.

p-fpv, 2023-03-17:
@MoesMo да, обновил до inav 6 rc3.
А вот для F405-xsd только betaflight нашёл, оставь ссылку, если вдруг готовый inav попадётся.

KenImhof, 2023-03-21:
@p-fpv,  when flashing inav_6.0.0_JHEMCUF411NOXEv3.zip with configurator 6.0.0 RC4 and selecting plane with tail, after the accelerometer is calibrated with FC save and reboot, it still shows accelerometer as not calibrated.  If I flash and select flying wing, the calibration works.  I can then change the mixer to plane with tail and the calibration continues to work.

p-fpv, 2023-03-21:
> @p-fpv, when flashing inav_6.0.0_JHEMCUF411NOXEv3.zip with configurator 6.0.0 RC4 and selecting plane with tail, after the accelerometer is calibrated with FC save and reboot, it still shows [...]

@KenImhof, I don't have a solution to this problem, as there is no this FC to check the firmware. Try the firmware from the archive JHEMCUF411NOXEv3_INAV6RC3.zip with the 6.0.0 RC4 configurator.. If everything works, then I will edit my comments and delete the old archives so that there is no confusion.

KenImhof, 2023-03-21:
@p-fpv, the calibration for plane with a tail is working with this firmware.
[image]

p-fpv, 2023-03-21:
@KenImhof, the same target was used in this firmware.
I reinstalled inav, maybe the problem is in it.
Try this file [nav 6 RC3.zip](https://github.com/iNavFlight/inav/files/11032102/inav.6.RC3.zip)

KenImhof, 2023-03-22:
@p-fpv, with this version the calibration data is retained and working; however, after the up-right calibration and completing the inverted calibration, the up-right calibration grays and has to be redone.  I also verified in plane mode that all three stabilized  servo outputs are working and the motor output is working from the configurator, but the motor output is not working from the transmitter.  After arming is enabled, the blue light just flashes on the flight controller and  the cli status shows the "CLI" flag even though I have enabled the output from the configurator.  The blue light also flashes when arming and disconnected from the configurator.  

Thanks for all your good work!!!!

MoesMo, 2023-03-22:
@p-fpv 
"Очень хорошая, но не официальная поддержка полета inav и beta в прошивке на заказ, не работают порты UART" )))
https://aliexpress.ru/item/1005004161082326.html?sku_id=12000028236272765&spm=a2g2w.productlist.search_results.0.2d644aa6kRRh35
На заказ говорят работаешь ).
Суть следующая, пытался выйти на связь с производителем,для того чтобы они предоствили два полётника для полноценного включения цели . Но видимо им это не особо интересно ,просто игнорируют .

p-fpv, 2023-03-22:
@MoesMo, если бы на заказ, так-то ничего кроме опыта по ковырянию прошивки больше не получил. А сейчас есть вроде как рабочий target, и собрать его из исходников не так и долго.
Думаю быстрее самому  отправить разработчикам, чем дождаться движений от производителя.

Viga7474, 2023-03-23:
> @MoesMo, если бы на заказ, так-то ничего кроме опыта по ковырянию прошивки больше не получил. А сейчас есть вроде как рабочий target, и собрать его из исходников не так и долго. Думаю быстрее самому отправить разработчикам, чем дождаться движений от производителя.

У меня на этом полетнике быстро сдох по непонятной причине барометр( после арма показания ползут вверх) и в "официальной" прошивке softserial похоже отключен, хотя я новичок,возможно не разобрался.

DzikuVx, 2023-03-23:
@MoesMo @Viga7474 @p-fpv please switch to English or I will be forced to lock conversation over here

DzikuVx, 2023-03-23:
The rules are the same for every target and described here https://github.com/iNavFlight/inav/blob/master/docs/policies/NEW_HARDWARE_POLICY.md

odddollar, 2023-03-26:
Now that INAV 6 is out on full release, I have managed to compile it for this target using instructions from @p-fpv 

[JHEMCUF411NOXEv3_INAV6.zip](https://github.com/iNavFlight/inav/files/11071325/JHEMCUF411NOXEv3_INAV6.zip)

Viga7474, 2023-03-26:
> Теперь, когда INAV 6 выпущен в полной версии, мне удалось скомпилировать его для этой цели, используя инструкции из@p-fpv
> 
> [JHEMCUF411NOXEv3_INAV6.zip](https://github.com/iNavFlight/inav/files/11071325/JHEMCUF411NOXEv3_INAV6.zip)

Have you tried this on a copter? I have only 1 and 4 motors running.

Viga7474, 2023-03-26:
It is possible that the problem is in these parameters, but I could not change them through the interface and the CLI. Maybe I'm doing something wrong.

# resources

# Mixer: motor mixer

mmix reset

mmix 0  1.000 -1.000  1.000 -1.000
mmix 1  1.000 -1.000 -1.000  1.000
mmix 2  1.000  1.000  1.000  1.000
mmix 3  1.000  1.000 -1.000 -1.000

# Mixer: servo mixer

# Outputs [servo]

odddollar, 2023-03-26:
> > Теперь, когда INAV 6 выпущен в полной версии, мне удалось скомпилировать его для этой цели, используя инструкции из@p-fpv
> > 
> > [JHEMCUF411NOXEv3_INAV6.zip](https://github.com/iNavFlight/inav/files/11071325/JHEMCUF411NOXEv3_INAV6.zip)
> 
> Have you tried this on a copter? I have only 1 and 4 motors running.

This is for a fixed wing. The target files were provided by @p-fpv, I just compiled it. It appears as though the outputs are fixed as one motor and 3 servos

odddollar, 2023-03-27:
I've just checked the target files, and it appears that all s1, s2, s3 and s4 should all be configured at motor outputs when a multicopter mixer is selected. So this target should also work for quad copters.

> It is possible that the problem is in these parameters, but I could not change them through the interface and the CLI. Maybe I'm doing something wrong.
> # resources
> # Mixer: motor mixer
>
> mmix reset [...]

Check if you've got an airplane or quad mixer selected.

odddollar, 2023-03-27:
I checked the target and all the outputs are set as motor outputs for a quadcopter. I don't know what the problem is then. I only have this flight controller in a plane and it works fine.

Viga7474, 2023-03-27:
> I checked the target and all the outputs are set as motor outputs for a quadcopter. I don't know what the problem is then. I only have this flight controller in a plane and it works fine.

Thanks for the reply. So I'll wait, maybe someone will test it on a copter or adapt the firmware for a copter.

nbertram, 2023-03-28:
I built this for the 6.0.0 stable and it appears to work properly, though I had an error "Not enough servo outputs/timers" in the CLI `status` that was preventing any PWM output from the servos.

Turned out the plane present had configured four servo outputs (in addition to the motor), so deleting the fourth from the mixer and configuring the others correctly made the error go away and I had output.

Just thought I'd put that here for anyone else trying on 6.0.0 and having an out of the box issue.

I'm also interested in having an actual fourth servo output so I can have separate aileron servos, and thought I should be able to repurpose the `PA1` current sense pin on to `TIM1` to achieve it? I'll report back if it works...

[23 more replies]

closed 2023-10-27 by DzikuVx