iNavFlight/inav #8046: RTH unexpected behaviour

akurekin, 2022-05-23:
## Current Behavior

Dangerous behaviour of quad when switching on RTH mode. I was flying at slow speed in Angle mode and when I switched RTH mode my quad started turning back to me, but it moved too much and the turning angle was close to 180 degrees. It actually turned upside down and then recovered. This accident happened multiple times at random. I enclosed the LOG file that shows this happening at time 0.4.31.171

## Steps to Reproduce

1. Fly the drone in Angle mode to the distance when failsafe gets activated and INAV switches to RTH mode. 
2. Observe the drone turning home
3. Seve your drone from falling upside down by compensating its turns manually

## Expected behavior

After switching to RTH mode the drone should smoothly start moving home

## Suggested solution(s)

Testing if Ardupilot has the same behaviour

- FC Board name and vendor: MATEK405SE
- INAV version string: 3.0.1
- 
[LOG00091.TXT](https://github.com/iNavFlight/inav/files/8751755/LOG00091.TXT)

breadoven, 2022-05-24:
Bit strange, the normal RTH worked fine but the Failsafe RTH sent full Pitch and Roll commands initially for some reason. Was this switched Failsafe mode or a real Failsafe due to signal loss ? How did you recover it with the sticks if it was in Failsafe ?

akurekin, 2022-05-24:
It was a real railsafe, you can check RSSI graph, RSSI=0 for short time interval and after that I could control the drone. However, the same behaviour happened  several times when I activated RTH manually.

closed 2022-06-06 by akurekin