iNavFlight/inav #8049: Lumenier H7 LUX beeper sound weird

ArnoTlse, 2022-05-23:
## Current Behavior
I have a Vifly Finder 2 connected to the GND, 5V, BUZ pad of a Lumenier H7 lux Flight controller flashed with MATEK H743 firmware.
I have noticed that the beeper makes a weird noise (low and clicking & metallic noise) for all Flight controller beeping tones. When I disconnect the Lipo from the quad, the Beeper, entering into its mode without lipos, sounds well, meaning that the Vifly beeper seems not be damaged. The beepper works well also on a MATEK F405 FC.
Then, I wonder if there is no issue around the mapping of this Buzzer pad and a Timer of the FC that could be in conflict with Motor output.
As anyone experiment this also ? 
How can I find the mapping of resources vs Pin of the FC ?

## Steps to Reproduce

1. Connect a lipo to the Quad
2. Beeper from motors plays well but beeper sound from Vifly Buzzer sound weird
3.
4.

## Expected behavior

Vifly finder 2 sounding well 

## Suggested solution(s)

## Additional context

---

- FC Board name and vendor: Lumenier H7 LUX
- INAV version string:  
-- INAV/MATEKH743 4.0.0 Dec 15 2021 / 15:18:48 (c20a9852) 
-- GCC-10.2.1 20201103 (release)

DzikuVx, 2022-05-24:
Try this one https://github.com/iNavFlight/inav/blob/master/docs/Settings.md#beeper_pwm_mode

closed 2022-05-24 by DzikuVx

ArnoTlse, 2022-05-24:
Hi Pawel,

Thanks a lot. It works fine now.
I will check what I have as beeper_pwm_mode value in the F405. So for the matek h743, I understand that the beeper pwm mode is set to On by default. By disabling it for the lumenier H7, it works fine.