iNavFlight/inav #8055: Frozen OSD/FC during RTH

StuweFPV, 2022-05-27:
Today i was testing some antenna setup with my test plane and did about 10 flights. Now and then I test RTH in order to fine tune the planes behavior and to make sure it will work. so today at a number of times. On flight 10 however the plane would not come home as usual. When i've noticed and wanted to abort RTH the whole FC went into a freeze! unable to regain control as i've watched the plane crash...

RTH was initiated by me via switch as a test. full RC cover was given at any time. 100%LQ at 10mW..

    This could be a hardware failure. However i do take my soldering tasks very seriously, always put capacitors on and have done more than 10h with this plane in over 50 flights. In all these years this has never happened to me. So just a hardware failure or really a bug in 4.1???

Strangely enough the Crossfire was still transmitting the GPS coordinates even after the crash (for once the battery stayed connected) so i was able to locate the plane exactly. However it sits 70ft high in a tree so that recovery is impossible. Hence no log files available. But this shows that it was not a complete freeze of iNAV/FC since Crossfire does not transmit coordinates without the help of the FC and the GPS antenna in my setup. Logically the GPS coordinates should be the ones at the moment of the freeze but they are not.

    https://youtu.be/L73tAnIXxLk

## Steps to Reproduce
1. Probably impossible to reproduce

## Expected behavior
1. Fly home when RTH is active - but it kept turning
2. Don't freeze but let me abort RTH anytime

## Suggested solution(s)
check the code for old/hidden leftovers, possible conflicts, bugs

## Additional context
Diff file of the current flight is attached

---

- FC Board name and vendor: Matek F722-Wing + Matek Analog pixel OSD module (via UART2)
# INAV/MATEKF722SE 4.1.0 Feb 11 2022 / 11:08:58 (611d26f4)
# GCC-10.2.1 20201103 (release)

[T-28 iNAV 4.1 v3.1.txt](https://github.com/iNavFlight/inav/files/8786010/T-28.iNAV.4.1.v3.1.txt)

breadoven, 2022-05-27:
Looking at the Diff you don't have RTH mode set only Failsafe mode. So you initiated RTH using Failsafe set to RTH ?. I assume you tried to cancel the RTH by switching Failsafe mode off ? Did it lock up the moment you switched off Failsafe mode or the moment you moved the sticks to cancel the failsafe or had it already locked up before you did anything to cancel the RTH ?

StuweFPV, 2022-05-27:
yes that is correct. I set failsafe as a mode and define the RTH procedure as the failsafe action. Crossfire is set to cut. worked well on over 40 planes so far and in many "real life" incidences where RTH saved me a plane. When i test, i switch on RTH and then immediately turn the switches back to ACRO but don't touch the sticks. the plane will perform RTH until i move the sticks more than 50% (or something around that value). In this case i'm much faster to react and don't have to worry about switch positions if something goes wrong - and it does a lot with a new airplane especially when you have the all famous inav drift per default on every plane.... (sorry this is besides the point but a shame that this old and so important issue is not being fixed - only very few people understand and do logging for filter adjustment..)

on this flight i've tried to cancel RTH after i realized something is wrong - so after the point where the plane should stop the turn and fly home. when watching the DVR it looks to me that exactly at the point where the plane should level (home arrow straight) the lock occurred. So my efforts to cancel were after the lock - hence no reaction of the plane - and trust me i've toggled every switch on my remote in those endless felt seconds before impact :)

closed 2022-12-01 by StuweFPV