iNavFlight/inav #8079: Modes not showing in iNav Configurator 5.0.0-rc1

billjow, 2022-06-03:
This is to follow on from #1534 which was closed for the wrong reason.

Have downloaded iNav Configurator 5.0.0, flashed it to my Cinewhoop using a Mamba F722-2IC flight controller, executed the DIFF ALL file from iNav 4.1. Going through the tabs, I could only see non-navigation modes so my Arm, Horizon (I don't bother with Angle) flight mode, and my Beeper. I normally have Nav WP, RTH, Altitude Hold, Waypoint Planner, Pos Hold etc. set up. I have now flown my quad, and all my modes are still operational but they don't show in Configurator. Screenshot below shows what I can see in Configurator. **If I slide the Hide unused modes back, I get other options, but not the nav ones.** My GPS and compass is working and I have successfully flight tested most of the nav modes. I have also tested Configurator on another PC and get the same behaviour.

## Current Behavior
[image]

[https://pastebin.com/ASjBVg3T](url)

version
INAV/MAMBAF722_I2C 5.0.0 May 26 2022 / 17:52:06 (8d70c7a1)
GCC-10.2.1 20201103 (release)

breadoven, 2022-06-03:
I take it this has an enabled Baro ?

billjow, 2022-06-03:
Yes, and it works fine in flight tests...