iNavFlight/inav #8080: iNav V5 RC2 - Broken MSP support for barometer

robustini, 2022-06-03:
## Current Behavior

Already with iNav V4.1 the support for MSP peripherals had some known problems, with the V5 RC2 the Mateksys M8Q-CAN gps the barometer is not recognized.
While with V4.1 it works with V5 and with the same parameters it doesn't work.
Tested on ZEEZF7 V1.2 on UART2. 3 and 4.

## Steps to Reproduce

1. Configure the device correctly in MSP
2. Look at the screen and it will give you the barometer meter in red.
3.
4.

## Expected behavior

## Suggested solution(s)

Fix the code.

## Additional context

[image]

---

# version
# INAV/ZEEZF7 5.0.0 Jun  2 2022 / 07:57:45 (3914c4f0)
# GCC-10.2.1 20201103 (release)

JulioCesarMatias, 2022-06-03:
Hi Marco, I'm a big fan of your work at ArduPilot. About this problem, I believe I found where the error is occurring, but to be sure of that, I need you to select the Barometer "FAKE" instead of "MSP".

JulioCesarMatias, 2022-06-03:
You can get the corrected build for your flight controller here >> https://github.com/iNavFlight/inav/actions/runs/2437143945

You can select "MSP" type Barometer and everything will work normally

robustini, 2022-06-04:
Hi @JulioCesarMatias, I'am honored to have you as a fan!
I looked at the code too but I couldn't find the problem, apparently there was a baro missing in the list, great job, thanks!!!

robustini, 2022-06-04:
Closed with https://github.com/iNavFlight/inav/issues/7576, work fine so please merge, thanks!

closed 2022-06-04 by robustini

robustini, 2022-07-09:
@DzikuVx  there is a problem with the configurator, irrelevant but there is.
In CLI the gps is set correctly in MSP, but then marks it as DJI Naza.

[image]

JulioCesarMatias, 2022-07-09:
> @DzikuVx there is a problem with the configurator, irrelevant but there is. In CLI the gps is set correctly in MSP, but then marks it as DJI Naza.
> 
> [image]

this is just a mismatch... Old Configurator vs iNav branch master for FC. Make this checkout with iNav 6.0 (Master version).

.Hex file >> https://github.com/iNavFlight/inav/actions/runs/2641160041

Configurator >> http://seyrsnys.myzen.co.uk/inav-configurator-next/