iNavFlight/inav #8086: INAV 5.0 RC2 AUTO DISARM ON LANDING FAIL TO INITIATE

abimaputra78, 2022-06-04:
## Current Behavior

https://user-images.githubusercontent.com/49124286/171986474-0301e944-2f41-4541-979a-5066b0e01e1c.mp4

## Steps to Reproduce

1. Set disarm_on_landing = ON
2. Gps Fix with sat > 6
3. During landing the multicopter still armed even no. 1 above
4.

## Expected behavior

During landing after RTH or failsafe the multicopter do disarm for safety reason 

## Suggested solution(s)
please fix auto disarm after landing

## Additional context

---

- FC Board name and vendor: KAKUTEH7 
- INAV version string: 5.0 RC2

Jetrell, 2022-06-04:
It can take some time to detect Landing, Up to 6 seconds. Or even more, under less than optimal conditions.
If the quad has an unstable base, and it leans to ones side at touchdown. Or the ground is not level, or its windy... These will all effect how long it takes to disarm.

breadoven, 2022-06-04:
It takes at least `nav_mc_auto_disarm_delay `(default 2 seconds) AFTER the FC has detected a stable landing has happened before it disarms. There is also a 2 seconds delay after the landing criteria are considered met before the `nav_mc_auto_disarm_delay `starts which is intended as a safety factor to avoid false disarms. The `nav_mc_auto_disarm_delay `can be set to a minimum of 0.1s though which will minimise the disarm delay.

However. the quad in the video would never have disarmed because it hadn't achieved a stable "landing" before it tipped over.

billjow, 2022-06-04:
Was not aware of that function, just tested it on my Green Hornet V3 cinewhoop on 5.0.0-rc1 - worked fine.

closed 2022-06-07 by DzikuVx