iNavFlight/inav #8089: MATEKF405TE no DSHOT on S3 (CH3N)

marcusmcelroy, 2022-06-05:
## Current Behavior
Although #7924 adds support for support for MATEKF405TE target (from release_5.0.0), DSHOT does not work on S3 output (PB15). 

As @MATEKSYS suggests in #7656 and on [Matek's FC tips](http://www.mateksys.com/?portfolio=f405-minite#tab-id-4), there is thought to be an INAV bug preventing this..

> I assume there is bug in INAV to support normal DMA on timer "N". BF supports it, so S3 on TIM1_CH3N (DMA2,6,6) works fine with Dshot with BF firmware.

Whilst the work-around proposed in #7656 and #7781 (use Multishot) does work, it makes for a clunky migration path from BF and has the obvious limitation of not being able to use RPM filtering. Furthermore, given EOL for other Matek mini FC's, I suspect this will be a common target for smaller quadcopter builds.

## Steps to Reproduce
1. Flash 5.0.0-RC2
2. Enable motor and servo output
3. ESC protocol = DSHOT300 or DSHOT600
4. Save/ reboot 
5. Motor test S3 (fail to spin)

## Expected behavior
DSHOT works on at least S1-S4 when using MATEKF405TE target for release >= 5.0.0.

## Suggested solution(s)
I've exhausted my skills, so perhaps someone could confirm that this is a bug, or a limitation of INAV DMA implementation, or a true limitation of MATEKF405TE?

## Additional context
Related issues
- #7656
- #7781

5.0.0-RC2 dump
- https://pastebin.com/X1yk2cMD

In BF, S3 uses AF1 (TIM1_CH3N) and DMA2,6,6
In INAV, S3 uses AF1 (TIM1_CH3N) and DMA2,6,0

Neither of those combinations work in INAV.

I've also tried compiling 5.0.0-RC2 with S3 (PB15) mapped to both AF1 (TIM1_CH3N) and AF3 (TIM8_CH3N), with various DMA request mapping, but I'm unable to get S3 (PB15) working with CH3N.

---
- FC Board name and vendor: 
MATEKF405 Mini TE (MATEKF405TE)

- INAV version string:
> # INAV/MATEKF405TE 5.0.0 Jun  2 2022 / 09:24:25 (3914c4f0)
> # GCC-10.2.1 20201103 (release)

mituritsyn, 2022-06-20:
Faced same issue. Why it was even merged if it evidently doesn't work? @MATEKSYS @DzikuVx any thoughts, guidelines, deadlines?

MATEKSYS, 2022-06-20:
> Faced same issue. Why it was even merged if it evidently doesn't work? @MATEKSYS @DzikuVx any thoughts, guidelines, deadlines?

Your ESC doesn't support multishot? oneshot?

marcusmcelroy, 2022-06-20:
What we're trying to establish with this issue is..

1) _why_ does DSHOT not work on S3?
2) how can we address that?

The workaround (use multishot or oneshot) works, but doesn't address these questions.

@MATEKSYS, you had previously suggested..

> I assume there is bug in INAV to support normal DMA on timer "N"

Given that DSHOT works in BF and Ardupilot, this _suggests_ an issue, or limitation with the INAV's handling of DMA. Could we focus there please?

aviphysics, 2022-07-07:
Is this issue getting attention? The multishot/singleshot work around is nice to have, but not a long term solution.

aviphysics, 2022-07-07:
@DzikuVx @MrD-RC is there someone that understands this type of thing we'll enough to look into it?

aviphysics, 2022-07-07:
@bkleiner is this something that you understand?

aviphysics, 2022-07-07:
BEEROTORF4 uses the same timer assignment on PB1 for M2, so it might be interesting to look at.

gokuhs, 2022-08-16:
I have the same issue on Matek F405-HDTE (MATEKF405TE) and with the ESC Mamba Master F55_96K. 

For check i try to flash BF and works property the ESC3.

mituritsyn, 2022-08-17:
@DzikuVx any ideas? assignees? due dates?

gokuhs, 2022-10-17:
I have an suggestion to bypass this problem:

In the target.c remove TIM_USE_MC_MOTOR in the S3. With that when inav assign the motors ignores S3 and we can use S4 for motor 3 and S5 for motor 4. 
[target.c.txt](https://github.com/iNavFlight/inav/files/9803541/target.c.txt)
With that i can use on a quad normaly. Is not a solution but another is the only way i can find to use in my drone.

m0nk1111, 2022-10-27:
is this ever gonna be fixed, or do i need to switch FC brand?

gokuhs, 2022-10-28:
@m0nk1111 I'm using this FC with my changes without problem. Please note the new wiring scheme; the Motor 3 is in the S4 and Motor 4 is in the S5. I attach the firmware of the versions 5.1.0 and the actual Master (i don't test this last one)
[MATEKF405TE.zip](https://github.com/iNavFlight/inav/files/9887685/MATEKF405TE.zip)

m0nk1111, 2022-10-31:
@gokuhs i appreciate your help. but i dont want to rewire S5 on the FC to S4 on my esc4in1 everytime i want to use INAV. also im using the jsh connector. Mateksys said on the product page, that S3, S5 and S7 have the same problem, so how why would this work?

gokuhs, 2022-11-02:
@m0nk1111 I know that it is not an ideal solution, but this bug has been open for a long time, and for me it was either to do something like that or to keep it in the drawer. I also use the JST, but of course I have to use both, and in the second only with one wire; the one from the S5. I know that on the Matek page they say that but for now it has worked well for me. And if you switch to Betaflight simply from the CLI you can forward ports without rewiring. I insist, it is not the most ideal, but at least we can put it to use.

sinisternm, 2022-11-23:
> @m0nk1111 I know that it is not an ideal solution, but this bug has been open for a long time, and for me it was either to do something like that or to keep it in the drawer. I also use the JST, but [...]

thank you for doing this, ill give it a try in the morning. shameful matek hasnt fixed this or offered better workarounds like yours.

sinisternm, 2022-11-24:
didn't work for me, bluejay showing just 3 esc's. tried both too, 6.0 is way bugged

gokuhs, 2022-11-24:
> didn't work for me, bluejay showing just 3 esc's. tried both too, 6.0 is way bugged. if i add another motor i see it, but S5 still seems to be S5, think im just gonna load BF on it and give up. boards garbage anyway, gyro noise its unfixable. thanks for nothing Matek.

Is strange... i'm flying with this firmware right now... Do you clean the configuration? And another thing... which is the "gyro noise"? Because flying, it flies, but it's true that I have a lot of vibrations that I can't remove by adjusting the pids...

sinisternm, 2022-11-26:
> Is strange... i'm flying with this firmware right now... Do you clean the configuration? And another thing... which is the "gyro noise"? Because flying, it flies, but it's true that I have a lot of [...]

clean, full erase? i full erased both and downloaded 6.0 just to test, just acted like it always did. i reuploaded diff though.
 on inav it flies like a spaz on multishot and alt/pos hold it would rocket up. tried everything, super clean build, caps/lc filters. main problem with the noisy gyro is it would nose down and barely recover randomly. i tried to tune with pidtoolbox many many times. googled for hours. the 3.5 inch i was trying doesn't have much info out there either. it flies like a dream on betaflight with dshot300/rpm filtering and default 3.5" presets. same situation without pos hold but its got RTH at least

MikeZakhar, 2023-02-05:
My setup is:
Matek f405-miniTE + ESC 4 in 1 MAMBA F30 mini 
I wanted to try one trick: Matek f405-miniTE supports up to 8 motors, so the idea was to use for quad instead of output 3 - output 6 for motor 3 (because 5,7 is not working for DShot too). And seems that's works. Partially)
I even can fly with this setup. But when I rise throttle to about 40-50% - voltage sensor becomes crazy, it's jumping up and down, telemetry sends me warming about low voltage, I can't monitory real voltage and battery remaining. 
Is this related with no DShot issue or it's another bug?

MikeZakhar, 2023-02-06:
> My setup is: Matek f405-miniTE + ESC 4 in 1 MAMBA F30 mini I wanted to try one trick: Matek f405-miniTE supports up to 8 motors, so the idea was to use for quad instead of output 3 - output 6 for [...]

I switched to multishot and voltage sensor becomes to normal values. So, I decided that's a part of the DShot issue.

aviphysics, 2023-04-10:
Any news on this issue. At this point I am thinking about submitting an alternate target that drops the problem channel

closed 2024-03-22 by DzikuVx

ClaudioFortes, 2024-04-05:
I always used INAV, but due to this issue because I can't use turtle mode without DSHOT, I migrated to Betaflight. With the new BF update, I was so excited to have an RTH as good as INAV, but I can't get a compass calibrated. So, I just tried to use INAV again, and DSHOT worked just fine! I'm using Configurator 7.1 and Firmware 7.1. 

I'm literally configuring the quad right now, so I can't say how it will be in the field.

commit 673d7435 (2023-02-10): Add targets using different timer for S3, in an attempt to fix DSHOT on
S3.

May fix #8089 and #8785

Add MATEKF405TE_S3DSHOT and MATEKF405TE_SD_S3DSHOT