iNavFlight/inav #8090: SA Broken for AKK FX3 in 5.0 RC1 and RC2

b14ckyy, 2022-06-05:
## Current Behavior
Smart Audio does not reliably work with AKK FX3 on both RC1 and RC1 of INAV 5.0. The OSD only occasionally after a few reboots shows correct channel and power level red from VTX. Changing power level does not work at all or maybe on 1 out of 10 retries. I am also not able to switch power level in flight with Programming. 

## Steps to Reproduce
1. Flash 5.0-RC2 on a Matek F411-WSE
2. connect Smart Audio for the AKK FX3 to ST1
3. configure SA on Soft-Serial 1
4. disable low power on disarm
5. check OSD element for VTX with monitor/goggles
6. change band, channel or power level in configurator or OSD menu
- only very occasionally the VTX will actually react to changes in settings, if at all
- Flashing back to 4.1 with the complete diff all from 5.0, and SA works reliable as expected

## Expected behavior
SA working the same way as in 4,1 and older

## Suggested solution(s)
Fix SA

## Additional context
This setup with an F411-WSE and AKK FX3 was the first setup I flew with 5.0. Will check other builds with different FC and VTX as well
Full DIFF: 
[SA fail diff.txt](https://github.com/iNavFlight/inav/files/8840401/SA.fail.diff.txt)
---

- FC Board name and vendor: Matek F411-WSE, AKK FX3
- INAV version string: # INAV/MATEKF411SE 5.0.0 May 26 2022 / 18:11:37 (8d70c7a1)

erstec, 2022-06-05:
@b14ckyy 
try `vtx_smartaudio_early_akk_workaround`, `vtx_softserial_shortstop` and `vtx_smartaudio_sixtynine` in different combinations.
https://github.com/iNavFlight/inav/blob/master/docs/VTx.md

b14ckyy, 2022-06-05:
> @b14ckyy try `vtx_smartaudio_early_akk_workaround`, `vtx_smartaudio_stopbits` and `vtx_smartaudio_sixtynine` in different combinations. https://github.com/iNavFlight/inav/blob/master/docs/VTx.md

vtx_smartaudio_alternate_softserial_method=OFF did the trick
vtx_smartaudio_early_akk_workaround ON or OFF makes no difference. 

Is this a new setting in 5.0? 
Because it is confusing. 

> Enable the alternate softserial method. This is the method used in iNav 3.0 and ealier.
Default is ON. But it worked previously also in INAV 3.0 and also 2.6. But now I have to set this setting to OFF to NOT use the old method to get SA to work. And it worked fine in 4.0 and 4.1 as well.

erstec, 2022-06-05:
> vtx_smartaudio_alternate_softserial_method=OFF did the trick vtx_smartaudio_early_akk_workaround ON or OFF makes no difference.
>
> Is this a new setting in 5.0? Because it is confusing.

I don't think so. SoftSerial was added with ability to adjust stopbit length, as originally IVAN have 3xlonger stopbit, rather it should be, becuase of timer interval and way softserial function process states (to comply with other manufacturer VTX'es).
So it is nothing strange that after those changes you don't need this option enabled.

b14ckyy, 2022-06-05:
OK so then this Option should be OFF by default then in 5.0. Because this would be the new way how it works? 
with a DIFF upgrade from 4.1 the new default setting should apply if no one has changed that setting before.

erstec, 2022-06-05:
> OK so then this Option should be OFF by default then in 5.0. Because this would be the new way how it works? with a DIFF upgrade from 4.1 the new default setting should apply if no one has changed that setting before.

There was no deep testing with all possible 3rd party SA VTX'es, so I would not be so categorical that if in your case it should be `OFF` doesn't mean it should be like that for all. If majority of  users will report that it should be changed to `OFF` - it will be.

Regarding second part of your message: migrating diff between major versions always need to pay attention to what are transferred. All those "workarounds" with deviations from Standard SA Protocol - are workarounds, but not standard, but what I know for sure is that with defaults it is working with TBS VTX'es, without altering defaults.

MrD-RC, 2022-06-05:
When `vtx_smartaudio_alternate_softserial_method` was added. The default value was set so that it would not mess up current setups. In other words, the change was bypassed. If the default value has been changed in 5.0 is should be put back.

erstec, 2022-06-05:
> When `vtx_smartaudio_alternate_softserial_method` was added. The default value was set so that it would not mess up current setups. In other words, the change was bypassed. If the default value has been changed in 5.0 is should be put back.

No, it was not changed in 5.0

MrD-RC, 2022-06-05:
Ok, so I don't get why we have a problem. By default the number of stop bits is the same as before that change, so it shouldn't effect anything?

b14ckyy, 2022-06-05:
This is what I don't understand. 

SA works fine on 4.1 and older, back to 2.6. I never touched vtx_smartaudio_alternate_softserial_method before. 
After flash to 5.0 SA was not working anymore with vtx_smartaudio_alternate_softserial_method=ON. But works with it set to OFF.

So from pure logic something must be different between vtx_smartaudio_alternate_softserial_method=ON in 5.0 and Soft Serial SA on 4.1.

erstec, 2022-06-05:
> Ok, so I don't get why we have a problem. By default the number of stop bits is the same as before that change, so it shouldn't effect anything?

Exactly!

Even more, softserial stopbit "duration" are only for Tramp serial configuration `vtxTrampInit()`, not for SA (I just rechecked the code). 

For SA there was option added to choose 1 or 2 stopbits, and default is 2, as it was before.

So, I don't understand how it can affect bahavior too...

@b14ckyy give me your diff, I will try to see what is a difference in what FC sends-out on 4.1 and on 5.0 (timings at least)

MrD-RC, 2022-06-05:
@b14ckyy nope, the `vtx_smartaudio_alternate_softserial_method` has not changed at all. It was added in 4.0 by the way. Pre 4.0 it would have only acted as if this setting was `ON`.

[image]

b14ckyy, 2022-06-05:
@erstec diff is attached in the opener

erstec, 2022-06-05:
@b14ckyy can you check on real HW test version I will provide you?

b14ckyy, 2022-06-05:
yes sure

erstec, 2022-06-05:
[inav_5.0.0_MATEKF411SE.hex.zip](https://github.com/iNavFlight/inav/files/8840535/inav_5.0.0_MATEKF411SE.hex.zip)

MrD-RC, 2022-06-05:
It is strange. Doing a compare of `vtx_smartaudio.c`. The only difference is the stopbits line (692). But if stopbits is set to 2 (the default value), there is no difference to how it would have worked in 4.0.

erstec, 2022-06-05:
> It is strange. Doing a compare of `vtx_smartaudio.c`. The only difference is the stopbits line (692). But if stopbits is set to 2 (the default value), there is no difference to how it would have worked in 4.0.

It seems I found the issue. Waiting for @b14ckyy test result, then provide one more test FW and do PR

b14ckyy, 2022-06-05:
@erstec All good now with this version. I have set both settings back to ON (akk workaround and old method) and SA is working as expected.

erstec, 2022-06-05:
@b14ckyy please test this one
[inav_5.0.0_MATEKF411SE.hex.zip](https://github.com/iNavFlight/inav/files/8840549/inav_5.0.0_MATEKF411SE.hex.zip)

b14ckyy, 2022-06-05:
> @b14ckyy please test this one [inav_5.0.0_MATEKF411SE.hex.zip](https://github.com/iNavFlight/inav/files/8840549/inav_5.0.0_MATEKF411SE.hex.zip)

works as well on this setup

erstec, 2022-06-05:
Good news! Thank you @b14ckyy for testing!

@MrD-RC, "real" isuue was my innatention, sorry about that...

Fix https://github.com/iNavFlight/inav/pull/8091

b14ckyy, 2022-06-05:
ok then let's close this here

closed 2022-06-05 by b14ckyy

MrD-RC, 2022-06-05:
> Good news! Thank you @b14ckyy for testing!
> 
> 
> 
> @MrD-RC, "real" issue was my innatention, sorry about that...
> 
> 
> 
> Fix https://github.com/iNavFlight/inav/pull/8091

It's fixed. It's all good. I've added a label and note to try and get this in 5.0.