iNavFlight/inav #8092: INAV RTH flies to the moon!

akurekin, 2022-06-06:
**PLEASE MAKE SURE YOU READ AND UNDERSTAND THE SOCIAL MEDIA SUPPORT CHANNELS. QUESTIONS ABOUT FLASHING, CONFIGURING, PILOTING MAY BE CLOSED WITHOUT FURTHER INTERACTION.**

* [Telegram channel](https://t.me/INAVFlight)
* [Facebook group](https://www.facebook.com/groups/INAVOfficial)
* [RC Groups thread](https://www.rcgroups.com/forums/showthread.php?2495732-Cleanflight-iNav-%28navigation-rewrite%29-project)

**Please double-check that nobody reported the issue before by using search in this bug tracker.**

**PLEASE DELETE THE TEXT ABOVE AFTER READING AND UNDERSTANDING IT**

****

## Current Behavior

When RTH is activated the drone shoots up at full throttle to about 20 meters, then starts its way to the home position. Some times it happens and some time not, more often at the wind. 
Same behaviour is observed in Position Hold when the drone goes down fast and then rapidly starts moving up, in a don - up pattern.    

## Steps to Reproduce

1. Take drone off the ground
2. Activate Position Hold
3. Move to about 10 meters up
4. Throttle down
5. At about3 meters above the ground throttle up to max

## Expected behavior

Drone starts ascending at slow speed

## Suggested solution(s)

It looks like a bug. At the moment when the drone shoots up to the skies, navPos[2] shows the drone goes down and navVel[2] shows negative velocity. At the same time baro shows that the drone goes up and acc[Z] is around 2g. This moment is shown in the log attached at 36.789 sec. of flight time.

After a large number of experiments with prop balancing, barometer covering in foam and playing with settings I found a work around by setting inav_w_z_gps_v = 0 .  

## Additional context

---

- FC Board name and vendor: MATEK 405SE
- INAV version string: 3.0.1
[LOG00135.TXT](https://github.com/iNavFlight/inav/files/8842517/LOG00135.TXT)

stronnag, 2022-06-06:
```
$ inav_states.rb  ~/dl/LOG00135.TXT 
INAV 3.0.1, states from 2.7.0
LOG00135.TXT: INAV 3.0.1 (753c4479) MATEKF405SE
Iteration	Time(s)	Elapsed(s)	State	FltMode	navFlag
   225984	 136.5	(   0.0)	nav_state_poshold_3d_in_progress (7)	ARM|ANGLE|NAVPOSHOLD|BLACKBOX		
   298592	 173.8	(  37.3)	nav_state_idle (1)	ARM|ANGLE|BLACKBOX		
   322464	 186.1	(  49.6)	nav_state_poshold_3d_in_progress (7)	ARM|ANGLE|NAVPOSHOLD|BLACKBOX		
   355882	 203.3	(  66.8)	end of log
Disarmed by: SWITCH
```
[image]

That the baro initialises at 8m is not helping. I doubt this is a software issue. It would be interesting to see what happens with contemporary firmware, 3.0 is no longer maintained.

akurekin, 2022-06-06:
The barometer shows some thermal drift over time, but it is very slow. It is also protected from wind, The vibrations are not that high as it follows from acc measurements. The same issue in INAV v4 and 5.  Any ideas of what the problem can be?

stronnag, 2022-06-06:
> What really looks strange is the velosity navVel[2] is estimated as negative when the copter goes up ! The velocity estimate looks wrong.

That is from the accelerometer, which is the crux of the matter.

Jetrell, 2022-06-06:
Comparing your `gyro_raw `data verses the filtered data. The `gyro's (yaw) axis `still has vibrations around 50 - 60hz.

The IMU data augments the barometric data, when abrupt changes in altitude occur.... This will cause the effect you are seeing if the sensor is blinded at certain frequencies by corrupt data, and has nothing to do with the barometer.
Filtering is very important. But its more important to find where those vibrations are coming from.
Being that they are low in the frequency range. That would likely indicate frame resonance. But checking the motors for balance, is important too. 
Makes sure the FC stack bolts are not over tightened. This can transfer vibrations to the IMU, by over-compressing the silicone dampers.

akurekin, 2022-06-07:
Thank you for your recommendations, I'll check if the frame resonance can be an issue. I  tried to put controller on a very soft foam with no success. It is also not clear how yaw vibrations can affect the height estimate, as they have different directions.  
But if we admit that it's a resonance issue, the resonance  may happen when the drone starts moving up when the motors get high power at the first moment to start pushing the copter up. This abrupt jump in power may cause additional vibration. It must be a way to make this transition smoother to reduce a chance of resonance at this moment