iNavFlight/inav #8109: pwm init error in aocoda F4

wang322588, 2022-06-11:
## Current Behavior
I flash the customed firmware of aocoda F4v3 while osd show "pwm init error" .And can not arm.
This error is usually caused by fireware internal error.I can't find information in inav wiki.

## Desired Behavior
No osd error and should be armed.

## Suggested Solution
It said the board support inav 4.1.0.
I think it is a customed firmware made by manufactor and not stable.maybe default resource config error.
Betaflight is fully support this target of AOCODAF405 and works fine.
I hope inav can support this board.Its cheap and with barometer.

## Who does this impact? Who is this for?
All users

## Additional context
product link (https://www.aliexpress.com/item/1005003575874474.html)
firmware link(https://drive.google.com/file/d/1hdPxVk8xBL4dKbcxAnYty0QuEPFo2F8O/view?usp=sharing)

DzikuVx, 2022-06-11:
This board is not supported by INAV. Please contact aocoda customer support . Or at least seller's customer support. INAV project never released hex for this flight controller

closed 2022-06-11 by DzikuVx

robustini, 2022-12-08:
> ## Current Behavior
>
> I flash the customed firmware of aocoda F4v3 while osd show "pwm init error" .And can not arm. This error is usually caused by fireware internal error.I can't find information [...]

Where did you find the target for this F4?

sensei-hacker, 2023-10-10:
For future readers, this board IS now supported in INAV.