iNavFlight/inav #8111: Cannot connect via USB to FLYWOOF745HEX after flashing inav

pasyn, 2022-06-11:
## Current Behavior
Cannot connect via USB to FLYWOOF745HEX after flashing inav. 
Inav-configurator times out with "No configuration received within 10 seconds, communication failed"

## Steps to Reproduce
1. Put FLYWOOF745HEX into DFU mode
2. Flash Inav with inav-configurator
3. After successful flash powercycle FC
4. Attempt to connect to FC via Inav-configurator

I have replicated this issue on both linux and windows with Inav versions 3.0.2, 4.1.0 and 5.0.0 using the corresponding versions of INAC-configurator.

I have successfully flashed and connected to betaflight using both the FLYWOOF745 and FLYWOOF745NANO targets.

I have also confirmed that Inav-configurator is functioning properly by successful flashing and connecting to a SPEEDYBEEF7.

## Expected Behavior
Board should be supported since version 3.0.0 according to [https://github.com/iNavFlight/inav/issues/6522](https://github.com/iNavFlight/inav/issues/6522)

## Additional context
The board I am attempting to use should be the same as the following link:
[https://flywoo.net/products/goku-hex-f745-fc-16mm-x-16mm-w-4pcs-ws2812-led](https://flywoo.net/products/goku-hex-f745-fc-16mm-x-16mm-w-4pcs-ws2812-led)
But it came on the hexplorer HD:
[https://flywoo.net/collections/micro-long-range-hexplorer-le-4-hexacopters/products/hexplorer-lr-4-hexa-copter-bnf](https://flywoo.net/collections/micro-long-range-hexplorer-le-4-hexacopters/products/hexplorer-lr-4-hexa-copter-bnf)
I am not entirely sure if there are any other differences but I do know that the board that comes with the BNF Hexplorer HD is missing the OSD chip. I assume it was desoldered from the board for weight savings as it is not needed with DJI vtx.

- FC Board name and vendor: FLYWOOF745HEX
- Target: FLYWOOF745
- INAV version: 4.1.0

According to the flywoo firmware documentation, the pinio should be the same as the FLYWOOf745 even though the fc board has the same look and form factor as the FLYWOOF745NANO. As far as I can tell the only difference between the two is the pin assignments for uart3 and i2c_2, neither of which should affect usb communication.

matstey, 2022-06-12:
I have been having the same issue. See this [discussion](https://github.com/iNavFlight/inav/discussions/7546). As far as i can tell right now commenting out `#define USE_MAX7456` in the target seems to solve the issue but this would mean adding a new target so i have been trying to see if i can make a change to handle the osd chip missing better.

pasyn, 2022-06-12:
Thanks for the lead. I found that flashing `inav_3.0.0_FLYWOO_F745HD.hex` from [https://flywoo.tawk.help/article/firmware-inav](https://flywoo.tawk.help/article/firmware-inav) works on the board without the OSD chip.

closed 2022-07-22 by JulioCesarMatias