iNavFlight/inav #8112: 5.0-RC3 Saving multi mission to EEPROM removes the mission index lines and merges all missions into one

b14ckyy, 2022-06-11:
## Current Behavior
If a Multi Mission is created and uploaded to FC (not EEPROM yet) and then downloaded, it is still a multi mission with selectable mission Index. But then storing the mission to EEPROM removes the Index markers. So after saving and loading from eeprom all multi missions are loaded as a single mission

## Steps to Reproduce

Mission 1:
[image]

Mission 2:
[image]

Uploaded all missions to FC: 
[image]

Saved to EEPROM and then loaded from EEPROM: 
[image]

## Expected behavior
EEPROM should keep the mission index for multi missions.

## Suggested solution(s)
fix

## Additional context
Reproduced by @stronnag and also happens with impload. 
---
INAV/MATEKF722WPX 5.0.0 Jun 10 2022 / 09:30:03 (dfa9f6f6)
GCC-10.2.1 20201103 (release)

- FC Board name and vendor: Matek F722-WPX

b14ckyy, 2022-06-11:
All good with 4.1 on same setup: 
[image]

JulioCesarMatias, 2022-06-11:
I'm not sure if this is a problem with saving WP's in flash memory... because if it was, missions 2 would be overwritten over missions 1, so this duplication wouldn't appear on the map... The two missions would be the same

b14ckyy, 2022-06-11:
The Configurator or MWP still send all waypoints of both missions in a row to the FC so the total waypoints are a sum of both missions. Not overwritten. it is the Mission End Flag (wp 53 here) that is missing after saving the eeprom. so INAV handles both missions as one big mission.

breadoven, 2022-06-11:
Hmmm, I can't replicate this in Configurator. Works as expected although my Configurator and INAV aren't completely up to date with the master. I'll update and try again but I'm not aware of anything that's changed related to loading and saving waypoints.

b14ckyy, 2022-06-11:
Yeah I am testing with RC3 INAV and RC2 Configurator. But also MWP shows the issue. @stronnag is also looking into it and not sure yet.

JulioCesarMatias, 2022-06-11:
> Hmmm, I can't replicate this using Configurator. Works as expected although my Configurator and INAV aren't completely up to date with the master. I'll update and try again but I'm not aware of anything that's changed related to loading and saving waypoints.

2 (me too)

b14ckyy, 2022-06-11:
Here is my mission file. It contains both variants as shown above.
[Terrain Follow Low Alt.zip](https://github.com/iNavFlight/inav/files/8884724/Terrain.Follow.Low.Alt.zip)

JulioCesarMatias, 2022-06-11:
I'm using iNav config and I can't reproduce the problem (Updated with Master)

[image]
[image]

b14ckyy, 2022-06-11:
@JulioCesarMatias have you written zu EEPROM and then loaded from EEPROM again? 
Can you please check with my mission file?

breadoven, 2022-06-11:
Updated INAV and Configurator and I can't replicate this on a HGLRCF722 FC.

Load 1 mission file
Open Multi mission -> add new mission -> load another mission from file
Save to FC
(Also tried Load from FC here)
Save to EEPROM
Load from EEPROM

Seems to work as expected with 165 end flags where they should be. Tried 53 WP file + 23 WP missions, made no difference to using smaller missions. Something related to particular boards perhaps ?

JulioCesarMatias, 2022-06-11:
I tested it over and over again and there was no problem, it worked fine

JulioCesarMatias, 2022-06-11:
> @JulioCesarMatias have you written zu EEPROM and then loaded from EEPROM again? 
> yes
> Can you please check with my mission file?
> yes, I spent at least 50 minutes recording and re-recording, and it didn't fail
> Also configurator can be problematic because it has a re-appeared bug in itself: [iNavFlight/inav-configurator#1553](https://github.com/iNavFlight/inav-configurator/issues/1553)

JulioCesarMatias, 2022-06-11:
I'm going to remove the "BUG" label, because I didn't find one, just like @breadoven

breadoven, 2022-06-12:
There isn't really any difference between saving to FC and saving to EEPROM in regard to the way data is transferred. When you save to FC it fills the INAV posControl.waypointList with waypoints. When you save to EEPROM it simply does the same thing initially, filling posControl.waypointList with waypoints, but then sends an extra command to save posControl.waypointList to EEPROM. So it's hard to see how saving to the FC changes anything here.

The only thing that can mess things up is if you Arm and then Disarm before saving to EEPROM because Arming changes posControl.waypointList. However, in this case you overwrite the Multimission and just end up with 1 mission, the mission that was loaded when you armed, it shouldn't result in the Multimissions merging into one.

https://github.com/iNavFlight/inav-configurator/issues/1553 and the excessive time it takes to transfer the waypoints suggests some sort of serial issue that is corrupting data.

b14ckyy, 2022-06-12:
OK so because of https://github.com/iNavFlight/inav-configurator/issues/1553 I was not able to download-check the WP mission in configurator and I don't even trust the upload to plan a WP mission. So I used MWPTools. 

With an F405-Wing it seems all works just fine. Direct USB.
With the F722-WPX I get the issues from https://github.com/iNavFlight/inav-configurator/issues/1553 that seems to be the same behavior as we had before in https://github.com/iNavFlight/inav-configurator/issues/1366

But for some freaking reason now it works while I am typing...... I have reflashed the FC back with 4.1 yesterday and now back to 5.0 with full diff and now it works.... WTF?! 

Now that this bug does not appear and I try it with configurator, all is fine here. Although the mission upoload takes FOREVER and the times are very inconsistent with the same mission: 

```
2022-06-12 @ 10:48:05 -- Start send point

2022-06-12 @ 10:48:30 -- Send time: 25214ms

2022-06-12 @ 10:48:30 -- End send point

2022-06-12 @ 10:48:31 -- Start send point

2022-06-12 @ 10:48:44 -- Send time: 13183ms

2022-06-12 @ 10:48:44 -- End send point

2022-06-12 @ 10:48:52 -- Start send point

2022-06-12 @ 10:49:05 -- Send time: 13132ms

2022-06-12 @ 10:49:05 -- End send point

2022-06-12 @ 10:49:07 -- Start send point

2022-06-12 @ 10:49:11 -- Send time: 3968ms

2022-06-12 @ 10:49:11 -- End send point
```

For the F405-Wing the time is always between 3000 and 4000ms. 

#############

I have no idea why the Mission transfer now works though. The loss of Index lines only seems to happen with MWP then. 
Now I can't even replicate https://github.com/iNavFlight/inav-configurator/issues/1553 anymore....

breadoven, 2022-06-12:
To do with serial.h change https://github.com/iNavFlight/inav/pull/8061/files#diff-1d244b1147ffb35376f8c7ad2d562ea8b916268e6f2148afc266cff8ef2eabbb perhaps ?

b14ckyy, 2022-06-12:
wasn't this the fix for Smart Audio not working over Softserial with Old Method ON? 

Btw I have doublechecked now with 4.1. INAV 4.1 and same MWP version successfully uploads multi missions to the FC and can also download as multi mission. But with 5.0 RC3 the multi mission index gets lost somehow. 
Yes it seems like with the configurator it works in both versions but the question is, what has changed to cause this discrepancy. 

Jonathan said nothing should have changed in regards of wp transfer from 4.1 > 5.0. But still the behavior changed.

b14ckyy, 2022-06-12:
Yep it now works all the time.... Yesterday I tried it over and over and got the same results with merged missions every single time and now works. I have no idea why.

b14ckyy, 2022-06-13:
OK after more and more testing it seems to be an issue with MWP only and not INAV or configurator. I'll close this here. 
Thanks guys for testing it out 👍

closed 2022-06-13 by b14ckyy