iNavFlight/inav #8114: INAV 5.0 RC 3.0 irregular behavior during alt and POS HOLD

abimaputra78, 2022-06-12:
## Current Behavior

During alt HOLD throttle acts irregularly and suddenly punch out with no reason. Pos HOLD it just not hold but unexpected hold behavior 

## Steps to Reproduce

1. Arm
2. Fly cruise with 20 mtrs of alt
3. Activated alt HOLD
4. Alt Hold as expected but suddenly throttle punch out as unexpected behavior. 
5. Disactivated Alt Hold, Activated Pos HOLD and HOLD behavior can not stable like previous INAV version

## Expected behavior

Altitude HOLD should perform stable altitude hold no unexpected throttle punch out and so does the pos HOLD

## Suggested solution(s)

Improve of altitude HOLD and POS Hold become stable like before

## Additional context

---

- FC Board name and vendor: Kakute H7
- INAV version string: 5.0 RC2

JulioCesarMatias, 2022-06-12:
on my multirotor the Alt-Hold works well... the Pos-Hold fires backwards, you can see it in this video >> https://drive.google.com/file/d/1UTq7lmSEVHoVsminCD49ZzAA7LzgenI1/view?usp=sharing

abimaputra78, 2022-06-12:
Yups agreed. Actually alt hold In 5.0 really great, but with note throttle
punch out suddenly. 🙏 No idea what caused this.

Jetrell, 2022-06-13:
There is a similar issue open in the discussions https://github.com/iNavFlight/inav/discussions/8096.. Have a read and see if you can extract any information to help in your case. 
These issues are generally all vibration related. And can appear occasionally after firmware changes, if there is a slight change to the filtering methods. 
I experienced this when going from 3.02 to 4.1.

abimaputra78, 2022-06-13:
Im afraid its not a vibration issue. 🙏

Jetrell, 2022-06-13:
You have checked for hardware vibrations? By looking at `raw gyro` log data, to be sure.

dadddda, 2022-06-27:
I'm also on iNav 5.0 RC3 and when I arm for the first time in angle mode and then switch to POSHOLD everything works fine, but after a little bit of flying, landing, disarming and then arming again in angle mode and changing the flight mode to POSHOLD makes the quad jump very rapidly on the Z axis for about 20-30cm and then stabilize to the ground.

Checked the blackbox and I don't see any unusual gyro/accelerometer vibrations. All seems good.

closed 2024-03-20 by b14ckyy