iNavFlight/inav #8119: JHEMCU GF16 F405 problem OSD

waleri69, 2022-06-13:
JHEMCU GF16  F405 Flight Controller
Purchased a GF-16. INAV firmware version 5.0.0. I'm setting it up for now. There is a problem with OSD! The OSD works and is configured using the transmitters knobs. In INAV 5.0. I can't configure OSD!
INAV does not read OSD and does not configure ! I tried to change the font and change the type of TV signal. Nothing works out. Help fix this problem!!!
I went through Betalight and checked. OSD in Betaflight is configured both from the computer and from the remote control. There are no problems in Betaflight.
In INAV 5.0.0. OSD is not readable or configurable from the computer.
Nothing works out. Help fix this problem!!!

DzikuVx, 2022-06-23:
I'm sorry, but I don't understand what is not working in your case. Maybe you could provide screenshots that shows what is your problem?

fa-lk, 2022-08-15:
@waleri69 
1. what target did you flash?
2. did you upload a font after flashing?

closed 2022-10-01 by DzikuVx