iNavFlight/inav #8122: MATEK M9N-5883 COMPASS INSTABILITY INAV 5.0-RC1

66nola66, 2022-06-14:
MATEK M9N-5883 COMPASS INSTABILITY INAV 5.0-RC1. Matek MINI-H743 FC using M9N-5883 mag was setup and calibrated per SOP in INAV 4.1 with no issues, both GPS and MAG were running better than expected, MAG was very accurate and had rock solid performance. Loaded INAV 5.0 RC1 and calibrated MAG using SOP but noticed immediately that the compass was drifting. removed FC and GPS/MAG from frame and attempted to calibrate being isolated from magnetic sources but had same results. Replace both FC and GPS/MAG to frame and reverted back to INAV 4.1 which resolved issue, GPS/MAG is again work flawlessly, MAG is rock steady without any drifting. 

**PLEASE DELETE THE TEXT ABOVE AFTER READING AND UNDERSTANDING IT**

****

## Current Behavior

## Steps to Reproduce

1.
2.
3.
4.

## Expected behavior

## Suggested solution(s)

## Additional context

---

- FC Board name and vendor:
- INAV version string:

Jetrell, 2022-06-17:
@66nola66 I was looking for changes between 4.1 and 5.0RC1, that may have influenced the mag drift you are experiencing with the M9N+5883.  
Nothing stands out, and I don't have this hardware.

But you could you try this firmware. It occurred right before some changes that _might_ be related.  **Don't fly with it, just do a bench test.** and report back.
[inav_5.0.0_MATEKH743.zip](https://github.com/iNavFlight/inav/files/8925128/inav_5.0.0_MATEKH743.zip)

66nola66, 2022-06-20:
Jetrell hello and thank you so much for the assistance. I loaded the Matek H743 test firmware you sent and yes it works flawlessly as the 4.1 firmware does. Just to verify and confirm., I loaded the 5.0-rc1 FW again and the problem returned with the compass drifting and being sluggish to come to it's correct heading if ever(kinda just wanders around). I went out on a limb and saw that there is now 5.0 RC3 firmware released and just for kicks loaded it and found that the issue is resolved and the 5.0-RC3 firmware is rock solid with my hardware(matek M9n-5883 GPS with H743mini FC).   So what ever it was seems to only be affecting the 5.0-RC1 FW(possibly RC2 also but I did not check it) I will fly the RC3 firmware tomorrow and let you know but all looks good so far. THANKS FOR YOUR HELP!!!!!
SK

Jetrell, 2022-06-20:
@66nola66 I'm glad it works now... 
But now that you have said it works with RC3 and didn't with RC1.. I'm totally not sure what it could be :-/ ... There was no significant change between the two.
Anyhow.. Have fun flying :+1:

closed 2022-06-29 by DzikuVx