iNavFlight/inav #8126: RTH agressive maneuvers

akurekin, 2022-06-15:
## Current Behavior

When my quad was in acro mode flying forwad at low speed I activated RTH and the quad turned upside down in the air to return home. After multiple swings It managed to stabilise itself and started return to home position. It looks like the copter was changing its orientation too aggressively when RTH was activated, and could not control the rotation speed properly with overshooting effect. I don't know if this is important and can affect this problem, but the battery on drone is located way below the propeller plane.  

## Steps to Reproduce

1. Take off
2. activate acro mode
3. fly at moderate speed away from home position
4. activate RTH

## Expected behavior

The drone should start returning home

## Suggested solution(s)

I think the manoeuvrers at the moment when RTH mode is activated, are too aggressive, the drone starts moving with the angle acceleration that is too high and overshoots motor power. The drone should change its orientation much more slowly, it is not a racing drone.  

## Additional context

---

- FC Board name and vendor: MATEK 405SE
- INAV version string: 5.0.0 RC2

akurekin, 2022-06-15:
There  is not problem with climbing up or descending down. It is controlled very well and I can fly confidently in acro mode. But it make a very aggressive manoeuvre when it starts returning home.

Jetrell, 2022-06-15:
By reducing the navigation climb rate, it will also reduce the instantaneous vertical acceleration that is occurring at the points the MC is also slowing and turning for home.  
As I said before. Its only a work around and not a fix for a less than perfect setup or loose tune.

I've done much RTH and Poshold activation's testing in Acro and Horizon modes, at higher speeds, and in the middle of turn... With quads up to 6inch. The worst they do is kick a little in windy conditions...  But my quads are well setup and tuned.

akurekin, 2022-06-15:
The drone was above the RTH hight and was not climbing up. It would be great if I could control angular acceleration, not vertical.

akurekin, 2022-06-16:
How about AXIS_ACCEL_MIN_LIMIT parameter? Can I reduce it? I don't need my drone be sharp in flight, it's a long range drone.

Jetrell, 2022-06-16:
I'm not aware of that being a CLI instruction..  It maybe a hard coded variable.

You still haven't told me to what level your PID's are tuned?

When I said spoke of a loose tune... I meant the Proportional, Integral and Derivative values are only stock, or only tuned to, maybe 50% of what gives a firm tune.
Tuning a MC to at least 90% of the best the motors and hardware can provide, does not have to be a performance thing... It just increases overall stability.

akurekin, 2022-06-16:
PID parameters are currently tuned as follows: P=40; I=25; D=20. These are set to minimum values that will provide reasonably good performance. I tried different PID settings and they did not have significant effect. I wonder if rate_accel_limit_roll_pitch parameter can be used to control angular acceleration in RTH mode?

Jetrell, 2022-06-16:
>I wonder if rate_accel_limit_roll_pitch parameter can be used to control angular acceleration in RTH mode?

No sorry. It only controls rotational speed based on stick input. It won't effect the initial attitude control when switch to RTH.

>Hal serupa sama dengan saya rth terlalu agresif di inav 5.. sekarang saya memakai inav 5.3rc...

 Tidak ada yang berubah antara 5.0RC1 dan 5.0RC3.

>saya sudah mencoba berbagai hal..dari setting GPS pid,alt pid,saat setel alt pid quad agak melambat tapi poshold goyah

 Anda perlu berhati-hati. Bahkan sedikit perubahan dalam pengaturan tersebut dapat menyebabkan perbedaan besar. Secara efektif menambah masalah yang Anda alami.

akurekin, 2022-06-16:
What parameters then control the flight aggressiveness in RTH mode ?

wawanlost, 2022-06-16:
> 

Maksud saya inav versi 3.02..navigasi benar dan bisa landing secara baik sampai melucuti motor otomatis,dan pas mau pendaratan inav 3 turun terus,(tidak pernah naik seperti inav 4 dan inav 5

wawanlost, 2022-06-16:
Diinav 5 saya harus menurunkan pid alt dan pid GPS...itu membuat quad tidak terlalu agresif,dan bisa melucuti motor otomatis walaupun dengan waktu yg agak lama,tapi motor akhirnya berhenti...entah apa perbedaan antara inav 3,4 dan 5 dalam mode navigasi...kalau dalam pesawat terbang saya belum merasakan perbedaan besar tapi untuk quad itu jelas berbeda..

akurekin, 2022-06-16:
It looks like the discussion has diverged from the original topic and I'll try to bring it back on it's  rails. I have to admit, I did not pay enough attention to PID tuning as I did not think that PIDs can affect slow angular fluctuations. i made another experiment today and increased D values for roll and pitch from 20 to 35 and have got well behaving drone on its way to home in RTH mode. I think the problem was in slow fluctuations due to low centre of gravity.  I don't want to be too optimistic  but it looks like the problem has been solved by increasing D values. I would like to thank Jetrel for pointing me into this direction.  
Regarding the problem that Wawanlost described, it looks like high frequency frame resonance or unbalanced propellers  that cause the drone to rapidly go up on RTH.

Jetrell, 2022-06-16:
> It looks like the discussion has diverged from the original topic and I'll try to bring it back on it's rails. I have to admit, I did not pay enough attention to PID tuning as I did not think that PIDs can affect slow angular fluctuations. 

Great news. I'm glad you picked up on the direction I was pointing :+1:

Jetrell, 2022-06-16:
> entah apa perbedaan antara inav 3,4 dan 5 dalam mode navigasi...kalau dalam pesawat terbang saya belum merasakan perbedaan besar tapi untuk quad itu jelas berbeda..

Ada perubahan yang dilakukan pada kinerja multicoper, di banyak area. Ini adalah penyaringan dan navigasi di sekitarnya. Tergantung pada bagaimana MC Anda disetel sebelumnya. Tidak semua perubahan ini mungkin kompatibel.. Tetapi mereka akan meningkatkan kinerja, setelah Anda mengatur agar sesuai dengan perubahan.

wawanlost, 2022-06-19:
ya semoga saja anda benar,karna saya mengikuti inav dari versi 1 hingga versi 5...mungkin inav 5 dikhususkan unuk perangkat atau hardware terbaru..supaya campactible. dan berjalan mulus...saya mengaharapkan mode navigasi inav5 bisa seperti inav 3 dan dibawahnya dengan pendaratan yang baik,bukan sudah turun lalu naik kembali keatas..terimakasih untuk developer..

Jetrell, 2022-06-20:
@wawanlost Jangan lupa untuk membaca catatan rilis. Untuk melihat hal-hal yang telah diperbarui.
Detektor pendaratan telah diubah untuk penggunaan sayap tetap.. Dan ada beberapa perubahan filter yang mungkin memerlukan sedikit perubahan setelan filter Anda saat ini.
Selain itu, ada algoritme baru yang ditambahkan untuk memprediksi titik pemberhentian dan ketinggian dengan lebih baik dalam mode navigasi.
Saya sendiri telah menggunakannya dalam versi alfa selama 5 bulan terakhir dengan dua paha depan ukuran berbeda... Keduanya mendarat dengan sempurna seperti pada versi iNav sebelumnya.. Dan melakukan navigasi WP dan RTH tanpa masalah.
Tapi seperti yang saya katakan.. Anda mungkin perlu mengubah sedikit.. Bahkan kecepatan mendarat dan mendarat di tab Penyetelan lanjutan, mungkin perlu sedikit diubah.

wawanlost, 2022-06-20:
> @wawanlostJangan lupa untuk membaca catatan rilis. Untuk melihat hal-hal yang telah dilakukan. Detektor telah mengubah untuk penggunaan sayap dan ada beberapa perubahan filter yang mungkin memerlukan sedikit perubahan pengaturan filter Anda saat ini. Selain itu, ada algoritme baru yang ditambahkan untuk memprediksi titik pemberhentian dan ketinggian dengan lebih baik dalam mode navigasi. Saya sendiri telah menggunakannya dalam versi alfa selama 5 bulan terakhir dengan dua paha depan ukuran berbeda... Keduanya mendarat dengan sempurna seperti pada versi iNav sebelumnya.. Dan melakukan navigasi WP dan RTH tanpa masalah. Tapi seperti yang saya katakan.. Anda mungkin perlu mengubah sedikit.. Bahkan kecepatan mendarat dan mendarat di tab Penyetelan lanjutan, mungkin perlu sedikit diubah.

Terimakasih atas jawabannya @Jetrell  bisakah kamu mengirimkan diff pengaturan terbaik anda...saya hampir frustasi sejak inav 4 dan 5...saya menemukan titik buntu di penyetelan navigasi...penyetelan acro memang inav 5 jauh lebih halus...dibanding inav 4 dan 3 dan dibawah nya .tapi masalah saya mode navigasi saya berubah jadi sangat agresif dan jarang sekali bisa mendarat dan melucuti otomatis..saya pernah membalikkan ke inav 3.02 dan inav 2.61...navigasi berjalan mulus hingga auto disarm...dan 1lagi masalah saya blackbox tidak bisa digunakan sejak inav 4 hingga inav 5... terimakasih

Jetrell, 2022-06-20:
@wawanlost Menyalin Diff Semua tidak akan berfungsi dengan baik... Penyetelan adalah sesuatu yang khusus untuk ukuran paha depan, berat, dan daya dorong. Itu sebabnya menyetel kecepatan dan ketinggian pendaratan, harus sesuai dengan quad individu Anda.
Sudahkah Anda menyesuaikan pengaturan Pendaratan Otomatis?

Community-verified icon
[image]

Apakah Anda memiliki rilis log blackbox terbaru? Saya tidak punya masalah dengan logging. Juga tidak ada masalah yang dilaporkan di forum mengenai hal itu.

wawanlost, 2022-06-21:
terima kasih @Jetrell ternyata agresifnya quad di inav 4 dan 5 sudah kutemukan penyebabnya...saya menurunkan throtlle scala jadi 0.8 itu berjalan sempurna baik inav 4 dan 5...tetapi saya harus menurunkan rasio pid baro untuk memaksa quad melucuti saat sampai tanah(dan posisi ini goyah sekali saat angin kencang)dari 8kali test penerbangan dia berhasil melucuti motor 5 kali setelah mencapai tanah ..dan 2kali berhasil tapi membutuhkan waktu 1menit hingga melucuti motor dan 1 kali test tidak berhasil saya telah menunggu hingga 2 menit lebih dan quad tidaak mau melucuti....dan seperti biasa saya coba flash kembali ke inav 2.61 dan 3.02...idengan setttingan seperti ini berjalan dengan mulus tidak memiliki masalah dengan navigasi atau position hold...pahkan quad landing dengan sangat mulus(settingan default dan pid default inav)...entah perubahan apa yang ada pada inav 4 dan 5....terlebih untuk blackbox saya sampai saat ini belum bisa dipakai di inav 4 dan 5.....target matek f405 terima kasih

wawanlost, 2022-06-21:
> @wawanlost Menyalin Diff Semua tidak akan berfungsi dengan baik... Penyetelan adalah sesuatu yang khusus untuk ukuran paha depan, berat, dan daya dorong. Itu sebabnya menyetel kecepatan dan ketinggian pendaratan, harus sesuai dengan quad individu Anda. Sudahkah Anda menyesuaikan pengaturan Pendaratan Otomatis?
> 
> [image]
> 
> Apakah Anda memiliki rilis log blackbox terbaru? Saya tidak punya masalah dengan logging. Juga tidak ada masalah yang dilaporkan di forum mengenai hal itu.

balckbox saya tidak berfungsi dari semenjak inav 4 dan inav 5 pun juga tidak berfungsi...di inav3 dan inav 2.61 itu berjalan dengan baik...fc saya Hakrc f405 DM banggod... target inav matek f405

wawanlost, 2022-06-21:
> diff all
> 
> # version
> # INAV/MATEKF405 5.0.0 Jun 10 2022 / 09:11:07 ([dfa9f6f](https://github.com/iNavFlight/inav/commit/dfa9f6f632af5f54bbef90d1f6e7bdd7f655af55))
> # GCC-10.2.1 20201103 (release)
> # start the command batch
> batch start
> 
> # reset configuration to default settings
> defaults noreboot
> 
> # resources
> # mixer
> mmix reset
> 
> mmix 0 1.000 -1.000 1.000 -1.000 mmix 1 1.000 -1.000 -1.000 1.000 mmix 2 1.000 1.000 1.000 1.000 mmix 3 1.000 1.000 -1.000 -1.000
> 
> # servo mix
> # servo
> # safehome
> # logic
> logic 0 1 -1 4 1 10 0 0 0 logic 1 1 -1 5 1 10 0 0 0 logic 2 1 -1 6 1 10 0 0 0 logic 3 1 0 25 0 2 0 0 0 logic 4 1 1 25 0 3 0 0 0 logic 5 1 2 25 0 4 0 0 0
> 
> # gvar
> # pid
> # feature
> feature -TX_PROF_SEL feature GPS feature PWM_OUTPUT_ENABLE
> 
> # beeper
> # blackbox
> blackbox -NAV_ACC blackbox NAV_POS blackbox NAV_PID blackbox MAG blackbox ACC blackbox ATTI blackbox RC_DATA blackbox RC_COMMAND blackbox MOTORS
> 
> # map
> # serial
> serial 0 4096 115200 115200 0 115200 serial 1 0 115200 115200 0 115200 serial 2 262144 115200 115200 0 115200 serial 3 2 115200 57600 0 115200 serial 4 64 115200 115200 0 115200
> 
> # led
> # color
> # mode_color
> # aux
> aux 0 0 0 1675 2100 aux 1 1 1 900 1300 aux 2 2 1 1300 1700 aux 3 11 4 1700 2100 aux 4 10 2 1700 2100 aux 5 3 4 1300 1700 aux 6 19 6 1700 2100 aux 7 43 6 1300 1700 aux 8 13 2 1300 1700 aux 9 27 3 1700 2100
> 
> # adjrange
> # rxrange
> # temp_sensor
> # wp
> #wp 0 invalid
> 
> # osd_layout
> osd_layout 0 0 25 1 H osd_layout 0 1 23 11 V osd_layout 0 7 17 10 V osd_layout 0 8 7 13 V osd_layout 0 9 12 10 H osd_layout 0 10 10 1 V osd_layout 0 11 24 7 V osd_layout 0 12 23 9 V osd_layout 0 13 2 10 V osd_layout 0 14 24 2 V osd_layout 0 15 1 6 V osd_layout 0 19 24 8 V osd_layout 0 22 24 5 V osd_layout 0 23 23 6 V osd_layout 0 24 13 3 V osd_layout 0 25 20 6 H osd_layout 0 28 1 11 V osd_layout 0 30 1 12 V osd_layout 0 32 23 10 V osd_layout 0 33 9 10 V osd_layout 0 34 3 1 H osd_layout 0 36 11 5 H osd_layout 0 38 17 11 V osd_layout 0 40 23 3 V osd_layout 0 45 0 0 V osd_layout 0 48 22 11 H osd_layout 0 85 2 8 V osd_layout 0 86 1 3 V osd_layout 0 87 1 4 V osd_layout 0 105 18 1 H osd_layout 0 106 1 1 V osd_layout 0 107 1 2 V osd_layout 0 109 22 0 V osd_layout 0 110 24 1 V osd_layout 0 111 12 2 H osd_layout 0 112 15 0 H osd_layout 2 0 25 2 H osd_layout 2 1 18 0 V osd_layout 2 6 22 11 H osd_layout 2 7 12 12 V osd_layout 2 8 9 13 H osd_layout 2 10 18 11 V osd_layout 2 11 19 2 V osd_layout 2 12 18 1 V osd_layout 2 13 8 9 V osd_layout 2 14 24 1 V osd_layout 2 15 2 9 V osd_layout 2 19 14 1 V osd_layout 2 20 1 0 V osd_layout 2 21 1 2 V osd_layout 2 22 25 8 V osd_layout 2 23 23 9 V osd_layout 2 24 13 5 V osd_layout 2 28 1 12 V osd_layout 2 29 1 11 H osd_layout 2 30 2 10 V osd_layout 2 31 24 0 V osd_layout 2 32 13 0 V osd_layout 2 33 7 11 V osd_layout 2 34 11 7 V osd_layout 2 37 21 5 H osd_layout 2 38 13 11 V osd_layout 2 40 23 7 V osd_layout 2 45 0 0 V osd_layout 2 86 1 6 V osd_layout 2 87 1 6 H osd_layout 2 105 26 11 V osd_layout 2 106 1 3 V osd_layout 2 107 1 5 V osd_layout 2 108 10 4 H osd_layout 2 110 24 2 V
> 
> # master
> set looptime = 500 set gyro_main_lpf_hz = 110 set gyro_main_lpf_type = PT1 set dynamic_gyro_notch_q = 150 set dynamic_gyro_notch_min_hz = 60 set setpoint_kalman_q = 200 set gyro_zero_x = -81 set gyro_zero_y = -4 set gyro_zero_z = -3 set ins_gravity_cmss = 989.316 set acc_hardware = MPU6000 set acczero_x = 62 set acczero_y = 1 set acczero_z = 241 set accgain_x = 4083 set accgain_y = 4091 set accgain_z = 4072 set align_mag = CW270 set mag_hardware = HMC5883 set magzero_x = -72 set magzero_y = -73 set magzero_z = -590 set maggain_x = 469 set maggain_y = 441 set maggain_z = 354 set baro_hardware = BMP280 set min_check = 1050 set max_check = 1600 set serialrx_provider = CRSF set blackbox_rate_denom = 2 set max_throttle = 1550 set motor_pwm_protocol = DSHOT600 set failsafe_delay = 3 set failsafe_procedure = RTH set failsafe_min_distance_procedure = RTH set align_board_roll = 1804 set align_board_pitch = 2 set current_meter_scale = 350 set current_meter_offset = 30 set model_preview_type = 3 set switch_disarm_delay = 50 set applied_defaults = 2 set rpm_gyro_min_hz = 70 set gps_provider = UBLOX7 set gps_sbas_mode = AUTO set gps_ublox_use_galileo = ON set airmode_type = THROTTLE_THRESHOLD set inav_reset_altitude = EACH_ARM set inav_reset_home = EACH_ARM set inav_w_z_baro_p = 0.350 set nav_disarm_on_landing = ON set nav_extra_arming_safety = OFF set nav_auto_speed = 650 set nav_auto_climb_rate = 200 set nav_manual_speed = 250 set nav_land_minalt_vspd = 80 set nav_land_slowdown_minalt = 800 set nav_land_slowdown_maxalt = 1500 set nav_emerg_landing_speed = 100 set nav_min_rth_distance = 1 set nav_rth_allow_landing = FS_ONLY set nav_rth_abort_threshold = 5000 set nav_rth_altitude = 2500 set safehome_max_distance = 1 set nav_mc_bank_angle = 32 set nav_mc_auto_disarm_delay = 500 set osd_left_sidebar_scroll = ALTITUDE set osd_right_sidebar_scroll = HOME_DISTANCE set osd_sidebar_scroll_arrows = ON set osd_esc_rpm_precision = 5 set i2c_speed = 800KHZ set name = Senma LongRange set vtx_band = 3 set vtx_channel = 3 set vtx_power = 3 set vtx_low_power_disarm = ON set dshot_beeper_enabled = OFF
> 
> # profile
> profile 1
> 
> set mc_p_pitch = 44 set mc_i_pitch = 60 set mc_d_pitch = 25 set mc_i_roll = 50 set mc_d_roll = 25 set mc_p_yaw = 45 set mc_i_yaw = 70 set max_angle_inclination_rll = 200 set dterm_lpf_hz = 90 set dterm_lpf_type =  [... cut ...] rate = 70 set pitch_rate = 70 set yaw_rate = 60
> 
> # battery_profile
> battery_profile 1
> 
> set vbat_cell_detect_voltage = 435 set vbat_max_cell_voltage = 430 set throttle_scale = 0.800 set throttle_idle = 10.000 set nav_mc_hover_thr = 1310
> 
> # battery_profile
> battery_profile 2
> 
> # battery_profile
> battery_profile 3
> 
> # restore original profile selection
> profile 1 battery_profile 1
> 
> # save configuration
> save

mungkin ini adalah setting terbaik inav 5 untuk saya saat ini..

z-renat, 2023-01-14:
Hi.
For a multirotor, when returning home, is it possible to make not a sharp, but a smooth set of speed at the beginning. And a smooth speed reset when reaching the home point before descending? The copter flies over the home point all the time, swings. The return speed is only 30km/h.
Machine translation, I'm sorry.

closed 2024-03-20 by b14ckyy