iNavFlight/inav #8133: 5.0 RC2 ESC telemetry Temp reading not working

0crap, 2022-06-18:
## Current Behavior

[image]
ESC temp reading stays at 0 degrees Celcius

## Steps to Reproduce

1. connect telemetry wire to FC
2. add ESC temp reading to OSD 
3. ESC temp is stuck at 0 degrees Celcius
4. ESC RPM counter works fine (So ESC to FC connection is confirmed to be correct.)

## Expected behavior

ESC temp reading gives correct values on OSD

## Suggested solution(s)

Ask kindly to look into this.

## Additional context

This is a new build, so I have no clue if this ever worked on older iNav releases,
Used components:
- Matek [H743-WING V2](http://www.mateksys.com/?portfolio=h743-wing-v2) flight controller
- Holybro [Tekko32 F4 45A](http://www.holybro.com/product/tekko32-f4-45a-esc/) BLHeli32 ESC
The FC is flashed with iNav 5.0 RC2
The ESC is flashed with BLHeli 32 v32.9 

---

- FC Board name and vendor: MATEK
- INAV version string: 
```
# version
# INAV/MATEKH743 5.0.0 Jun 10 2022 / 09:38:43 (dfa9f6f6)
# GCC-10.2.1 20201103 (release)
```

MrD-RC, 2022-06-18:
Working fine on my 5.0 install.
- Matek F405-STD
- iFlight SucceX-E Slick 45A ESC (no idea which version of BLHeli_32)

[image]

erstec, 2022-06-18:
Same. Works fine on H743-SLIM + T-motor F55A PRO

0crap, 2022-06-18:
Which BLHeli32 version did you use?

erstec, 2022-06-18:
> Which BLHeli32 version did you use?

32.8 I believe, need to check when will back home after weekend

0crap, 2022-06-19:
I flashed the ESC to v32.8 and it gives the same issue, ESC temp stays at 0 degrees C.

Mateyhv, 2022-06-19:
Matek F405 Wing - v5.0.0 RC3 with Tekko32 F3 35A v32.8 here. Reporting temp and RPM just fine, no difference since 4.1, 4.0, etc.

DzikuVx, 2022-06-20:
tested with BLHeli 32.9. multiple FCs and INAV 5. All the time wors like expected

closed 2022-06-20 by DzikuVx

0crap, 2022-06-20:
> tested with BLHeli 32.9. multiple FCs and INAV 5. All the time wors like expected

It does not work "all the time", that's why I reported it.
Anyway in the link above, to the BLHeli32 Github issue, you can read why this happens.
In short, this case is not iNav related.

Thx to all that replied to this issue.