iNavFlight/inav #8138: Omnibus f4 v3 plus with gyroscope bmi270

LexResident, 2022-06-19:
Hi, I have a Aliexpress F4 V3 Plus with gyroscope bmi270 or bmi160 (can't define exactly).
On BETAFLIGHT the board is called OmnibusF4SD, but INAV do not have such a board. The gyroscope is not detected on the INAV OMNIBUSF4V3 (OMNIBUS F4, OMNIBUSF4PRO and etc) firmware. Is there any board I could choose to flash INAV on it to make everything work.

Sincerely, DENIS.

julianschweizer, 2022-06-21:
Looks like that the OMNIBUSF4V3 target.h [OMNIBUSF4V3](https://github.com/iNavFlight/inav/blob/master/src/main/target/OMNIBUSF4/target.h) uses the MPU6000. 

I recommend building your own INAV version based on the OMNIBUSF4V3 and looking/copying from betaflight target by adding/trying the BMI270 or BMI160 driver. Its pretty straight forward with Visual Studio Code or WSL. See: https://github.com/iNavFlight/inav/tree/master/docs/development

I had the same issue with some diatone FC and did the same.

julianschweizer, 2022-06-21:
Well, the Ali listing says "Betaflight only" .... .

You can try to compile your own firmware. I did this with 4.1 and the following targets for you. Give it a shot. You may need to modify it further. Please read https://github.com/iNavFlight/inav/tree/master/docs/development

[target OMNIBUSF4.zip](https://github.com/iNavFlight/inav/files/8948739/target.OMNIBUSF4.zip)
[inav_4.1.0_OMNIBUSF4V3.zip](https://github.com/iNavFlight/inav/files/8948742/inav_4.1.0_OMNIBUSF4V3.zip)

LexResident, 2022-06-22:
With this firmware, the accelerometer is also not detected.
I will understand and compile for my flight controller.
Please recommend a good guide for this.

JulioCesarMatias, 2022-07-02:
We really need to add this support... Since the MPU-6000 is no longer produced, it has now been replaced by the BMI270

Anandan1963, 2022-07-23:
for me also same issue.
inav is not showing gyro,but showing accelerometer,baro.
please help to solve this issue

JulioCesarMatias, 2022-08-15:
I was looking at this just now, I made some modifications and I have a build for you guys to test... the firmware was compiled in version 6.0, you need to download configurator 6.0 here >> http://seyrsnys.myzen.co.uk/inav-configurator-next/
[inav_6.0.0_OMNIBUSF4V3.zip](https://github.com/iNavFlight/inav/files/9340532/inav_6.0.0_OMNIBUSF4V3.zip)

LexResident, 2022-08-23:
Thank you very much! On this firmware BMI270 works! I didn't even expect to be helped. Thank you!

LexResident, 2022-08-23:
Many will be happy if BMI270 support is added to permanent support.

Zileg72, 2022-08-23:
Hi JulioCesarMatias!
It's working! Good work! 
Thank you!!!

JulioCesarMatias, 2022-08-23:
> Many will be happy if BMI270 support is added to permanent support.

I opened a PR, now it's waiting for Pawel to approve and merge.

closed 2022-09-03 by DzikuVx

wawanlost, 2022-11-17:
saya tidak dapat menemukan barometer pada papan ini...saya sudah memflash inav 6.0...itu hanya mendapatkan accelerometer..baro tidak terbaca...apakah kesalahan ada pada hardware..?...padahal saya baru saja membelinya

DzikuVx, 2022-11-17:
Niestety nie rozumem o czym piszesz. A szkoda ...

godi182, 2022-12-31:
I have the same problem . gyroscopio works in betafligbetaflight but it is not recognized in inav . please help me . fc matek f405 std bmi270 giroscope
[image]

parahromis, 2023-07-03:
[image]
Hello. I checked several versions - Inav, Clean Flight, Betaflight, the magnetometer does not work. Accelerometer, gyroscope, barometer - work without problems.

Brynod, 2023-07-09:
> [image] Hello. I checked several versions - Inav, Clean Flight, Betaflight, the magnetometer does not work. Accelerometer, gyroscope, barometer - work without problems.

I had this problem.... Try changing I2C speed to 200khz,the software will that this is too slow but this is the only way I could get compass to work.
Hope this helps

shrinibas7, 2025-03-21:
> I was looking at this just now, I made some modifications and I have a build for you guys to test... the firmware was compiled in version 6.0, you need to download configurator 6.0 here >> http://seyrsnys.myzen.co.uk/inav-configurator-next/ Firmware >> [inav_6.0.0_OMNIBUSF4V3.zip](https://github.com/iNavFlight/inav/files/9340532/inav_6.0.0_OMNIBUSF4V3.zip)

Hey .. i was thinking can you make the firmware for ardupilot in ardupilot bmi270also not working