iNavFlight/inav #8147: INAV 5.0.0 RC3 MATEK F411-WTE is freezing and resets config to defaults

julianschweizer, 2022-06-22:
## Current Behavior
MATEK F411-WTE Flight Controller is randomly "freezing/unresponsive" in configurator 5.0.0 RC2 with battery / USB plugged in. Reboot causes configuration reset. This happened with 2 different units. 

Other: Semms like FC is rebooting sometimes while switching configurator tabs (espeacially OSD) or after pluging in the battery. I suspect the GPS via softserial.

## Steps to Reproduce
Random ocurrance while configuring new flashed hardware with USB and Battery connected.

## Additional context
- Wireing of hardware to fc attached
- DIFF All attached

- FC Board name and vendor: INAV/MATEKF411TE
- INAV version string: 5.0.0 Jun 10 2022 / 09:25:02 (dfa9f6f6)

[Anschlussplan_Hee_Wing_F01.pptx](https://github.com/iNavFlight/inav/files/8955427/Anschlussplan_Hee_Wing_F01.pptx)
[INAV_cli_20220622_094416_DIFF_ALL.txt](https://github.com/iNavFlight/inav/files/8955522/INAV_cli_20220622_094416_DIFF_ALL.txt)

AndreaCCIE, 2022-06-22:
Mine doesn't freeze, but I'm on analog and I have another issue (#8143). Does it happen even with the DJI unit disconnected?

closed 2022-06-23 by julianschweizer

0crap, 2022-06-23:
@julianschweizer It helps others if you write down why you closed the issue.

IvanBiv, 2022-07-09:
Similar problem on Matek F411se.
Firmware INAV 5.0.0.
Playback:
1) I connect the USB cable, configure in INAV Configurator, save.
2) I click Disconnect in INAV Configurator.
3) Connect the power.
4) Through the OSD, I see what is freezed on the INAV logo at the time of loading.
5) When I turn off / turn on the power on the OSD, I see indicators, the device has booted up, but the elements on the OSD are not mine and not in my order. I connect to INAV Configurator, it offers to configure the device, i.e. configuration is reset.

ManuelWoessner, 2022-09-20:
Got the same problem with my FC. Luckily I had my Laptop with me in the field to restore the configuration. But I never had such a behaviour with my other drones FCs. Since my FixWing is uncontrollable with the default configuration, I hope this will not happen in midair, a crash is then inevitable... That gives me a really bad feeling. It runs INAV 5.1 firmware. I couldnt find information about a weak EEPROM etc of the used STM32F411CE microcontroller. If there is a solution to this issue, sharing it with me will be greatly appreciated.

IvanBiv, 2022-09-20:
> 

I use 4.x.x version now

ManuelWoessner, 2022-09-20:
> > 
> 
> I use 4.x.x version now

Thx for the fast response =)
And this behaviour never occured again? Then I am going to downgrade the firmware, too.

IvanBiv, 2022-09-20:
I did not take risks, immediately after the incident (it’s good that I didn’t lose the plane), I returned it to the old version.  I will try to work with the ardupilot, the inav horizon is problematic, which means you can’t rely on the Inav autopilot.

IvanBiv, 2022-09-20:
> > > 
> > 
> > 
> > I use 4.x.x version now
> 
> Thx for the fast response =) And this behaviour never occured again? Then I am going to downgrade the firmware, too.

My answer was deleted here. I will repeat the answer in other words.
No, I did not try again, I rolled back the lead.  High risk of recurrence.  Lucky I didn't lose the plane.  I have no complaints about the developers.  We choose what to use.