iNavFlight/inav #8148: Fixed wing heading status not resetting after landing disarm

breadoven, 2022-06-22:
Heading status remains trusted even after a fixed wing has landed and been disarmed which doesn't make sense. GPS CoG is no longer valid so Heading status can't be trusted and should be reset. The following change would fix this in imu.c (https://github.com/iNavFlight/inav/blob/3cb8e80c96dcb5380666e8b1a0644454a75f8e3a/src/main/flight/imu.c#L555):

```
        if (canUseMAG) {
            useMag = true;
            gpsHeadingInitialized = true;   // GPS heading initialised from MAG, continue on GPS if compass fails
        }
        else if (canUseCOG) {
            if (gpsHeadingInitialized) {
                courseOverGround = DECIDEGREES_TO_RADIANS(gpsSol.groundCourse);
                useCOG = true;
            }
            else {
                // Re-initialize quaternion from known Roll, Pitch and GPS heading
                imuComputeQuaternionFromRPY(attitude.values.roll, attitude.values.pitch, gpsSol.groundCourse);
                gpsHeadingInitialized = true;

                // Force reset of heading hold target
                resetHeadingHoldTarget(DECIDEGREES_TO_DEGREES(attitude.values.yaw));
            }
        } else if (!ARMING_FLAG(ARMED)) { 
            gpsHeadingInitialized = false;
        }
```

Or is there some reason for not resetting `gpsHeadingInitialized  `in these circumstances ?

Jetrell, 2022-06-24:
>Or is there some reason for not resetting `gpsHeadingInitialized` in these circumstances ?

The main case I can think of, is if a fixedwing with undercarriage, was to make a quick landing (for a goggle battery change).. and disarm.
Once re-armed, the planes GPS heading would instantly show on the OSD at _minimal_ ground speed, while its taxing back to the flight-line.

There are other setting that that work in a similar way - 
`inav_reset_altitude `= Never, First_arm, Each_arm
`inav_reset_home` = Never, First_arm, Each_arm

I suppose the question is.. Has there been a case of the un-trusted heading, not being instantly trusted again after rearming, once the plane is _moved_ for re-launch?

Jetrell, 2022-06-24:
> Or is there some reason for not resetting gpsHeadingInitialized in these circumstances ?

The main case I can think of... is if a fixedwing with undercarriage, was to make a quick landing (for a goggle battery change).. and disarmed.
Once re-armed, the planes GPS heading would instantly show on the OSD at minimal ground speed, while its taxing back to the flight-line.

I suppose the question is.. Has there been a case of the un-trusted heading, not being instantly trusted again after rearming, once the plane is moved for re-launch?

There are other user setting that can be made work in a similar way -
`inav_reset_altitude` = Never, First_arm, Each_arm
`inav_reset_home` = Never, First_arm, Each_arm

breadoven, 2022-06-24:
The reason I noticed this was because of PR #8134 and the fact heading status remained Trusted even after landing and disarming which affects the fixed wing flying status flag if you launch again without rebooting.

Not sure why this is the way it is. I guess you shouldn't lose heading status unless the GPS fails in which case things will happen, i.e. emergency landing, because GPS position is no longer trusted. The only time you could have a valid GPS position but not a reliable course over ground (and therefore heading) if ground speed drops below 3 m/s (strong headwind). Is it better to ignore a dubious heading in this case and carry on or do something to mitigate it ? I guess the only other option is to land or fly manually ... creeping forward into wind.

`gpsHeadingInitialized `should probably always be set `false `whenever there is no valid GPS heading, not just when disarmed.

RTH and WP mode both use heading status to trigger an emergency landing but I can't see it ever triggering in flight once heading status has initially been set as Trusted so seems a bit pointless including it.

Jetrell, 2022-06-25:
> 
> Not sure why this is the way it is. I guess you shouldn't lose heading status unless the GPS fails in which case things will happen.

Today when I was testing the manual throttle auto launch. I landed and disarmed 5 times in one flight.  
Each time the heading on the OSD was perfectly fine (from known markers) after re-launch.... I also ran some RTH tests after some of these rearms, just to be sure that the GPS heading was correct for the navigation use.. I didn't experience any problems.

breadoven, 2022-06-25:
If the GPS is working then INAV uses the GPS course over ground once speed is > 3 m/s and the heading will quickly align with this after launch so you won't see any heading errors given this happens in < 1s.

I'll just do a PR for this to tidy it up. I'm sure someone will question it if it causes problems.

Jetrell, 2022-06-26:
>The only time you could have a valid GPS position but not a reliable course over ground (and therefore heading) is if ground speed drops below 3 m/s (strong headwind). Is it better to ignore a dubious heading in this case and carry on or do something to mitigate it ? 

I guess the reason this doesn't often happen, is due to `nav_fw_min_vel_speed_boost`, being coded to 7m/s. And boosting the `cruise_thr` to `max_thr`.
It would either be a large gale, a poorly setup plane.. Or very under powered.

MrD-RC, 2022-06-26:
Certainly when navigation modes are involved. I've flown backwards before, intentionally. But would always increase speed to forward flight before doing anything else. 

Another example is a low powered plane that even at 100% throttle cannot maintain forward flight against the wind. This would be poor planning in most cases. But sometimes the wind is surprising.

closed 2022-07-01 by DzikuVx