iNavFlight/inav #8163: RTHP: Returning home by retracing the path.

sdellava, 2022-06-26:
## Current Behavior
The RTH proceudre make the aircraft fly straigh to the home point.

## Desired Behavior
Allow the aircraft to revere fly the path from the homepoint to the lost singal point.

## Suggested Solution
The RTH procedure makes the aircraft fly in a straight line toward the takeoff point.
Depending on the terrain geography or simply the presence of buildings, the aircraft risks crashing and being lost.

Even by specifying a minimum height to be reached before returning in a straight line, it is not certain that the specified altitude will be reached quickly enough.

A very simple solution is to have the FC fly back the same path, and with INAV using missions, it is very easy to accomplish this feature:
- while the aircraft is in flight the FC stores GPS points in a special mission with a predefined frequency (e.g., once per second). 
- At the time when RTH needs to be activated, simply the FC activates this mission by following the points in reverse order.

The aircraft will retrace its path backward shortly reaching a position where radio contact is possible. From there the pilot will be able to regain control as he normally does with the RTH procedure avoiding any risk of collision with obstacles in the environment.

## Who does this impact? Who is this for?
Every INAV users

stronnag, 2022-06-26:
Duplicate of #7988

closed 2022-06-26 by stronnag

sdellava, 2022-06-26:
tnx!