iNavFlight/inav #8190: Rx Link Loss with current master build

breadoven, 2022-07-03:
Updated my custom firmware yesterday to the master version and now no Rx link on 2 boards that have been flashed, HGLRCF722 and FLYWOOF411 targets. The ELRS Rx seems to bind (LED steady) and there's basic telemetry but otherwise no Rx link. INAV is in Failsafe with the RX LINK system message showing. Status in the CLI doesn't tell you any more. Serial port and Receiver setting are as before. Another board that hasn't been updated works fine so it's not the transmitter. The working board was last updated from master on the 28/6/22 so the problem is caused by something from that date. The next commit after that master update appears to be related to https://github.com/iNavFlight/inav/pull/8176.

I've not changed anything related to the Rx in my custom changes so it would appear to be something in the current master. I noticed there are various changes to Rx related stuff. Any idea what's broken here or is there some settings that needs updating ?

JulioCesarMatias, 2022-07-03:
looks like @robustini is having the same problem...

breadoven, 2022-07-03:
Reverted the last master merge and all is working OK again ... so something merged since https://github.com/iNavFlight/inav/pull/8176 has broken the Rx link.

OptimusTi, 2022-07-03:
Latest master working for me. No issues with rxlink

closed 2022-07-04 by DzikuVx