iNavFlight/inav #8196: Matek F405-AIO can't see AUTOLEVEL mode

dsaboot, 2022-07-04:
## Current Behavior

Matek F405-AIO board. Flashed with INAV V5.
Continuously trim servos  enabled.
Cant' see in MODES - AUTOLEVEL

[image]
[image]

## Steps to Reproduce

1. Flash MATEK F405-AIO with MATEKF405 firmware V5
2. go to MODES tab
3. no "AUTOLEVEL" mode in list
4.

## Expected behavior

AUTOLEVEL mode in list available

## Suggested solution(s)

## Additional context

---

# version
# INAV/MATEKF405 5.0.0 Jun 20 2022 / 10:56:14 (d2adf35c)
# GCC-10.2.1 20201103 (release)

# start the command batch
batch start

# resources

# mixer

mmix reset

mmix 0  1.000  0.000  0.000  0.000
mmix 1  1.000  0.000  0.000  0.000

# servo mix
smix reset

smix 0 1 0 50 0 -1
smix 1 1 1 50 0 -1
smix 2 2 0 -50 0 -1
smix 3 2 1 50 0 -1

# servo
servo 1 1000 2000 1468 -100
servo 2 1000 2000 1448 100
servo 3 1000 2000 1500 -100

# safehome

# logic

# gvar

# pid

# feature
feature -TX_PROF_SEL
feature THR_VBAT_COMP
feature MOTOR_STOP
feature GPS
feature BLACKBOX
feature PWM_OUTPUT_ENABLE
feature FW_LAUNCH
feature FW_AUTOTRIM

# beeper
beeper -ON_USB

# blackbox
blackbox -NAV_ACC
blackbox NAV_POS
blackbox NAV_PID
blackbox MAG
blackbox ACC
blackbox ATTI
blackbox RC_DATA
blackbox RC_COMMAND
blackbox MOTORS

# map

# serial
serial 2 2097152 115200 115200 0 115200
serial 3 256 115200 115200 0 115200
serial 4 2 115200 115200 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1600 2100
aux 1 1 1 1275 1375
aux 2 12 1 1100 1225
aux 3 53 1 1775 1875
aux 4 11 1 1625 1725
aux 5 10 1 1950 2075
aux 6 28 1 1450 1550
aux 7 3 1 1775 1875
aux 8 21 3 1700 2100
aux 9 13 8 1600 2100

# adjrange

# rxrange

# temp_sensor

# wp
#wp 0 invalid

# osd_layout
osd_layout 0 0 14 1 V
osd_layout 0 1 1 11 V
osd_layout 0 3 8 6 V
osd_layout 0 7 12 15 V
osd_layout 0 8 14 11 V
osd_layout 0 9 1 13 H
osd_layout 0 11 1 14 V
osd_layout 0 12 1 13 V
osd_layout 0 13 24 7 V
osd_layout 0 14 17 1 V
osd_layout 0 15 1 7 V
osd_layout 0 20 1 2 V
osd_layout 0 21 1 1 V
osd_layout 0 22 10 15 V
osd_layout 0 23 24 6 V
osd_layout 0 24 23 1 V
osd_layout 0 25 5 5 H
osd_layout 0 26 1 8 V
osd_layout 0 28 22 14 V
osd_layout 0 30 0 12 V
osd_layout 0 31 12 5 H
osd_layout 0 32 1 12 V
osd_layout 0 33 1 13 H
osd_layout 0 34 11 2 H
osd_layout 0 35 16 13 V
osd_layout 0 38 9 14 H
osd_layout 0 40 23 13 V
osd_layout 0 41 1 3 V
osd_layout 0 42 1 4 V
osd_layout 0 45 0 0 V
osd_layout 0 46 0 8 H
osd_layout 0 47 23 10 H
osd_layout 0 53 9 13 V
osd_layout 0 85 24 7 H
osd_layout 0 86 1 15 V
osd_layout 0 87 0 15 H
osd_layout 0 88 1 6 H
osd_layout 0 107 1 5 V
osd_layout 1 0 14 1 V
osd_layout 1 1 2 1 V
osd_layout 1 3 8 6 V
osd_layout 1 7 1 3 H
osd_layout 1 9 1 13 H
osd_layout 1 11 1 14 H
osd_layout 1 12 24 14 H
osd_layout 1 13 25 5 V
osd_layout 1 14 18 1 H
osd_layout 1 15 24 2 V
osd_layout 1 16 2 9 V
osd_layout 1 17 2 10 V
osd_layout 1 20 1 2 H
osd_layout 1 21 1 1 H
osd_layout 1 22 23 1 H
osd_layout 1 23 24 1 V
osd_layout 1 24 13 1 H
osd_layout 1 26 2 8 H
osd_layout 1 28 22 14 H
osd_layout 1 30 1 3 V
osd_layout 1 31 21 4 H
osd_layout 1 32 1 12 H
osd_layout 1 33 1 13 H
osd_layout 1 34 11 2 H
osd_layout 1 35 16 13 H
osd_layout 1 38 2 2 V
osd_layout 1 40 23 13 H
osd_layout 1 41 2 5 V
osd_layout 1 42 2 4 H
osd_layout 1 45 0 0 V
osd_layout 1 46 23 8 H
osd_layout 1 47 23 10 H
osd_layout 1 62 2 8 H
osd_layout 1 63 2 8 V
osd_layout 1 67 2 12 V
osd_layout 1 68 2 13 V
osd_layout 1 85 22 7 H
osd_layout 1 86 1 4 H
osd_layout 1 87 0 10 H
osd_layout 2 0 14 1 V
osd_layout 2 1 1 13 V
osd_layout 2 3 8 6 V
osd_layout 2 7 1 3 H
osd_layout 2 9 1 13 H
osd_layout 2 11 1 14 V
osd_layout 2 12 24 14 V
osd_layout 2 13 23 7 V
osd_layout 2 14 1 2 V
osd_layout 2 15 2 7 V
osd_layout 2 16 2 10 V
osd_layout 2 17 2 11 V
osd_layout 2 18 2 12 V
osd_layout 2 20 19 1 V
osd_layout 2 21 1 1 V
osd_layout 2 22 23 2 V
osd_layout 2 23 24 2 V
osd_layout 2 24 13 1 H
osd_layout 2 26 2 8 H
osd_layout 2 28 22 14 H
osd_layout 2 30 1 4 V
osd_layout 2 31 21 4 H
osd_layout 2 32 1 12 H
osd_layout 2 33 1 13 H
osd_layout 2 34 11 2 H
osd_layout 2 35 16 13 H
osd_layout 2 38 25 13 V
osd_layout 2 40 23 13 H
osd_layout 2 44 0 0 V
osd_layout 2 45 0 0 V
osd_layout 2 46 23 9 H
osd_layout 2 47 23 10 H
osd_layout 2 85 22 7 H
osd_layout 2 86 1 4 H
osd_layout 2 87 0 10 H
osd_layout 3 0 2 2 V
osd_layout 3 1 24 10 V
osd_layout 3 7 14 1 V
osd_layout 3 9 24 6 V
osd_layout 3 10 2 1 V
osd_layout 3 11 25 8 V
osd_layout 3 15 1 7 V
osd_layout 3 16 7 2 V
osd_layout 3 17 7 3 V
osd_layout 3 18 7 4 V
osd_layout 3 19 25 7 V
osd_layout 3 26 1 8 V
osd_layout 3 27 0 5 V
osd_layout 3 28 23 1 V
osd_layout 3 30 1 12 V
osd_layout 3 31 24 3 H
osd_layout 3 32 24 9 V
osd_layout 3 33 24 5 V
osd_layout 3 41 1 9 V
osd_layout 3 42 1 10 V
osd_layout 3 56 7 10 V
osd_layout 3 57 7 11 V
osd_layout 3 58 7 11 H
osd_layout 3 61 16 6 H
osd_layout 3 62 16 6 H
osd_layout 3 63 7 6 V
osd_layout 3 67 2 13 V
osd_layout 3 68 2 14 V
osd_layout 3 75 7 5 V
osd_layout 3 76 16 5 V
osd_layout 3 77 16 6 V
osd_layout 3 85 1 6 V
osd_layout 3 108 7 4 H
osd_layout 3 109 7 7 V
osd_layout 3 110 22 11 H
osd_layout 3 111 23 9 H

# master
set gyro_main_lpf_hz = 25
set dynamic_gyro_notch_q = 250
set dynamic_gyro_notch_min_hz = 30
set gyro_zero_x = 10
set gyro_zero_y = -3
set gyro_zero_z = -7
set ins_gravity_cmss =  978.841
set acc_hardware = MPU6500
set acczero_x = -8
set acczero_y = -11
set acczero_z = -50
set accgain_x = 4088
set accgain_y = 4084
set accgain_z = 4071
set align_mag = CW270FLIP
set mag_hardware = NONE
set baro_hardware = BMP280
set max_throttle = 1950
set motor_pwm_protocol = STANDARD
set failsafe_procedure = RTH
set bat_voltage_src = SAG_COMP
set platform_type = AIRPLANE
set model_preview_type = 8
set imu_dcm_ki = 0
set small_angle = 180
set imu_acc_ignore_rate = 8
set imu_acc_ignore_slope = 3
set disarm_kill_switch = OFF
set applied_defaults = 3
set gps_provider = UBLOX7
set gps_sbas_mode = EGNOS
set airmode_type = STICK_CENTER_ONCE
set nav_extra_arming_safety = ALLOW_BYPASS
set nav_wp_radius = 1500
set nav_manual_climb_rate = 800
set nav_land_slowdown_minalt = 800
set nav_land_slowdown_maxalt = 2500
set nav_emerg_landing_speed = 200
set nav_min_rth_distance = 0
set nav_rth_allow_landing = FS_ONLY
set nav_rth_alt_mode = AT_LEAST_LINEAR_DESCENT
set nav_rth_abort_threshold = 60000
set nav_rth_altitude = 5000
set nav_fw_bank_angle = 45
set nav_fw_climb_angle = 30
set nav_fw_dive_angle = 30
set nav_fw_pitch2thr_smoothing = 3
set nav_fw_loiter_radius = 6000
set nav_fw_cruise_speed = 1400
set nav_fw_control_smoothness = 3
set nav_fw_launch_velocity = 250
set nav_fw_launch_accel = 1500
set nav_fw_launch_detect_time = 30
set nav_fw_launch_motor_delay = 120
set nav_fw_launch_spinup_time = 150
set nav_fw_launch_timeout = 8000
set nav_fw_launch_climb_angle = 25
set nav_fw_allow_manual_thr_increase = ON
set frsky_pitch_roll = ON
set smartport_fuel_unit = PERCENT
set mavlink_version = 1
set osd_rssi_alarm = 40
set osd_time_alarm = 30
set osd_alt_alarm = 2000
set osd_dist_alarm = 5000
set osd_current_alarm = 20
set osd_crosshairs_style = TYPE3
set osd_sidebar_scroll_arrows = ON
set osd_speed_source = 3D
set i2c_speed = 800KHZ
set name = :WDTE
set dji_esc_temp_source = IMU

# profile
profile 1

set fw_p_pitch = 15
set fw_i_pitch = 5
set fw_d_pitch = 5
set fw_ff_pitch = 172
set fw_p_roll = 15
set fw_i_roll = 3
set fw_d_roll = 7
set fw_ff_roll = 67
set fw_p_yaw = 20
set fw_i_yaw = 0
set fw_ff_yaw = 100
set max_angle_inclination_rll = 550
set dterm_lpf_hz = 10
set fw_turn_assist_pitch_gain =  0.200
set nav_fw_pos_z_p = 15
set nav_fw_pos_z_d = 5
set nav_fw_pos_xy_p = 60
set d_boost_min =  1.000
set d_boost_max =  1.000
set fw_level_pitch_trim =  3.657
set rc_expo = 30
set rc_yaw_expo = 30
set roll_rate = 19
set pitch_rate = 7
set yaw_rate = 3

# battery_profile
battery_profile 1

set bat_cells = 4
set vbat_min_cell_voltage = 270
set vbat_warning_cell_voltage = 320
set battery_capacity = 4600
set battery_capacity_warning = 920
set battery_capacity_critical = 460
set throttle_idle =  5.000
set nav_fw_cruise_thr = 1450
set nav_fw_min_thr = 1100
set nav_fw_max_thr = 2000
set nav_fw_pitch2thr = 13
set nav_fw_launch_thr = 1950

# end the command batch
batch end

#

MrD-RC, 2022-07-04:
That flight controller doesn't have a barometer. Barometer is required for AUTO LEVEL.

dsaboot, 2022-07-04:
Barometer connected to I2C bus and work.

dsaboot, 2022-07-04:
> That flight controller doesn't have a barometer. Barometer is required for AUTO LEVEL.

Barometer connected to I2C bus and work.

"set baro_hardware = BMP280"

MrD-RC, 2022-07-04:
So the baro symbol at the top of configurator is blue?

MrD-RC, 2022-07-04:
Also autolevel will be lower down the page in the tuning modes section.

closed 2022-09-03 by DzikuVx