iNavFlight/inav #8203: FuriousFPV F-35 No Telememety INAV 5.0

Basti77, 2022-07-09:
Hello.
After updating from 2.2.1 (where everything was working) I am getting
the telemetry connection does not work. 

Before the update I made a backup of the old config. 
and made screenshots of all menus of the old Invav Configurator, because I knew that I have to maintain all settings manually and cannot simply import them.
+Fly the new PIDs.

As far as I can judge on the ground, everything works except telemetry.

The cabling has not been changed. 

Remote control.
XLite (old model) with R9

Receiver: R9-Slim Plus 

is there another trick?

INAV
Current build 
5.0 FF35

[S800-5.0.txt](https://github.com/iNavFlight/inav/files/9077528/S800-5.0.txt)
[S800-2.2.1.txt](https://github.com/iNavFlight/inav/files/9077529/S800-2.2.1.txt)

Basti77, 2022-07-09:
After a lot of trial and error, I found the solution:

set telemetry_halfduplex = OFF
shouldn't that be set as the default for this board?

closed 2022-07-09 by Basti77

MrD-RC, 2022-07-09:
No. Default should be auto. That being on or off depends on the receiver. Not the flight controller.