iNavFlight/inav #8207: Magnetometer not detected on INAV/FLYWOOF745NANO 5.0.0

irom77, 2022-07-11:
## Current Behavior

Getting Sensor status: MAG=UNAVAILABLE on INAV/FLYWOOF745NANO 5.0.0 and I2C Errors: 60

## Steps to Reproduce

1. Connect quad FC to USB

## Expected behavior

Magnetometer (FLYWOO GOKU GM8 MINI V2.0 GPS W/ COMPASS)  to be detected

## Suggested solution(s)

## Additional context

https://pastebin.com/0NAm1FrD
```
# status
System Uptime: 28 seconds
Current Time: 2022-07-10T00:38:31.610+00:00
Voltage: 0.00V (1S battery - NOT PRESENT)
CPU Clock=216MHz, GYRO=MPU6000, ACC=MPU6000, BARO=BMP280
STM32 system clocks:
  SYSCLK = 216 MHz
  HCLK   = 216 MHz
  PCLK1  = 54 MHz
  PCLK2  = 108 MHz
Sensor status: GYRO=OK, ACC=OK, MAG=UNAVAILABLE, BARO=OK, RANGEFINDER=NONE, OPFLOW=NONE, GPS=OK, IMU2=NONE
Stack size: 6144, Stack address: 0x20010000, Heap available: 1820
I2C Errors: 60, config size: 9704, max available config: 32768
ADC channel usage:
   BATTERY : configured = ADC 2, used = ADC 2
      RSSI : configured = ADC 3, used = none
   CURRENT : configured = ADC 1, used = none
  AIRSPEED : configured = none, used = none
System load: 10, cycle time: 510, PID rate: 1960, RX rate: 49, System rate: 9
Arming disabled flags: HWFAIL RX CLI
VTX: not detected
```
---

# version
# INAV/FLYWOOF745NANO 5.0.0 Jun 20 2022 / 10:10:33 (d2adf35c)
# GCC-10.2.1 20201103 (release)

- FC Board name and vendor: FLYWOOF745NANO
- INAV version string:5.0.0

JulioCesarMatias, 2022-07-22:
Something new?

JulioCesarMatias, 2022-08-20:
Something new?

rpandassociates, 2023-07-12:
I’m experiencing same issue frustrating
Configs are set according to gps specs in my case is a iflight v2 gps/compass for barometer and magnometer 
I checked continuity on all wires from fc to gps plug  good solder joints no shorts.

It’s like these pads are not assigned ??? Dead ?

going to pop it in another model and see if betaflight detects the magnetometer if so then that will tell us something :) it would fc specific 

had high hopes for this inav on this hexplorer but about over it at this point haha

closed 2023-11-03 by DzikuVx