iNavFlight/inav #8209: RSSI rescaling issue with ELRS

robustini, 2022-07-11:
## Current Behavior

I had an unexpected failsafe with my long range racer yesterday, checking the log I think I see that the RSSI value went to a fixed 23%, and at that moment the failsafe was activated (first flight in the log).
When this drop occurred, I notice that the rx channels responded without problems and the ELRS link quality was at the maximum values in the sensors on the tx, i.e. at 100, de rest tx and rx were very close, no more than 70 meters in line of air.
I noticed that a [rescaling](https://github.com/iNavFlight/inav/pull/7630) has been done, so I wonder if it really works correctly or if I am doing something wrong.
Doing bench tests I noticed that AUX12, which in my case is RSSI while AUX11 is LQ, drops dramatically, perhaps too much compared to LQ which remains at maximum.

## Steps to Reproduce

1. Put the power of the tx module to a minimum
2. Move the tx away from the rx 

## Expected behavior

AUX16 will drop a lot as you can see from the image while AUX15 always stays at maximum, as if it didn't work.

## Suggested solution(s)

I used to use AUX16 before, now I have set AUX15 (LQ) to see if this is more reliable.

## Additional context

[blackbox_log_2022-07-10_191625.zip](https://github.com/iNavFlight/inav/files/9081020/blackbox_log_2022-07-10_191625.zip)

[image]

---

INAV/ZEEZF7 5.0.0 Jul  4 2022 / 08:25:36 (e3b3e2b9)
GCC-10.2.1 20201103 (release)

robustini, 2022-07-11:
The remote control where I made the red circle, no further than 50 meters from the racer.
The tx warns me in audio by reading the value when the dynamic power changes, it remained at a minimum that is 10 mW, so the link was good when the RSSI was at that 23% (but shouldn't it go to 0?).

[image]

breadoven, 2022-07-11:
Last time I checked ELRS only supports 12 channels. 

And out of interest why are you assigning RSSI and LQ to Aux channels ? I assume you are using the WIDE switch mode ?

robustini, 2022-07-11:
> Last time I checked ELRS only supports 12 channels.
> 
> And out of interest why are you assigning RSSI and LQ to Aux channels ? I assume you are using the WIDE switch mode ?

ELRS V3.0 RC1, full res at 12 and 16 ch.

robustini, 2022-07-11:
https://www.expresslrs.org/2.0/info/signal-health/

0crap, 2022-07-11:
Like breadoven says, there is no need to assign any AUX channels.
RSSI and LQ works out of the box with iNav 5.0

Als try ExpressLRS v2.5.1
v3.0 is still a Release Candidate, although I don't expect any problems regarding RSSI and LQ.

robustini, 2022-07-11:
> Like breadoven says, there is no need to assign any AUX channels. RSSI and LQ works out of the box with iNav 5.0
> 
> Also try ExpressLRS v2.5.1 v3.0 is still a Release Candidate, although I don't expect any problems regarding RSSI and LQ.

Ok, and where do log them?

0crap, 2022-07-11:
Just enable LQ, RSSI, SNR in the OSD, nothing more to do.
Failsafe has nothing to do with AUX channels, just works out of the box.
For RX logging I always grab it from my Taranis (OpenTX) radio.
RSSI in % can also be found in the iNav BBL, but % does not tell me much.
I prefer dBm and that is wat the Taranis log will give you.

robustini, 2022-07-11:
Yes, but this does not eliminate the problem I have reported.
If the failsafe is referring to RSSI and not LQ just look at my log to see what happened.
Link stable and failsafe trips.
I think if i select the AUX channel related to the LQ maybe it takes that as a reference, I try.

0crap, 2022-07-11:
The receiver will see if there is a failsafe or not.
There is nothing you have to setup or do for it to work.
You even do not have to setup the OSD for it, but the OSD is handy so you see for yourself if the RX is working fine.
You will also learn by looking at the OSD values to see if you are near the end of your range.

breadoven, 2022-07-11:
Have to admit it's not easy trying to find out what does trigger Failsafe with ELRS. I get the impression looking at the INAV code it uses "no pulses" since there doesn't appear to be a Failsafe flag set by the Rx with CRSF when failsafe occurs, it just times out when no frames are received. Could be wrong here. So Failsafe is set by the output INAV receives from the Rx not anything related to values being received via Aux channels.

0crap, 2022-07-11:
> So Failsafe is set by the output INAV receives from the Rx not anything related to values being received via Aux channels.

Correct.

robustini, 2022-07-11:
> Really, forget about the BBL. Try it over with setting up your OSD properly and use the OpenTX logs. Nothing is wrong with iNav and ELRS. Good luck.

In fact unfortunately I had not activated the log on the tx, which I have now set.
I'm curious to see if it still happens what the critical values are.

breadoven, 2022-07-11:
You don't use the Aux channel values to trigger Failsafe mode I take it ?

I just set up the ELRS stuff, disabled the RSSI channel, didn't do anything else specific to get Failsafe to work and it just worked all by itself when I turned the Tx off.

0crap, 2022-07-11:
> Why wrong ? INAV simply looks to see if it's receiving a valid signal from the RX based on certain criteria. If those criteria aren't met it decides the signal is bad and goes into Failsafe. Either INAV is misinterpreting the signal from the Rx or the ELRS LQ, RSSI values have an issue.

In the BBL I think you can even find that by looking at rxSignalReceived and/or rxFlightChannelsValid.

robustini, 2022-07-11:
> You don't use the Aux channel values to trigger Failsafe mode I take it ?
> 
> I just set up the ELRS stuff, disabled the RSSI channel, didn't do anything else specific to get Failsafe to work and it just worked all by itself when I turned the Tx off.

I try and tell you flying what happens, thanks.

robustini, 2022-07-11:
> In the BBL I think you can even find that by looking at rxSignalReceived and/or rxFlightChannelsValid.

Yes, rxSignalReceived go to 0 during the failsafe, rxFlightChannelsValid stay to 1.

breadoven, 2022-07-11:
Out of interest, if you do use the Aux channels for LQ or RSSI, does that mean they aren't available for anything else, e.g. switches/pots ? Not obvious how this works from the ELRS info.

robustini, 2022-07-11:
Ok, but I still don't understand what could have happened, without the tx log it is not easy to understand.
I was in dynamic power with the tx module to do a test, at 10 mW, max 250, the tx it warns me vocally as soon as that value changes, it never said anything, always 10 mW.

robustini, 2022-07-11:
> Out of interest, if you do use the Aux channels for LQ or RSSI, does that mean they aren't available for anything else, e.g. switches/pots ? Not obvious how this works from the ELRS info.

AUX11 (ch15) is RSSI, AUX12 is LQ.
And they work on the test bench, you see them vary in the rx channels monitor.
I put the tx in the microwave and in fact AUX11 goes from the maximum value to the one I posted above, but to get AUX12 down I had to go a long way from the rx, but both work.

0crap, 2022-07-11:
> Out of interest, if you do use the Aux channels for LQ or RSSI, does that mean they aren't available for anything else, e.g. switches/pots ? Not obvious how this works from the ELRS info.

What I've seen is that they are always very high up in the AUX range.
Never possible to use that for anything at all that I know.

robustini, 2022-07-11:
> 

AUX11 changes immediately instead.
With the tx two meters away from the sensor RSS1 on the tx it is at -50, and AUX11 seems to me to be around 1900, what I mean is that they work.

0crap, 2022-07-11:
> Ok, but I still don't understand what could have happened, without the tx log it is not easy to understand. I was in dynamic power with the tx module to do a test, at 10 mW, max 250, the tx it warns me vocally as soon as that value changes, it never said anything, always 10 mW.

You will never know without the OpenTX log.
I'm still on v2.5.1 and use fixed output power.
It works much better vs dynamic power.

On v3 they changed how it works by including RSNR in the mix.
Not tested that. Will see when v3 is on stable release.

robustini, 2022-07-11:
Surely testing an RC is not the best, but trusting the failsafe that works nice here the danger is relative.
That's why I started testing with dynamic power.
Today I try to replicate the problem but with the log on the tx active.

0crap, 2022-07-11:
> > 
> 
> AUX11 changes immediately instead. With the tx two meters away from the rx the sensor RSS1 on the tx it is at -50, and AUX11 seems to me to be around 1900, what I mean is that they work.

Really, forget about all this.
The only test you should do is this:
https://www.expresslrs.org/2.0/quick-start/pre-1stflight/#bench-test

robustini, 2022-07-11:
And I already did, I have those values.

breadoven, 2022-07-11:
> What I've seen is that they are always very high up in the AUX range. Never possible to use that for anything at all that I know.

Well I just tested this and if I set channel 12 as the RSSI Channel and enable RSSI in the OSD  ... it just shows the output value of the pot which is assigned to this channel as the "RSSI" value in the OSD. Makes no sense. I assume you have to have nothing else assigned to the channel if you're using it as the RSSI channel ?

And weirder still I noticed that if I move my Tx in a certain way close to the Rx then channels 15 and 16 start jumping around even though this is 12 channel ELRS. Keep the Tx still and the channels stop moving. How does that work ?

robustini, 2022-07-11:
It is obvious that you don't have to have anything on the CH15 of the tx if you assign AUX11 as RSSI
i use 12ch mode, so AUX11 (ch15) is RSSI and AUX12 LQ.

robustini, 2022-07-11:
And instead I removed the source RSSI, and the value is always and in any case logged in the same way in the BBL.

0crap, 2022-07-11:
> @0crap with DJI Caddx you cannot set RSSI or LQ, but only RSSI (on the left). And you have to choose which source to use (on the right). Betaflight is better about this thing.

OMG, you are using DJI!
You should have told that before.
I have no clue about DJI, just analog here.
For analog it's simple, just select the correct OSD items like below.

[image]

I suggest you ask for help on the ExpressLRS Discord.
A lot of really helpful people there which should be able to help you out.

robustini, 2022-07-11:
With the analogue it is obvious that you have both values, even with the new Betaflight you have them on the digital, but with iNav for now not.

OptimusTi, 2022-07-11:
There is no "rescaling" issue. RSSI% is working correctly. You need to check your setup (rf noise, antennas, ect.) and learn how to setup the OSD with DJI.

robustini, 2022-07-11:
> There is no "rescaling" issue. RSSI% is working correctly. You need to check your setup (rf noise, antennas, ect.) and learn how to setup the OSD with DJI.

there isn't much to learn in setting up the OSD for DJI... :P

OptimusTi, 2022-07-11:
I'm going to test with my setup and check the code again. Looks that everything is working as expected. Please check your hardware/setup. I'm going to add hysteresis to the rssi anyways so if there's an issue it will be fixed in the next PR.

robustini, 2022-07-11:
> I'm going to test with my setup and check the code again. Looks that everything is working as expected. Please check your hardware/setup. I'm going to add hysteresis to the rssi anyways so if there's an issue it will be fixed in the next PR.

I took a smooth flight a while ago.
Power set to 100 mW, RSSI source on OSD working, in all honesty I thought better as a link at 868 MHz.
If you want I can share both BBL and OpenTX logs for comparation.

OptimusTi, 2022-07-11:
No thanks. No need.

I don't understand what you said about 868MHz. If you are in Europe it's illegal to use 915MHz.

robustini, 2022-07-11:
> No thanks. No need.
> 
> I don't understand what you said about 868MHz. If you are in Europe it's illegal to use 915MHz.

I know, in fact as I wrote I use 868! :-)

OptimusTi, 2022-07-11:
Do me a favor and (remove propellers). Turn on your quad and goggles and wait for RSSI to show. Then disconnect the battery from your quad? What RSSI do you see?

robustini, 2022-07-11:
Tomorrow I'll try and I'll tell you what I read, now I don't have the Goggles here sorry.
However from what I remember the Goggles should leave you the last read good data.
So if RSSI was at 95 it remains 95, I'll write it to you tomorrow.
For example, the GPS coordinates remain the last good read, so that if the battery is disconnected in a crash you can understand where it fell.

OptimusTi, 2022-07-11:
Let me know what you find. 

For RSSI %, if there is signal loos, the value should be 0% because there's no signal. If it's not going to 0% on signal loss, then it needs to be fixed. This is how RSSI% works for all RF links.

For analog OSD, in the case or signal loss, RSSI in dBm remains with the last value and it was designed to do that to help visualize RF health along the the other RF elements.

closed 2023-10-27 by DzikuVx