iNavFlight/inav #8219: Bluetooth help

Some-Won-1, 2022-07-12:
I am using a Kakute H7 and need the bluetooth connection to set orientation for Inav 5.0. It can connect with an Android tablet and iphone, but there is nothing seen with using the desktop computer. I can use two different dongles to scan for bluetooth devices, but nothing seen after the scan. Each dongle is working with other things at home no problem. Inav doesn't have a version of configurator for Android so I'm stuck. HELP!

Some-Won-1, 2022-07-12:
Perhaps I'm not using the desktop configurator correctly. Do I need to pair with using the OS (win7) first or just go from the configurator?

OptimusTi, 2022-07-12:
You need to setup "serial over Bluetooth" COM ports.

Some-Won-1, 2022-07-13:
Ha yes, Open Settings was hiding in the right click icon in the lower right status bar. It's grayed out else where. There are two versions of BT drivers in the desktop.
Now the configurator connects for a short time Successful then says it can not receive data. Hum...

JulioCesarMatias, 2022-07-22:
Something new?