iNavFlight/inav #8220: inav 5: Manual mode prevents auto from being selected

Natris, 2022-07-12:
I have recently upgraded to inav5 with my matek f411 wte. I have saved a diff, wiped the fc, connected to it, selected wing as default, and copied all settings into cli. All is working fine.

After that I have done the same with another fc; since it is a new plane, and also a wing, and has identical layout of peripherials, I have copied all but the profile's rates.

I am using channel 7 to switch between modes. In my radio i use two 3 level switches: 3 from one switch, and 2 are overriden from the second switch. So there are exactly 5 positions. These are: empty (overriden from second switch), RTH (overridden from second switch), angle, navigate in the direction, and manual mode.
On the original plane, all is working fine. But on the other plane, the empty slot is not there; it seems like "manual" is somehow interfering with it. And since empty mode selects acro, I now do not have a way how to select acro.
Deleting "manual" fixes the issue; re-adding causes it ocurring again. 

On this position of channel 7 there is nothing selected; yet acro is not highlighted. I do not know if it is a display bug or if acro is not in use...
https://drive.google.com/file/d/1-FiTY_8BXSv1TKZfxkK3N7OQ1Xzm-2kq/view?usp=sharing

Original config that I have inserted (without pids)
https://drive.google.com/file/d/14w-EVydZjZosP3wil3ZSHzyYuz0lCwbo/view?usp=sharing

Current diff (with the acro not highlighting):
https://drive.google.com/file/d/1mWoFicOFij3OtePX8vs62Bp0VQP1-KwT/view?usp=sharing

I have simply deleted the manual mode for now...

dump:
https://pastebin.com/YSitCK4Q

# version
# INAV/MATEKF411TE 5.0.0 Jun 20 2022 / 11:11:14 (d2adf35c)
# GCC-10.2.1 20201103 (release)

breadoven, 2022-07-13:
Don't entirely understand how you've set up the switches but the priority of your mode selections is:
RTH
Manual
WP or CRUZ (navigate in the direction ?)
Angle
Acro (Not selectable though, just becomes active if no other mode is selected)

If `ACRO `isn't showing as the current mode then the mode that's overriding will be, e.g. `MANU `will display if it's Manual mode.

Natris, 2022-07-13:
OK I have identified the issue. It seems like if you have an unconfigured magnetometer, it breaks the channel selection logic; simply removing the magnetometer fixes the behaviour. Since the plane is not complete yet, I can not say if configuring it also fixes it, as I do not want to tilt it right now.

closed 2022-07-13 by Natris

MrD-RC, 2022-07-13:
You don't want to use a compass on fixed wing anyway.