iNavFlight/inav #8235: Docker build fail on macOS x86: /src dir dubious ownership

petertorelli, 2022-07-16:
## Current Behavior

Docker fails after setup. Suggestion doesn't fix it, nor does adding it to local `.git/configure`.

```bash
*** Building targets [MATEKF722PX]

fatal: detected dubious ownership in repository at '/src'
To add an exception for this directory, call:

	git config --global --add safe.directory /src

```

## Steps to Reproduce

1. clone the repository
2. ./build.sh MATEKF722PX

## Expected behavior
To build.

- FC Board name and vendor: MATEK F722HD
- INAV version string: head pointer cf0a4dd10e9dffd9a792ad1825210b0ba22a3239

JulioCesarMatias, 2022-07-16:
Why don't you try with WSL? Use my method >> https://github.com/iNavFlight/inav/pull/8236/files

petertorelli, 2022-07-16:
> Why don't you try with WSL? Use my method >> https://github.com/iNavFlight/inav/pull/8236/files

How does this fix the docker issue? I can build fine with cmake/make because the iNav documents are well-written.

JulioCesarMatias, 2022-07-16:
> > Why don't you try with WSL? Use my method >> https://github.com/iNavFlight/inav/pull/8236/files
> 
> How does this fix the docker issue? I can build fine with cmake/make because the iNav documents are well-written, but I'd like to use docker due to the systems I have access to.

This is not to fix the Docker problem... this is for you to build iNav in another way without using Docker, just!

petertorelli, 2022-07-22:
Why would you close the issue and leave a broken docker file in the main branch, including bad docker documentation? Isn't that just going to make the project look bad?

JulioCesarMatias, 2022-07-22:
I thought you solved the problem, why haven't there been any changes here... I'm sorry! So I ask a question: Did you manage to solve the problem?

przemekgalazka, 2022-10-19:
hi, I run into the same problem. How to sort it out using docker build as I am on macbook with apple m1 chip

skeet70, 2022-11-14:
@petertorelli do you mind giving #8558 a try? I was googling a similar issue in a totally different place and came across this one. Figured I could apply what I learned here, though I've never touched this project. If it doesn't work for you, that PR should at least be a starting point for someone to fix this. It let me build the project locally and the produced files are usable. I run docker in rootless mode though, which has a known issue with producing files from a container like this (random uid/guid, regardless of Dockerfile settings):

```console
λ ~/code/oss/inav/ master* ll build
total 2472
drwxr-xr-x 1 100999 100099      68 Nov 14 13:33 bin
-rw-r--r-- 1 100999 100099   17288 Nov 14 13:31 CMakeCache.txt
drwxr-xr-x 1 100999 100099     396 Nov 14 13:31 CMakeFiles
-rw-r--r-- 1 100999 100099    1477 Nov 14 13:31 cmake_install.cmake
-rw-r--r-- 1 100999 100099      41 Nov 14 13:31 docker_cmake.rev
-rw-r--r-- 1 100999 100099 1298029 Nov 14 13:38 inav_6.0.0_MATEKF722PX.hex
-rw-r--r-- 1 100999 100099 1203633 Nov 14 13:31 Makefile
drwxr-xr-x 1 100999 100099      82 Nov 14 13:31 src
```

for a normal user with a normal Docker setup, the uid/guid should result in their filesystem telling them that they own the files. You're on Mac, which has a different bind mount implementation, so things may be weird in another way! Hopefully it works for you though.

PR #8558 (2022-12-23): Fix git security issue with docker build
Fixes #8235. I don't use this project and likely never will, but happened across this issue looking for a solution to my own problem and figured I could fix it. Feel free to close if you don't want it fixed :smile:.

Make the docker user id and group id match the running user's, so that files built by the container are seen as owned by the current user. Also set the current directory as "safe" in git so it's able to inspect the git history.

_note_: this will successfully run in a rootless docker install, but the created files will have random uid/guids [... cut]
closed 2022-12-23 by DzikuVx