iNavFlight/inav #8244: v5.0 - "pos_hold_deadband" unable to arm

StuweFPV, 2022-07-19:
## Current Behavior
I upgraded a plane from 2.6.1 to 5.0 and was unable to arm with the error "unable to arm pitch roll not centered". I got this error quite often so I've checked that "small_angle" is set to 180 and the "pos_hold_deadband" was already increased to 50. Yes the servos are quite off due to some mechanical limits. each aileron is set to 1600 and the elevator 1460. The rest is spot on 1500. I have then gradually increased the "pos_hold_deadband" to 150 without success. Wondering if the limit of 250 is the SUM of all "off 1500" so i set it to 250 and it worked. However:

- I wonder what the consequences are when going flying with such a high value of 250?? I did not find any detailed description
- When activating Autolaunch the Servos now get centered which is scary as i know this plane needs the servos to be offset like it is set in the Output tab - Will the FC be fast enough to level the plane accordingly as with centered servos it will crash for sure!
- Why does something like this fly fine under 2.6.1 and then suddenly not arm at all under 5.0??? What has changed?

## Expected behavior
When you just update a working inav version I expect that it will at least WORK under the new version. Not the new features and fancy stuff but just arm and fly like it did under the old version.

## Suggested solution(s)
Please check if this is a bug or some sort of "new pre-arm check" that made its way somewhere between 3.0-5.0. Why is that even there? I have the option under "Outputs" to change the center of my servos but as a consequence I cannot autolaunch or even arm my plane? somehow this does not make sense. I would prefer this "check" to be disable or at least have the option as a user to disable this check at my own risk.

---
INAV/MATEKF722SE 5.0.0 Jun 20 2022 / 11:19:12 (d2adf35c)
GCC-10.2.1 20201103 (release)

MrD-RC, 2022-07-19:
Pitch/roll not centred usually means you have trim on the transmitter. Check the receiver page. The roll, pitch, and yaw should all be 1500 at rest.

breadoven, 2022-07-19:
This Arming block is caused by `control_deadband `rather than `pos_hold_deadband `(`pos_hold_deadband `is used to trigger RC adjustments during navigation).

The stick position arming check just looks at `rcCommand `on roll and pitch to see if it exceeds `control_deadband`. If it's blocking it's because there is `rcCommand `input when the stick is centred.

Disable Launch, Arm, record a log and see what values you have for `rcCommand `Roll/Pitch when the stick is centred, Should be 1500 or you'll have this problem.

StuweFPV, 2022-07-19:
I went flying today with this plane and did not crash. Last night arming worked well on the bench. Today on the field I got the same message again about unable to arm due to roll/pitch center. So remembering breadoven's words i changed the control_deadband to 30 and the pos_hold_deadband from 250 back to 10. This now seems to work. At least it worked today on 5 flights. But what a strange mistery this is....

Good thing i don't have too many planes where the servos are off center that much so hopefully my other 18 updates will go less painfully. But the flying behavior in 5.0 really sucks... both my 2 planes on v5.0 can't hold level trim. they dive left down. One a pusher and one a jet. never had this under any previous versions with identical settings. I'm honestly considering downgrading again to 2.6.1 - the last really stable working version with a usable autotune that made sense. Sorry to say but since then the releases got worse and worse. I would miss the nice fancy stuff like profiles and switch display - but in the end what counts the most to me is a good stable reliable flying airplane. my peak was 2.6.1
thanks for your help

closed 2022-07-22 by JulioCesarMatias