iNavFlight/inav #8245: Alternate way to enter OSD Menu

StuweFPV, 2022-07-19:
## Current Behavior
Due to some mechanical limits I had to limit my servos in the output tab to 75% way. Everything works fine except entering the OSD menu via stick command. (I never used the other stick commands but I guess they are not working either). I've noticed on other planes that this function requires a servo way of at least ~95% in order to work. Sometimes this is not possible and it also makes little sense as the limitation of a servo way is a valid and widely used setting in iNAV.

## Desired Behavior
Have a alternate way to enter the OSD Menu besides the classy stick command in case of servo limitations or old habits / personal preference.

## Suggested Solution
Add a alternate solution to enter the OSD menu like other systems have i.e. Eagle Tree or PitLab. One way would be to allow the user to define a channel and toggle 3 times within 5 seconds. I.e. use the mode switch and toggle it fast 3x will enter the OSD menu. Toggle once will exit the menu.

## Who does this impact? Who is this for?
Anybody who likes to use the OSD menu in flight or on the ground and has a limitation on one of the servos.
Users that fly more EagleTree / PitLab than iNAV and are tied to their habit.
Mode 1, 3 and 4 pilots that are totally confused how their friends get into that menu.

breadoven, 2022-07-19:
Stick commands use raw stick inputs rather than mixed inputs so should be unaffected by mixed settings.

Check in the receiver tab you are getting the full range of stick movement, 1000 - 2000.

See also https://github.com/iNavFlight/inav/wiki/Fixed-Wing-Guide.

MrD-RC, 2022-07-19:
You shouldn't be limiting in the transmitter. You should do the limiting on the outputs page in iNav.

breadoven, 2022-07-19:
Or limit servo travel in the Mixer page by setting Weight < 100% and adjust centres mechanically.

StuweFPV, 2022-07-19:
Sorry if i was not clear in my description. i never limit, trim or change anything on my transmitter. I do the settings on the Output page. My current file looks like this:

# servo mix
smix reset

smix 0 0 0 100 0 -1
smix 1 1 0 100 0 -1
smix 2 2 1 100 0 -1
smix 3 3 2 100 0 -1
smix 4 4 1 100 0 -1

# servo
servo 0 1000 2000 1528 **80**
servo 1 1000 2000 1528 **80**
servo 2 1000 2000 1488 -100
servo 3 950 2050 1461 125
servo 4 1000 2000 1512 100

I went flying today and in the meantime i'm at 90% but this is still not enough to enter the OSD menu. What's wrong with having a switch to enter that menu?

breadoven, 2022-07-20:
Can you confirm what values you get in the Configurator Receiver tab for Roll and Pitch stick range. Should be 1000 - 2000.

StuweFPV, 2022-07-21:
I did check the values comming from the receiver and they looked ok. not perfect on 2000 but somewhere around 1970. however i found that for some strange reason the rudder CH4 was double reversed. Meaning it's fine for flying and also for the stab but when going full up and right with all sticks where all values should go towards 2000 the rudder went down to 1000. I guess that was the problem. It works now but still a switch would be better - also for entering in-flight :)

another question to the experts: what is the difference rsp. the consequence if i reduce the servo in outputs or the mixer % in mixer. lets assume a straight plane that does not require any mixers. so where should i limit the servo??
cheers guys!

StuweFPV, 2022-07-21:
ok thanks - then at least i did that correctly :)

but in the end does it have any effect or influence any automated modes or pid's or whatever if you limit your servo to 80% in the outputs or the mixer??? or does it not matter at all?

MrD-RC, 2022-07-22:
The mixer mixes how the different control surfaces mix with each other. The outputs determine the limits of the output devices. In that respect, you can think of them the same as the mixer and outputs on a transmitter. 

The navigation and stick commands are all inputs that are assigned to as sources for the mixes. So the changes made in the mixer or outputs effect the navigation in the same way that they would in the flight modes where you are in control. The only exception being manual, where you will have a little more movement, due to the 10-20% room allowed for the stabilisation to work.

zebu14, 2024-03-09:
+1 for the feature to enter OSD menu with a switch input :)

As a former eagletree user, the "mode switch toggle" is very convenient ton enter the menu.

b14ckyy, 2024-03-20:
@zebu14 as we don't support the CMS mid flight, there is no point in sacrificing a channel for the menu. But you can configure a automatic stick channel command on your radio to be executed when you push a button. that's what we do here. https://inavfixedwinggroup.com/guides/transmitter-models/inav-fixed-wing-group-pro-opentx-model/

closed 2024-03-20 by b14ckyy