iNavFlight/inav #8250: No dataflash chip found MATEKF722SE

georgekot, 2022-07-21:
My FC has dataflash chip and i can use it with betaflight... I tried to flash every version of inav and still i couldnt fix it. I also tried simiular FC like MATEKF722 PX etc... but still nothing. I installed every drive i had to and still nothing. I had this problem in the past with betaflight but i used an old txt file with some saved settings and the problem was solved. Im new to inav and i dont have an old txt file that works  to upload.... Help i lost like 6 days trying to fix it... 
PS.
When i flashed ivan the 1st problem was that it couldnt read my gyro, and all the sensors. I fixed this problem by typing in cli set gyro_to_use=1 and all the sensors fixed. Is there any cli command that enables dataflashcip...?how can i fix it.....

JulioCesarMatias, 2022-07-22:
The MATEKF722SE only works with SD Card, it does not have a flash memory chip installed. The MATEKF722-PX, on the other hand, has a flash memory chip (25Q256JVEQ) where iNav is not supported.

closed 2022-07-22 by JulioCesarMatias

JulioCesarMatias, 2022-07-22:
Open a new ISSUE asking to add support for flash memory 25Q256JVEQ