iNavFlight/inav #8251: new target for AocodaRC F7MINI

CGuasco, 2022-07-22:
## Current Behavior

No available target for Aocoda RC F7MINI

## Desired Behavior

please add a new target for the board
only exist for Aocoda RC F7 dual, but not work

## Suggested Solution

## Who does this impact? Who is this for?

Many Aircraft model enthusiasts,there is eight motor or servo driver.
it is easy to update the latest firmware. 
And, finally, is cheap!

## Additional context

in Betaflight work good, but there is pratically no support for GPS  (example Alt Hold & Pos Hold)

JulioCesarMatias, 2022-07-22:
https://github.com/iNavFlight/inav/blob/master/docs/policies/NEW_HARDWARE_POLICY.md

CGuasco, 2022-07-23:
So.... Only constructor can request a new target?
!!! GREAT !!!
Is not so difficult to eliminate second Gyro in supported F7 Dual
(the only difference, according constructor is second Gyro chip).
Hope in github there are source code for different F7Dual, if i can found it... I will do for myself.
Congratulations.
"We don't have interest to create a new target" will be a more honest answer.

CGuasco, 2022-08-02:
New target "Aocoda RC F7 MINI" I did it myself!
For those interested, see here:
https://github.com/iNavFlight/inav/wiki/Building-custom-firmware
starting from the target MATEKF722SE and SPEEDYBEEF7MINI (very similar to aocoda) and using the Betaflight targets as a reference.
I still have some problems with the i2c bus, but the rest works, in the next w.e. i study i2c problems.
see: http://www.robot-eyes.com/public/INAV-AOCODA.png
------------
@JulioCesarMatias esperava uma atitude parecida com a sua de um alemão, nunca de um BeloHorizontino

closed 2022-08-02 by CGuasco

JulioCesarMatias, 2022-08-02:
> 

Você quer que eu faça o que? Quem tem que ter interesse em fazer o iNav funcionar com a FC é a fabricante... Se o fabricante não quiser ter o trabalho de trazer o suporte, ele pode enviar uma FC para algum dos Dev's para que o desenvolvimento seja iniciado. Eu não vou gastar dinheiro do meu bolso para isso... Imagina o quanto de dinheiro eu gastaria sempre que uma FC aparece no mercado para ser adicionada ao iNav?!?!

Ultraexpert, 2022-09-10:
Hey did anyone manage a working hex for 5.1 Thanks

CGuasco, 2022-09-10:
I have a 5.0 working, but is unofficial because the INAV staff thinks this FC is not worthy. (see comments above) so.... sorry, but AOCODA F7 cannot mount INAV.
Seems that Aocoda don't support the team???
If you search on the site you will find the procedures to create your custom version.
Sorry, but this is the INAV policy.

Ultraexpert, 2022-09-10:
Thanks for the info. Do you have a hex you could upload? Or just the h file

Ultraexpert, 2022-09-13:
> I have a 5.0 working, but is unofficial because the INAV staff thinks this FC is not worthy. (see comments above) so.... sorry, but AOCODA F7 cannot mount INAV.
> Seems that Aocoda don't support the team???
> If you search on the site you will find the procedures to create your custom version.
> (or in my previous post)
> Sorry, but this is the hard INAV policy.

I really don’t understand why this target hasn’t been included? I am willing to send the hardware to a developer in order to get this done. Are you willing to post your working hex? Or the target so I can compare? I have found some community members willing to test with their own hardware. Thanks

CGuasco, 2022-09-13:
I also don't understand why this target is not appreciated by the INAV team.
I am a user and not a developer.
You have to ask @JulioCesarMatias.
He told me he can't spend time and money on a card that doesn't support INAV (see his post back to my grievances). Which support not given can be hypothesized!
However I cannot upload the code. It's against the rules, INAV doesn't want it.
At most I could email it as it is.

sensei-hacker, 2022-12-13:
The target from the manufacturer is here:
https://github.com/dlt2018/inav/tree/aocodarcf7mini/src/main/target/AOCODARCF7MINI

The manufacturer is trying to work with the core devs to get it merged, but there seems to be communication issues.

derFrickler, 2023-02-02:
Just tried the new Target on INAV 6 RC1. 
I was trying to setup an X8 Quad with PWM Motor Signal, Output M5 and M6 are not working, all other outputs produce a valid signal, but no PWM signal on M5/M6.

sensei-hacker, 2023-02-02:
> Just tried the new Target on INAV 6 RC1. I was trying to setup an X8 Quad with PWM Motor Signal, Output M5 and M6 are not working, all other outputs produce a valid signal, but no PWM signal on M5/M6.
> 
> The Betaflight Target for this Board although only supports 7 Motors...

derFrickler, can you try the official target?
[image]

derFrickler, 2023-02-02:
Actually it was this version that i used: 

Flight controller info, identifier: INAV, version: 6.0.0
Running firmware released on: Jan 29 2023 10:25:59
Board: AO7M, version: 0

Ultraexpert, 2023-02-09:
Did you use version 1or2 I see in the target file that v2 has to be used to use m5/m6

derFrickler, 2023-02-09:
I tried different Versions, was never able to get the 8 motors working. 
@sensei-hacker has confirmed that this is a but and max 7 motors are possible right now.

sensei-hacker, 2023-02-09:
> Did you use version 1or2 I see in the target file that v2 has to be used to use m5/m6

V2 hardware is needed in order to have 8 outputs.
V1 hardware has 7.

Ultraexpert, 2023-02-09:
@derFrickler thank you I saw the line in the target file. Looks like motor 7 is the culprit

Ultraexpert, 2023-03-09:
I tried both versions of the 6.0rc3 and none work for me. I get no motor response or servo response

sensei-hacker, 2023-03-09:
@Ultraexpert if you want to head over to the INAV Discord I'd be glad to give you a hand getting your build working.

sensei-hacker, 2023-03-09:
Sinde this was apparently unclear -
If you have a V1 board, that board has 7 PWM outputs.

INAV refers to them as S1-S7

If you try to tell INAV that you've connected something to S12 or S8 or S14, things will not work.

sensei-hacker, 2023-03-09:
Working config file for the version 1 hardware, since it seems to be tricky for some:
https://github.com/sensei-hacker/inav_unofficial_targets/blob/da0105394de6abd477e9283bf4df5219fa96940c/src/main/target/AOCODARCF7MINI/zohd_aocoda_mini_v1_inav_6_dump.txt

The above is for a plane.

Ultraexpert, 2023-03-10:
@sensei-hacker thank you for helping