iNavFlight/inav #8252: Speedybee F7 V3 support

Ivosch, 2022-07-23:
Hello all,
do you plan to support Speedybee F7 V3 flight controller? When? Or can I use V2 firmware on V3 hardware?

I.

DzikuVx, 2022-07-23:
Planned for  INAV 5.1

petertorelli, 2022-08-09:
@Ivosch Have you tried V2?

AmineBahadi, 2022-08-09:
hello,

I've tried the v2 it didn't work it's not detecting the gyro and opticalflow and i tried to change the pinio to match the v3 and it didn't work to 

it has something to do with removing USE_EXTI, USE_GYRO_EXTI and USE_MPU_DATA_READY_SIGNAL defines i think 

If any one can help it will be great 🙏

Tjjohnston, 2022-08-10:
I just bought the V3 and wired it all up, not thinking I would have any issues since SpeedyBee said it was compatible. Now I am stuck with a flight controller that I cant use. My SpeedyBee V2 ESC died in my second flight, support is giving me the run around. I bought V3, still cant use it. Any idea how long this will take? I am really not looking forward to unsoldering everything to return it.

petertorelli, 2022-08-12:
Hi, @AmineBahadi ... Where did you find the GPIO port/pin mappings for the IMU on the V3 board?

AmineBahadi, 2022-08-12:
Hi @petertorelli flash the board on betaflight, and then go to the cli tab on the betaflight GUI type the command : resource and you will have all the ports/pins mapping

Ultraexpert, 2022-08-12:
I hacked together a horrid version and get errors when building… it’s a start maybe

AmineBahadi, 2022-08-12:
Hi @Ultraexpert i see that you used the betaflight code here it will not work the inav variables are not the same as betaflight 😅

I did manage to build a version but there's stuff need to be changed on firmware that are obsolete.

Ultraexpert, 2022-08-12:
> Hi @Ultraexpert i see that you used the betaflight code here it will not work the inav variables are not the same as betaflight 😅
> 
> I did manage to build a version but there's stuff need to be changed on firmware that are obsolete.

Gotcha I saw that obsolete stuff. Sadly I don’t have the hardware to get the actual pins. By variables do you mean the pin assignment?

Ultraexpert, 2022-08-12:
Let me know if my Build can spin the motors?

Tjjohnston, 2022-08-12:
I just saw this email, I’m working one of my rare 18 hour work days today. I won’t be done until late this evening. I’ve got the V3 hardware I’m happy to test things for the build, I’ve got a custom drone build that can handle crashes it’s fully protected.

Tjjohnston, 2022-08-12:
Oh, if you guys wanted to post one of the builds I can test it, I didn't see it wasn't posted.

Ultraexpert, 2022-08-12:
@AmineBahadi would you be interested in letting me see your build version? or upload your progress? I'm shocked this has been almost a month and nobody can edit a couple paragraphs? @Tjjohnston the version I added will be on my fork of iNav, check the timers I know I made it rough

Ultraexpert, 2022-08-12:
just saw a message from DzikuVx its done guys! all you need to do is wait a few days!

Tjjohnston, 2022-08-12:
Where did you hear it was done?? Did he say when it would be out or did he
say just generally soon? I heard it will be out with 5.1 but that would be
the end of the month.

Ultraexpert, 2022-08-12:
From Pawel [DzikuVx]
@Ultraexpert thanks for the contribution, but the official target is just tested by Speedybee and should be merged in a few days

petertorelli, 2022-08-13:
@Ultraexpert 
I think it would be helpful to know how to do these ports (and offload @DzikuVx from having to do everything). If the STM processor is the same as one in the iNav library, and if all the peripherals are known, then it should just be a matter of assigning the correct hardware connectivity for UARTs, I2C, etc. Right? (It's not like the clocking changes... so that's about all there is!) Anyway, if @AmineBahadi is right and something in the HAL had to change due to interrupts, I'm really curious to see what needed to be fixed in iNav itself to support this board. (Also, at one of the milestones to abstract the peripheral layer so that nRF-based flight controllers (or NXP) could be used, too.)

Unless there already is a porting guide somewhere that I overlooked...?

closed 2022-08-23 by DzikuVx

Tjjohnston, 2022-10-11:
Does anyone know if speedybee got this done?

Ultraexpert, 2022-10-11:
@Tjjohnston did Speedybee do what?

hlalsaed, 2023-05-07:
> I just bought the V3 and wired it all up, not thinking I would have any issues since SpeedyBee said it was compatible. Now I am stuck with a flight controller that I cant use. My SpeedyBee V2 ESC [...]

I got exactly the same problem you had, I don’t recommend any body to get speedybee f7v3