iNavFlight/inav #8265: is Preset Airframes comming back in INAV5 ? and Request to include Development boards into the board lists

synersignart, 2022-07-28:
## Current Behavior
I miss this feature on the old INAV configurator 

## Desired Behavior

some Air frames still missing proper Motor and Servo Mapping on certain boards
Custom frame are not too flexible to choose where the motor or servos can be place 

## Suggested Solution

id Request to to explore  Development boards 
like the STM  bluepill & Blackpill
             STM Discovery board
             STM Neucleo

all as Target with a wide selection of Sensors 
intended for  experimenters  who wish to build from development boards

MrD-RC, 2022-07-28:
This is really two separate issues.

Presets will not be coming back to configurator. To be honest. Most of them were bad. Plus, the new defaults and autotune works well for most planes.

Which servo/motor setups do you feel are missing? Please leave a separate issue on the [Configurator GitHub](https://github.com/iNavFlight/inav-configurator/issues).

wawanlost, 2022-07-29:
menambahkan preset model sangat baik...seprti sayap tetap dan multicopter...dan mengaktifkan kembali ppm receiver...itu sangat dihargai,,,terimakasih

synersignart, 2022-07-30:
> This is really two separate issues.
>
> Presets will not be coming back to configurator. To be honest. Most of them were bad. Plus, the new defaults and autotune works well for most planes.
>
> Which [...]

ah I see ok , well check to this .id wish it is available in multicoter Autotune.

synersignart, 2022-07-30:
> menambahkan preset model sangat baik...seprti sayap tetap dan multicopter...dan mengaktifkan kembali ppm receiver...itu sangat dihargai,,,terimakasih

unfortunately the developers decided to drop the PPM . but if your like me who still uses old PPM and PWM radios there is a PPM PWM SBUS converter in the market you can use . be aware it may require an SBUS inverter circuit on some models

closed 2022-08-04 by DzikuVx